arendst/Tasmota #22468: Failure when doing pairing with matter.js/OpenHab matter controller.

TylerDurden-23, 2024-11-11:
### PROBLEM DESCRIPTION
Openhab uses the matter.js standard implementation when doing matter pairing.
When trying to pair it with a tasmota device using a pairing code, the process fails because Tasmota reports a:

21:24:16.219 MTR: rejected packet due to invalid MIC

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [ ] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [ ] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [ ] Searched the problem in [discussions](https://github.com/arendst/Tasmota/discussions)
- [ ] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [ ] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [X] Device used (e.g., Sonoff Basic): Multiple ESP32 devices
- [X] Tasmota binary firmware version number used: 14.3.0
  - [X] Pre-compiled
  - [ ] Self-compiled
- [X] Flashing tools used: OTA Update
- [ ] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```lua
  Configuration output here:

- [ ] Set `weblog` to 4 and then, when you experience your issue, provide the output of the Console log:
```lua
  Console output here:
21:24:15.341 MTR: UDP received from [fe80::2b25:b8e6:3c79:91a8%st1]:55188
21:24:15.348 BRY: GC from 27737 to 17870 bytes, objects freed 164/224 (in 2 ms) - slots from 456/486 to 229/456
21:24:15.353 MTR: >Received  (     0) PBKDFParamRequest from [fe80::2b25:b8e6:3c79:91a8%st1]:55188
21:24:15.368 MTR: +Session   ( 46826) from '[fe80::2b25:b8e6:3c79:91a8%st1]:55188'
21:24:15.383 MTR: <Replied   (     0) PBKDFParamResponse
21:24:15.390 MTR: sending packet to '[fe80::2b25:b8e6:3c79:91a8%st1]:55188'
21:24:15.419 MTR: UDP received from [fe80::2b25:b8e6:3c79:91a8%st1]:55188
21:24:15.425 MTR: .          Removed packet from sending list id=186881462
21:24:15.428 MTR: >Received  (     0) PASE_Pake1 from [fe80::2b25:b8e6:3c79:91a8%st1]:55188
21:24:16.060 MTR: <Replied   (     0) PASE_Pake2
21:24:16.064 MTR: New Commissioning (PASE id=46826) from [fe80::2b25:b8e6:3c79:91a8%st1]:55188
21:24:16.070 MTR: sending packet to '[fe80::2b25:b8e6:3c79:91a8%st1]:55188'
21:24:16.075 MTR: UDP received from [fe80::2b25:b8e6:3c79:91a8%st1]:55188
21:24:16.081 MTR: .          Duplicate unencrypted message = 57291015 ref = 57291015
21:24:16.168 MTR: UDP received from [fe80::2b25:b8e6:3c79:91a8%st1]:55188
21:24:16.174 MTR: .          Removed packet from sending list id=186881463
21:24:16.177 MTR: >Received  (     0) PASE_Pake3 from [fe80::2b25:b8e6:3c79:91a8%st1]:55188
21:24:16.187 MTR: <Replied   (     0) StatusReport
21:24:16.193 MTR: sending packet to '[fe80::2b25:b8e6:3c79:91a8%st1]:55188'
21:24:16.196 MTR: add_session local_session_id=46826 initiator_session_id=50357
21:24:16.203 MTR: UDP received from [fe80::2b25:b8e6:3c79:91a8%st1]:55188
21:24:16.215 MTR: decode header: local_session_id=46826 message_counter=114224690
21:24:16.219 MTR: rejected packet due to invalid MIC
21:24:16.693 MTR: UDP received from [fe80::2b25:b8e6:3c79:91a8%st1]:55188
21:24:16.697 MTR: decode header: local_session_id=46826 message_counter=114224690
21:24:16.700 MTR: .          Duplicate encrypted message = 114224690 counter=114224690
21:24:16.995 MTR: UDP received from [fe80::2b25:b8e6:3c79:91a8%st1]:55188
21:24:17.000 MTR: decode header: local_session_id=46826 message_counter=114224690
21:24:17.003 MTR: .          Duplicate encrypted message = 114224690 counter=114224690
21:24:17.691 MTR: UDP received from [fe80::2b25:b8e6:3c79:91a8%st1]:55188
21:24:17.695 MTR: decode header: local_session_id=46826 message_counter=114224690
21:24:17.698 MTR: .          Duplicate encrypted message = 114224690 counter=114224690
21:24:18.509 MTR: UDP received from [fe80::2b25:b8e6:3c79:91a8%st1]:55188
21:24:18.513 MTR: decode header: local_session_id=46826 message_counter=114224690
21:24:18.516 MTR: .          Duplicate encrypted message = 114224690 counter=114224690

Openhab/matter.js Output:
18:10:39.344 [TRACE] [.internal.util.MatterWebsocketService] - MdnsBroadcaster: Announcement Generator: Fabric id: 94ebef79d182a732/15483111525282463271 qname: 94EBEF79D182A732-D6DF0FE63237BE27._matter._tcp.local port: 54666 interface: vEthernet (Default Net)
18:10:39.344 [TRACE] [.internal.util.MatterWebsocketService] - MdnsBroadcaster: Announcement Generator: Fabric id: 94ebef79d182a732/15483111525282463271 qname: 94EBEF79D182A732-D6DF0FE63237BE27._matter._tcp.local port: 54666 interface: vEthernet (nat)
18:10:39.352 [TRACE] [.internal.util.MatterWebsocketService] - MdnsBroadcaster: Announcement Generator: Fabric id: 84138387a2e5c1f7/8418225184102985051 qname: 84138387A2E5C1F7-74D38B3409B4AD5B._matter._tcp.local port: 62234 interface: vEthernet (Default Net)
18:10:39.352 [TRACE] [.internal.util.MatterWebsocketService] - MdnsBroadcaster: Announcement Generator: Fabric id: 84138387a2e5c1f7/8418225184102985051 qname: 84138387A2E5C1F7-74D38B3409B4AD5B._matter._tcp.local port: 62234 interface: vEthernet (nat)
18:10:39.361 [TRACE] [.internal.util.MatterWebsocketService] - MdnsScanner: Adding operational device 84138387A2E5C1F7-74D38B3409B4AD5B._matter._tcp.local in cache (interface 10, ttl=120) with TXT data: SII: 500 SAI: 300 SAT: 4000 T: 0 DT: undefined PH: undefined ICD: 0 VP: undefined DN: undefined RI: undefined PI: undefined
18:10:39.370 [TRACE] [.internal.util.MatterWebsocketService] - MdnsScanner: Adding operational device 84138387A2E5C1F7-74D38B3409B4AD5B._matter._tcp.local in cache (interface 10, ttl=120) with TXT data: SII: 500 SAI: 300 SAT: 4000 T: 0 DT: undefined PH: undefined ICD: 0 VP: undefined DN: undefined RI: undefined PI: undefined
18:10:39.652 [TRACE] [.internal.util.MatterWebsocketService] - MessageExchange: Resubmit message 99458278 (retransmission attempt 2, backoff time 610ms))
[... 25 lines cut ...]
18:10:40.583 [TRACE] [.internal.util.MatterWebsocketService] - ExchangeManager: Message » id: 65534/952/247800933 type: 1/2 acked: undefined reqAck: true duplicate: false payload: 1536001724020024031d2404031818290324ff0c18
18:10:41.183 [TRACE] [.internal.util.MatterWebsocketService] - MessageExchange: Resubmit message 247800933 (retransmission attempt 3, backoff time 915ms))
18:10:41.184 [TRACE] [.internal.util.MatterWebsocketService] - ExchangeManager: Message » id: 65534/952/247800933 type: 1/2 acked: undefined reqAck: true duplicate: false payload: 1536001724020024031d2404031818290324ff0c18
18:10:41.490 [INFO ] [pport.EventAdminConfigurationNotifier] - Sending Event Admin notification (configuration successful) to org/ops4j/pax/logging/Configuration
18:10:42.099 [TRACE] [.internal.util.MatterWebsocketService] - MessageExchange: Resubmit message 247800933 (retransmission attempt 4, backoff time 1477ms))
18:10:42.100 [TRACE] [.internal.util.MatterWebsocketService] - ExchangeManager: Message » id: 65534/952/247800933 type: 1/2 acked: undefined reqAck: true duplicate: false payload: 1536001724020024031d2404031818290324ff0c18
18:10:43.579 [TRACE] [.internal.util.MatterWebsocketService] - MessageExchange: Wait additional 4752ms for processing time and peer resubmissions after all our resubmissions
18:10:48.332 [TRACE] [.internal.util.MatterWebsocketService] - InteractionMessenger: Retransmission limit reached, trying to reconnect and resend the message.
18:10:59.563 [DEBUG] [.internal.util.MatterWebsocketService] - Controller: Error executing function pairNode:
18:10:59.564 [DEBUG] [.internal.util.MatterWebsocketService] - Controller: Stack trace: Error
18:10:59.564 [TRACE] [.internal.util.MatterWebsocketService] - matter: Sending response: {"type":"response","message":{"type":"resultError","id":"587af880-9726-4cb1-8765-25a385d32966","result":"undefined","error":""}}
18:10:59.564 [TRACE] [.internal.util.MatterWebsocketService] - none:       at StandardTimer.eval [as callback] (webpack://matter-server/./node_modules/@matter/general/dist/cjs/util/DataReadQueue.js?:53:24)
18:10:59.564 [DEBUG] [internal.client.MatterWebsocketClient] - onWebSocketText {"type":"response","message":{"type":"resultError","id":"587af880-9726-4cb1-8765-25a385d32966","result":"undefined","error":""}}
[... cut ...]
18:10:59.567 [TRACE] [.internal.util.MatterWebsocketService] - none:   at StandardTimer.eval [as callback] (webpack://matter-server/./node_modules/@matter/general/dist/cjs/util/DataReadQueue.js?:53:24)
18:10:59.567 [TRACE] [.internal.util.MatterWebsocketService] - none:   at Timeout.eval [as _onTimeout] (webpack://matter-server/./node_modules/@matter/general/dist/cjs/time/Time.js?:133:12)
18:10:59.567 [TRACE] [.internal.util.MatterWebsocketService] - none:   at listOnTimeout (node:internal/timers:573:17)
18:10:59.567 [TRACE] [.internal.util.MatterWebsocketService] - none:   at process.processTimers (node:internal/timers:514:7)

s-hadinger, 2024-11-12:
Thanks for the detailed logs. This is weird, it happens very early in the provisioning process. This part of code hasn't changed in ages and always worked with all Matter controllers.

Does the error happen each time? Do you have a way to reproduce?

s-hadinger, 2024-11-12:
I tried using matter.js as controller and I can reproduce indeed

closed 2024-11-12 by s-hadinger

TylerDurden-23, 2024-11-12:
Hi, thanks for the extremely fast fix. I already retested it with the OpenHab Matter implementation - and it works now! :)

PR #22470 (2024-11-12): Matter provisioning with matter.js controller
## Description:

Fix Matter provisioning that appears when using matter.js controller:
- the "N" parameter in encrypting and decrypting frames must be `sec_flags` and not `flags`- it never appeared before because those two flags are usually `00`
- reduced max frame size to avoid error: `Ignoring UDP message with size 1237 from udp://xxx%en0:5540, which is larger than the maximum allowed size of 1232.`

**Related issue (if applicable):** fixes #22468

## Checklist:
  - [x] The pull request is done against the latest development branch
  - [x] Only r [... cut]