arendst/Tasmota #22479: Can't set a 'HumOffset' bigger/smaller than -10.0

mircsicz, 2024-11-13:
### PROBLEM DESCRIPTION
Can't set a 'HumOffset' bigger/smaller than -10

I've followed this [docs](https://tasmota.github.io/docs/Commands/#sensors)

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [ ] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [ ] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [ ] Searched the problem in [discussions](https://github.com/arendst/Tasmota/discussions)
- [ ] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [ ] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [x] NodeMCU
- [x] Tasmota binary firmware version number used: 14.3.0.6
  - [x] Pre-compiled
- [x] Flashing tools used: tasmotizer
- [ ] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```lua
19:57:50.045 RSL: RESULT = {"GPIO0":{"0":"None"},"GPIO1":{"0":"None"},"GPIO2":{"1184":"DHT11"},"GPIO3":{"0":"None"},"GPIO4":{"0":"None"},"GPIO5":{"0":"None"},"GPIO9":{"0":"None"},"GPIO10":{"0":"None"},"GPIO12":{"0":"None"},"GPIO13":{"0":"None"},"GPIO14":{"0":"None"},"GPIO15":{"0":"None"},"GPIO16":{"0":"None"},"GPIO17":{"0":"None"}}
```
- [ ] Provide the output of this command: `Status 0`:
```lua
19:58:34.818 RSL: STATUS = {"Status":{"Module":18,"DeviceName":"NodeMCU-Office","FriendlyName":["DVES_%06X"],"Topic":"tasmota_1F3D3D","ButtonTopic":"0","Power":"0","PowerLock":"0","PowerOnState":3,"LedState":1,"LedMask":"FFFF","SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0,"InfoRetain":0,"StateRetain":0,"StatusRetain":0}}
19:58:34.826 RSL: STATUS1 = {"StatusPRM":{"Baudrate":115200,"SerialConfig":"8N1","GroupTopic":"tasmotas","OtaUrl":"http://ota.tasmota.com/tasmota/release/tasmota-sensors.bin.gz","RestartReason":"Software/System restart","Uptime":"0T00:24:07","StartupUTC":"2024-11-13T18:34:27","Sleep":50,"CfgHolder":4617,"BootCount":9,"BCResetTime":"2024-11-13T18:07:01","SaveCount":30,"SaveAddress":"F6000"}}
19:58:34.830 RSL: STATUS2 = {"StatusFWR":{"Version":"14.3.0.6(50b6f74-sensors)","BuildDateTime":"2024-11-12T20:51:03","Boot":31,"Core":"2_7_8","SDK":"2.2.2-dev(38a443e)","CpuFrequency":80,"Hardware":"ESP8266EX","CR":"397/699"}}
19:58:34.834 RSL: STATUS3 = {"StatusLOG":{"SerialLog":2,"WebLog":2,"MqttLog":0,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["JuMi-Officina","Recovery"],"TelePeriod":300,"Resolution":"558180C0","SetOption":["00008009","2805C80001000600003C5A0A192800000000","00000080","00006000","00004000","00000000"]}}
19:58:34.853 RSL: STATUS4 = {"StatusMEM":{"ProgramSize":719,"Free":284,"Heap":19,"ProgramFlashSize":4096,"FlashSize":4096,"FlashChipId":"16301C","FlashFrequency":40,"FlashMode":"DOUT","Features":["0809","8F9A8787","0415A005","B7F7BFCF","05DA9BC4","64367CC7","00084052","20000000","54000020","0008C081","00000000"],"Drivers":"1,2,!3,!4,!5,!6,7,!8,9,10,12,!14,!16,!17,!20,!21,!24,29,!34,62,!65,!66,!68","Sensors":"1,2,3,4,5,6,7,8,9,10,11,12,13,14,15,17,18,19,20,21,22,26,31,34,37,39,40,42,43,45,51,52,55,56,58,59,62,64,66,67,74,98,103,105","I2CDriver":"7,8,9,10,11,12,13,14,15,17,18,20,24,29,31,36,41,42,44,46,48,69,76,89"}}
19:58:34.862 RSL: STATUS5 = {"StatusNET":{"Hostname":"tasmota-1F3D3D-7485","IPAddress":"192.168.30.72","Gateway":"192.168.30.254","Subnetmask":"255.255.255.0","DNSServer1":"192.168.30.254","DNSServer2":"0.0.0.0","Mac":"B4:E6:2D:1F:3D:3D","Webserver":2,"HTTP_API":1,"WifiConfig":4,"WifiPower":17.0}}
19:58:34.865 RSL: STATUS6 = {"StatusMQT":{"MqttHost":"10.10.191.91","MqttPort":1883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_1F3D3D","MqttUser":"tasmota","MqttCount":0,"MAX_PACKET_SIZE":1200,"KEEPALIVE":30,"SOCKET_TIMEOUT":4}}
19:58:34.871 RSL: STATUS7 = {"StatusTIM":{"UTC":"2024-11-13T18:58:34Z","Local":"2024-11-13T19:58:34","StartDST":"2024-03-31T02:00:00","EndDST":"2024-10-27T03:00:00","Timezone":"+01:00","Sunrise":"07:57","Sunset":"17:11"}}
19:58:34.879 RSL: STATUS10 = {"StatusSNS":{"Time":"2024-11-13T19:58:34","DHT11":{"Temperature":30.2,"Humidity":81.0,"DewPoint":26.6},"TempUnit":"C"}}
19:58:34.884 RSL: STATUS11 = {"StatusSTS":{"Time":"2024-11-13T19:58:34","Uptime":"0T00:24:07","UptimeSec":1447,"Heap":19,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":0,"Wifi":{"AP":1,"SSId":"JuMi-Officina","BSSId":"AC:84:C6:6B:4A:B8","Channel":1,"Mode":"11n","RSSI":100,"Signal":-22,"LinkCount":1,"Downtime":"0T00:00:04"}}}
```
- [ ] Set `weblog` to 4 and then, when you experience your issue, provide the output of the Console log:
```lua
0:11:11.533 WIF: Checking connection...
20:11:11.993 CMD: HumOffset -11.0
20:11:11.995 SRC: WebConsole from 192.168.30.68
20:11:11.997 CMD: Grp 0, Cmd 'HUMOFFSET', Idx 1, Len 5, Pld -11, Data '-11.0'
20:11:11.001 RSL: RESULT = {"HumOffset":-10.0}
```

### TO REPRODUCE
_Steps to reproduce the behavior:_
On Console enter 'HumOffset -11.0'

```
20:00:37.415 CMD: HumOffset -11.0
20:00:37.421 RSL: RESULT = {"HumOffset":-10.0}
```
Where the RESULT is just what worked to set last...

### EXPECTED BEHAVIOUR
_A clear and concise description of what you expected to happen._
A prompt back telling me that it set the 'HummOffset' to what I requested

### SCREENSHOTS
[image]

sfromis, 2024-11-13:
Tasmota is in general not issuing detailed error messages to hand-hold you through incorrect usage, that would not be feasible with the large amount of available functions in the limited space of a small MCU like the ESP8266. As designed, no issue to be fixed.

Also it would be irrelevant with such a huge offset, as you cannot expect **badly** wrong values to be fixable with a simple up/down correction.

mircsicz, 2024-11-13:
OK, I get it...

The DHT11 were just the first an easiestly available, I've other's at hand as well...

closed 2024-11-13 by mircsicz