arendst/Tasmota #22482: Softreset init problem Tasmota 4.3.3 on Wemos D1 and DHT22

EkkiBrue, 2024-11-14:
Tasmota v14.3.0 on Wemos D1 with a DHT22 (AM2302) connected to 5v and D4

Pinout: DHT22 (AM2302) — Wemos D1

```
+	 	—	 	1	 	—	 	5v
OUT	 	—	 	2	 	—	 	D4
-	 	—	 	3	 	—	 	G
```

I reported this in a resolved DHT22 thread - but it seems the topic there was already closed . So I have to bring this up again as a new bug as it´s not exactly correlating to the [former problem](https://github.com/arendst/Tasmota/issues/5619):

After a softreset from inside Tasmota, the data immediately becomes invalid (null). Only unplugging and replugging the DATA-Cable will restart the "send data process" again.

After reset via resetbutton or unplugging the power cord, data is displayed correctly and seems to remain.

I made a short video to make clear what my actual problem is. 

[https://m1z.eu/tasmobug](https://m1z.eu/tasmobug)
pw: tasmota_14.3.0

Hope it helps to make clear that the bug has to do something with the init-part of the reset-code of Tasmota...

tsmoneta, 2024-11-14:
Does your version of the DHT22 have an onboard 10 Kohm resistor between VCC and the data pin? If not, you need to add one.

tsmoneta, 2024-11-16:
The pull-up resistor on a DHT22 sensor is used to maintain the data line at the Vcc level when the sensor is not actively sending data, ensuring proper communication with the microcontroller by defining a clear voltage state for the data signal to transition between high and low levels during data transmission; it's essentially a mechanism to keep the data line in a known state when not being actively driven by the sensor itself. Some versions include the resistor, some don't.

EkkiBrue, 2024-11-17:
Hi @tsmoneta . Thank you for your detailed answer. I guess it´s not the problem as my platine based DHT22 seems to have the resistor onboard. Please look at this picture: My Modell is the one on the black PCB.

[https://deepbluembedded.com/arduino-dht22-library-code-examples tutorial/#:~:text=How%20to%20connect%20DHT22%20to%20Arduino%3F
](https://deepbluembedded.com/arduino-dht22-library-code-examples-tutorial/#:~:text=but%20not%20beneficial.-,DHT22%20Sensor%20Pinout,-There%20are%20two)

https://deepbluembedded.com/wp-content/uploads/2023/06/DHT22-Pinout-768x890.jpg

crhass, 2024-11-28:
I think you are using GPIO2, try GPIO14

https://github.com/arendst/Tasmota/issues/3522

closed 2024-12-29 by github-actions[bot]