arendst/Tasmota #22492: Word "&curren" inside a variable or inside a comment is replaced by "¤" character after saving a file on file system

Phlavio-SPR, 2024-11-15:
### PROBLEM DESCRIPTION
Trying to use an API to get wheather information from "api.open-meteo.com", I've found that current version os Tasmota for ESP32 cant process the word "&curren" on file system. Ater using this word inside any file, on a variable or on a comment, the system itself replaces it to "¤", making it impossible to use that API.

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [ ] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [ ] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [ ] Searched the problem in [discussions](https://github.com/arendst/Tasmota/discussions)
- [ ] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [ ] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [ ] Device used (e.g., Sonoff Basic): _____
- [ ] Tasmota binary firmware version number used: _____
  - [ ] Pre-compiled
  - [ ] Self-compiled
- [ ] Flashing tools used: _____
- [ ] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```lua
  Configuration output here:

```
- [ ] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```lua
  Rules output here:

```
- [ ] Provide the output of this command: `Status 0`:
```lua
  STATUS 0 output here:

```
- [ ] Set `weblog` to 4 and then, when you experience your issue, provide the output of the Console log:
```lua
  Console output here:

```

### TO REPRODUCE
_Steps to reproduce the behavior:_
Just create a file on ESP32 version with a comment or variable containing the word "&curren", saving it and then reading it back.

### EXPECTED BEHAVIOUR
_A clear and concise description of what you expected to happen._

This problem prevents me to use an API that has the word "&current" on its address. It works fine when the the device is restarted for the first time after saving it, but doesn't work on subsequent booting because the actual address is modified by itselt to ""¤t" afterwards.

### SCREENSHOTS
When you do this:

[image]

After saving the file, you get this:

[image]

### ADDITIONAL CONTEXT
_Add any other context about the problem here._

**(Please, remember to close the issue when the problem has been addressed)**

sfromis, 2024-11-15:
Berry has no problem *processing* a file including `&curren` (or other [HTML  character entities](https://en.wikipedia.org/wiki/List_of_XML_and_HTML_character_entity_references#List_of_character_entity_references_in_HTML)), as such files can be written and read without the issue you see.

Instead, this issue is *only* about using that simplistic file edit function available in the web UI, which becomes unworkable. This is not really recommended for programming anyway, to instead use a better editor, and uploading files.

If you "insist" on coding via the low grade web UI edit function, you can still use a workaround like this (available all places you use a string):
`print('here is a string including &' 'current and again &' 'current')`
Not very elegant, but workable.

This implicit concatenation is resolved at compile time, hence no runtime "cost". It is designed for having long string spanning multiple lines, but work on one line too.

Phlavio-SPR, 2024-11-15:
@sfromis Thank you for a lighting fast reply.
Both options solve the problems I was facing.
Since there is something not quite well with low grade web UI, should I close this issue for now?

sfromis, 2024-11-15:
While you can work around it, I'd still say that it is a detail worthy of fixing, as the [UfsEditor](https://github.com/arendst/Tasmota/blob/0947860c831de00f8f05aa474f548bcc52ce8127/tasmota/tasmota_xdrv_driver/xdrv_50_filesystem.ino#L1571C6-L1571C15) function of `xdrv_50_filesystem.ino` is not doing any escaping. This can also hit code including `textarea` tags.

s-hadinger, 2024-11-17:
I thought we fixed the lack of html escaping a while ago. Worth revisiting what we missed

sfromis, 2024-11-17:
And example of where it breaks at a `</textarea>` tag:
https://github.com/Staars/berry-examples/blob/main/bleKeyLegacy.be
[image]

sfromis, 2024-11-21:
Works better now, for both types.

closed 2024-11-25 by arendst