arendst/Tasmota #22510: SwitchMode 5 not triggering on first press after ~5s have elapsed

madduck, 2024-11-18:
### PROBLEM DESCRIPTION
`SwitchMode 5` doesn't work if device was left untouched for ~5 seconds.

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [X] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [X] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [X] Searched the problem in [discussions](https://github.com/arendst/Tasmota/discussions)
- [X] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [X] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [X] Device used (e.g., Sonoff Basic):Shelly Plus 2PM
- [X] Tasmota binary firmware version number used: 14.3.0
  - [X] Pre-compiled
  - [ ] Self-compiled
- [X] Flashing tools used: WebGUI
- [X] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```lua
  Configuration output here:
21:17:05.723 MQT: stat/tas-room-martin/RESULT = {"NAME":"Shelly Plus 2PM","GPIO":[320,0,0,0,34,192,0,0,225,224,0,0,0,0,193,0,0,0,0,0,0,608,640,3458,0,0,0,0,0,9472,0,4736,0,0,0,0],"FLAG":0,"BASE":1}
21:17:05.954 MQT: stat/tas-room-martin/RESULT = {"Module":{"0":"Shelly Plus 2PM"}}
21:17:06.165 MQT: stat/tas-room-martin/RESULT = {"GPIO0":{"320":"Led_i1"},"GPIO1":{"0":"None"},"GPIO2":{"0":"None"},"GPIO3":{"0":"None"},"GPIO4":{"34":"Button3"},"GPIO5":{"192":"Switch_n1"},"GPIO6":{"0":"None"},"GPIO7":{"0":"None"},"GPIO8":{"0":"None"},"GPIO9":{"0":"None"},"GPIO10":{"0":"None"},"GPIO11":{"0":"None"},"GPIO12":{"225":"Relay2"},"GPIO13":{"224":"Relay1"},"GPIO14":{"0":"None"},"GPIO15":{"0":"None"},"GPIO16":{"0":"None"},"GPIO17":{"0":"None"},"GPIO18":{"193":"Switch_n2"},"GPIO19":{"0":"None"},"GPIO20":{"0":"None"},"GPIO21":{"0":"None"},"GPIO22":{"0":"None"},"GPIO23":{"0":"None"},"GPIO24":{"0":"None"},"GPIO25":{"608":"I2C SCL1"},"GPIO26":{"640":"I2C SDA1"},"GPIO27":{"3458":"ADE7953 IRQ3"},"GPIO32":{"0":"None"},"GPIO33":{"9472":"ADE7953 RST"},"GPIO34":{"0":"None"},"GPIO35":{"4736":"ADC Temp1"},"GPIO36":{"0":"None"},"GPIO37":{"0":"None"},"GPIO38":{"0":"None"},"GPIO39":{"0":"None"}}
```
- [ ] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```lua
  Rules output here:

```
- [X] Provide the output of this command: `Status 0`:
```lua
  STATUS 0 output here:
21:17:58.793 MQT: stat/tas-room-martin/STATUS = {"Status":{"Module":0,"DeviceName":"tas-room-martin","FriendlyName":["Martin Deckenlicht","Martin Steckdosen"],"Topic":"tas-room-martin","ButtonTopic":"0","Power":"01","PowerLock":"00","PowerOnState":3,"LedState":1,"LedMask":"FFFF","SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[5,5,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0,"InfoRetain":0,"StateRetain":0,"StatusRetain":0}}
21:17:58.799 MQT: stat/tas-room-martin/STATUS1 = {"StatusPRM":{"Baudrate":115200,"SerialConfig":"8N1","GroupTopic":"tasmotas","OtaUrl":"http://ota.tasmota.com/tasmota32/release/tasmota32solo1.bin","RestartReason":"Vbat power on reset","Uptime":"0T09:08:34","StartupUTC":"2024-11-18T11:09:24","Sleep":50,"CfgHolder":4617,"BootCount":18,"BCResetTime":"2024-10-16T09:56:53","SaveCount":247}}
21:17:58.805 MQT: stat/tas-room-martin/STATUS2 = {"StatusFWR":{"Version":"14.3.0(release-tasmota32)","BuildDateTime":"2024-10-15T08:21:27","Core":"3_1_0","SDK":"5.3.1.240924","CpuFrequency":160,"Hardware":"ESP32-U4WDH-D v3.1","CR":"451/699"}}
21:17:58.810 MQT: stat/tas-room-martin/STATUS3 = {"StatusLOG":{"SerialLog":2,"WebLog":2,"MqttLog":0,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["whakaputa",""],"TelePeriod":10,"Resolution":"558180C0","SetOption":["00008009","0F05C80001000600003C5A0A192800000000","00000080","00006000","00004000","00000000"]}}
21:17:58.833 MQT: stat/tas-room-martin/STATUS4 = {"StatusMEM":{"ProgramSize":1997,"Free":882,"Heap":130,"StackLowMark":3,"PsrMax":0,"PsrFree":0,"ProgramFlashSize":4096,"FlashSize":4096,"FlashChipId":"164020","FlashFrequency":40,"FlashMode":"DIO","Features":["0809","9F9AD7DF","0015A001","B7F7BFCF","05DA9BC4","E0360DC7","480840D2","20200000","D4BC482D","810A80B1","00000014"],"Drivers":"1,2,3,!4,!5,7,!8,9,10,11,12,!14,!16,!17,!20,!21,!24,26,!27,29,!34,!35,38,50,52,!59,!60,62,!63,!66,!67,!68,!73,82,!86,!87,!88,121","Sensors":"1,2,3,5,6,7,8,9,10,11,12,13,14,15,17,18,19,20,21,22,26,31,34,37,39,40,42,43,45,51,52,55,56,58,59,64,66,67,74,85,92,95,98,103,105,109,127","I2CDriver":"7,8,9,10,11,12,13,14,15,17,18,20,24,29,31,36,41,42,44,46,48,58,62,65,69,76,77,82,89"}}
21:17:58.842 MQT: stat/tas-room-martin/STATUS5 = {"StatusNET":{"Hostname":"tas-room-martin-7304","IPAddress":"10.39.230.13","Gateway":"10.39.230.1","Subnetmask":"255.255.255.0","DNSServer1":"10.39.230.1","DNSServer2":"0.0.0.0","Mac":"FC:B4:67:0E:7C:88","IP6Global":"","IP6Local":"fe80::feb4:67ff:fe0e:7c88%st14","Ethernet":{"Hostname":"","IPAddress":"0.0.0.0","Gateway":"0.0.0.0","Subnetmask":"0.0.0.0","DNSServer1":"10.39.230.1","DNSServer2":"0.0.0.0","Mac":"00:00:00:00:00:00","IP6Global":"","IP6Local":""},"Webserver":2,"HTTP_API":1,"WifiConfig":4,"WifiPower":16.0}}
21:17:58.849 MQT: stat/tas-room-martin/STATUS6 = {"StatusMQT":{"MqttHost":"mqtt.smarthome.gern","MqttPort":1883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_0E7C88","MqttUser":"tas-room-martin","MqttCount":5,"MAX_PACKET_SIZE":1200,"KEEPALIVE":30,"SOCKET_TIMEOUT":4}}
21:17:58.884 MQT: stat/tas-room-martin/STATUS7 = {"StatusTIM":{"UTC":"2024-11-18T20:17:58Z","Local":"2024-11-18T21:17:58","StartDST":"2024-03-31T02:00:00","EndDST":"2024-10-27T03:00:00","Timezone":"+01:00","Sunrise":"08:04","Sunset":"17:05"}}
21:17:58.898 MQT: stat/tas-room-martin/STATUS9 = {"StatusPTH":{"PowerDelta":[0,0],"PowerLow":[0,0],"PowerHigh":[0,0],"VoltageLow":[0,0],"VoltageHigh":[0,0],"CurrentLow":[0,0],"CurrentHigh":[0,0],"MaxPower":[0,0],"MaxPowerHold":[10,10],"MaxPowerWindow":[30,30],"MaxEnergy":[0,0],"MaxEnergyStart":[0,0]}}
21:17:58.917 MQT: stat/tas-room-martin/STATUS10 = {"StatusSNS":{"Time":"2024-11-18T21:17:58","Switch1":"OFF","Switch2":"OFF","ANALOG":{"Temperature1":46.5},"ENERGY":{"TotalStartTime":"2024-10-16T09:56:53","Total":0.382,"Yesterday":0.000,"Today":0.382,"Power":[49,0],"ApparentPower":[79,0],"ReactivePower":[20,0],"Factor":[0.62,0.00],"Frequency":50,"Voltage":236,"Current":[0.203,0.000]},"TempUnit":"C"}}
21:17:58.930 MQT: stat/tas-room-martin/STATUS11 = {"StatusSTS":{"Time":"2024-11-18T21:17:58","Uptime":"0T09:08:34","UptimeSec":32914,"Heap":132,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":20,"MqttCount":5,"Berry":{"HeapUsed":4,"Objects":46},"POWER1":"ON","POWER2":"OFF","Wifi":{"AP":1,"SSId":"whakaputa","BSSId":"6A:AF:97:72:B1:BC","Channel":1,"Mode":"HT40","RSSI":94,"Signal":-53,"LinkCount":9,"Downtime":"0T00:02:40"}}}

```
- [ ] Set `weblog` to 4 and then, when you experience your issue, provide the output of the Console log:
```lua
  Console output here:

21:21:00.919 BRY: GC from 7468 to 4159 bytes, objects freed 12/46 (in 0 ms) - slots from 49/91 to 49/91
21:21:00.931 MQT: tele/tas-room-martin/STATE = {"Time":"2024-11-18T21:21:00","Uptime":"0T09:11:36","UptimeSec":33096,"Heap":136,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":5,"Berry":{"HeapUsed":4,"Objects":46},"POWER1":"ON","POWER2":"OFF","Wifi":{"AP":1,"SSId":"whakaputa","BSSId":"6A:AF:97:72:B1:BC","Channel":1,"Mode":"HT40","RSSI":100,"Signal":-50,"LinkCount":9,"Downtime":"0T00:02:40"}}
21:21:00.947 MQT: tele/tas-room-martin/SENSOR = {"Time":"2024-11-18T21:21:00","Switch1":"OFF","Switch2":"OFF","ANALOG":{"Temperature1":44.9},"ENERGY":{"TotalStartTime":"2024-10-16T09:56:53","Total":0.384,"Yesterday":0.000,"Today":0.384,"Period":[0,0],"Power":[49,0],"ApparentPower":[79,0],"ReactivePower":[20,0],"Factor":[0.62,0.00],"Frequency":50,"Voltage":236,"Current":[0.202,0.000]},"TempUnit":"C"}
21:21:01.250 ADE: Channel1 ACCMODE 0x280C00, VRMS 6126417, Period 4475, IRMS 22325, WATT -215, VA 348, VAR 86
21:21:01.252 ADE: Channel2 ACCMODE 0x280C00, VRMS 6127650, Period 4472, IRMS 1782, WATT 0, VA 28, VAR 0
21:21:02.250 ADE: Channel1 ACCMODE 0x280C00, VRMS 6124055, Period 4474, IRMS 22258, WATT -214, VA 347, VAR 87
21:21:02.251 ADE: Channel2 ACCMODE 0x280C00, VRMS 6124716, Period 4472, IRMS 1782, WATT 0, VA 28, VAR 0
21:21:02.964 SRC: Switch
21:21:02.976 MQT: stat/tas-room-martin/RESULT = {"POWER2":"ON"}
[... cut ...]
### EXPECTED BEHAVIOUR
I expect the trigger to work on first press, even after waiting 5 seconds.

### SCREENSHOTS
[image]

Observe GPIO18, the input button going to high, but nothing happens. Only on second press within 5 seconds does GPIO12 go to high. Then I wait a few seconds and press the button again. Observe GPIO18 going high, but GPIO12 only goes low on second press.

### ADDITIONAL CONTEXT
I have this issue consistently with all Shelly Plus 2PM running the same current Tasmota firmware.

sfromis, 2024-11-18:
As detailed on Discord, don't assume the funky behavior to be Tasmota, when the device is Shelly Plus 2PM, which seems to not behave straight for the mains AC switching inputs.

madduck, 2024-11-18:
> As detailed on Discord, don't assume the funky behavior to be Tasmota, when the device is Shelly Plus 2PM, which seems to not behave straight for the mains AC switching inputs.

SwitchModes 3, 4, and 6 work just fine, which leads me to believe that the difference is in software, because — and this is also shown in the GPIO diagram — the hardware seems to work fine.

madduck, 2024-11-18:
> As already suggested, if you want to demonstrate odd behavior, do it on a device with "normal" inputs.

I am not a hardware person, but I still fail to see how this is a hardware issue when the GPIOs work as expected even on AC, and only one of the four very similar SwitchMode, which are a software construct (right?), is affected.

barbudor, 2024-11-18:
The difference between switchmode 5 and 6 is exactly about hardware
One is active low while he other is active high
By using the wrong mode, it won't work

You are exactly in the case where your swich inactive position is understood as active by mode 5
So the 5 seconds you see are actually the 4 seconds of the hold delay

madduck, 2024-11-18:
> You are exactly in the case where your swich inactive position is understood as active by mode 5 So the 5 seconds you see are actually the 4 seconds of the hold delay

Thank you @barbudor, this noww makes perfect sense and I can close the issue. Sorry for having wasted everyone's time. `SwitchMode 5` is designed for the case when the input GPIO is high by default, and lowering it means `TOGGLE`. I got confused by all the 1's and 0's and ON's and OFF's and rising and falling edges.

closed 2024-11-18 by madduck