arendst/Tasmota #22535: Sonoff Basic R4 Magicswitch bounces when toggled by command

crhass, 2024-11-23:
### PROBLEM DESCRIPTION
_A clear and concise description of what the problem is._
I have noticed that when toggling the sonoff relay from the web interface or MQTT command, the relay will sometimes bounce back to its previous state. MAGICSWITCH_MASKING_WINDOW_LEN appears not to be active for switching by command but may solve the problem.

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [ ] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [ ] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [ ] Searched the problem in [discussions](https://github.com/arendst/Tasmota/discussions)
- [ ] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [ ] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [ ] Device used (e.g., Sonoff Basic): _____
- [ ] Tasmota binary firmware version number used: _____
  - [ ] Pre-compiled
  - [ ] Self-compiled
- [ ] Flashing tools used: _____
- [ ] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```lua
  Configuration output here:

```
- [ ] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```lua
  Rules output here:

```
- [ ] Provide the output of this command: `Status 0`:
```lua
  STATUS 0 output here:

```
- [ ] Set `weblog` to 4 and then, when you experience your issue, provide the output of the Console log:
```lua
  Console output here:

```

### TO REPRODUCE
_Steps to reproduce the behavior:_
Apply the standard magicswitch template
Set MAGICSWITCH_MASKING_WINDOW_LEN to 20
The relay does not bounce when using the magicswitch but does bounce when switched from the web interface or MQTT.

### EXPECTED BEHAVIOUR
_A clear and concise description of what you expected to happen._
Expect the relay to change state once and not trigger the magicswitch feature.

### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._

### ADDITIONAL CONTEXT
_Add any other context about the problem here._
I had to increase the MAGICSWITCH_MASKING_WINDOW_LEN from the default 5 (250ms) to 20 (1 sec) to prevent the bouncing when using the physical switch but I still get the issue when switching by software. Should the masking be applied to all switching?

**(Please, remember to close the issue when the problem has been addressed)**

barbudor, 2024-11-23:
I will try to reproduce but I believe your problem is hardware related as the magic switch features detects a power loss while you are not actuating the switch.

What is your load? Passive or inductive? What power? 
I fear that the load switch creates a disturbance on the mains that the Basic sees as a small power loss.

crhass, 2024-11-23:
Load is 4x approx 5w GU10 LED lamps.

I am seeing the magic switch detect pulses of around 5000 to 50000 when switching the relay with MQTT. 

It is not every time, probably about 10% erroneous detection.

I suspecting it could be the load characteristics too but would be nice to be able to filter it. 

It is not general power noise as I have no phantom switching other to than this

I tried switchdebounce 1000 which I suspect may have reduced the frequency of this issue. 

Do you know if Switchdebounce 69 works with the basic R4

I am going to try a rule to create a longer delay, maybe 2 sec

I would like to find a software solution as I am sure others could have similar loads.

Thanks for responding

barbudor, 2024-11-23:
Switchdebounce will have no impact on the MagicSwitch which use a logic totally indépendant from Switches

I don't see how you could workaround that with rules.

I think the only solution is indeed to apply the masking window for any power switch whatever is the source
I'll try to find time for that during the week end

I understand you know how to compile your own so I'll provide you patches or a branch to test

crhass, 2024-11-23:
Thank you for your help, I will be sure to test this on my setup and feedback to you if it is successful.

crhass, 2024-11-23:
I can confirm I am also seeing this with different GU10 lamps, with a single lamp and also with no load so I am ruling out the load now.

barbudor, 2024-11-23:
I'm fighting with my build env
I can't test but this should be ok
Can you please try to apply the changes listed there and report ?
https://github.com/arendst/Tasmota/pull/22539/files

crhass, 2024-11-23:
Just compiled and uploaded and pleased to confirm it works and the issue is resolved. Really appreciate you working on this so quickly. Many Thanks again.

Off to turn my lighting automations back on in home assistant now.

PR #22539 (2024-11-25): Issue#22535 applying masking window to any power change
## Description:

**Related issue (if applicable):** fixes #22535 

Apply a masking window to the MagicSwitch input in case of a power state change whatever is the source.
Should avoid load switch to create disturbances that is falsy understood as a MagicSwitch pulse

tested by OP #crhass

## Checklist:
  - [X] The pull request is done against the latest development branch
  - [X] Only relevant files were touched
  - [X] Only one feature/fix was added per PR and the code change compiles without warnings
  - [-] The code change is tested an [... cut]