arendst/Tasmota #22559: Unknown command failure with RfSend on sonoff rf bridge v2.2

rhabacker, 2024-11-29:
### PROBLEM DESCRIPTION
With the latest tasmota development version (version 14.3.0.7 without the portisch firmware) on a sonoff rf bridge v2.2, I can see the `RfReceived` messages when I press a button on an rf switch, but I cannot use the `RfSend` command with the appropriate parameters.

According to https://tasmota.github.io/docs/Commands/#rf-transceiver I have tried the examples mentioned there, but they fail (see below)

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [x] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [x] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [x] Searched the problem in [discussions](https://github.com/arendst/Tasmota/discussions)
- [x] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [x] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [x] Device used (e.g., Sonoff Basic): Sonoff Rf Brigde
- [x] Tasmota binary firmware version number used: 14.3.0.7 without the portisch firmware
  - [x] Pre-compiled
  - [ ] Self-compiled
- [x] Flashing tools used: ota
- [x] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```lua
  Configuration output here:
22:53:48.540 MQT: stat/tasmota_sonoff/RESULT = {"GPIO0":{"32":"Button1"},"GPIO1":{"3200":"Serial Tx"},"GPIO2":{"0":"None"},"GPIO3":{"3232":"Serial Rx"},"GPIO4":{"0":"None"},"GPIO5":{"0":"None"},"GPIO9":{"0":"None"},"GPIO10":{"0":"None"},"GPIO12":{"0":"None"},"GPIO13":{"320":"Led_i1"},"GPIO14":{"0":"None"},"GPIO15":{"0":"None"},"GPIO16":{"0":"None"},"GPIO17":{"0":"None"}}
```
- [x] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```lua
22:54:23.371 MQT: stat/tasmota_sonoff/RESULT = {"Rule1":{"State":"OFF","Once":"OFF","StopOnError":"OFF","Length":0,"Free":511,"Rules":""}}
22:54:23.597 MQT: stat/tasmota_sonoff/RESULT = {"Rule2":{"State":"OFF","Once":"OFF","StopOnError":"OFF","Length":0,"Free":511,"Rules":""}}
22:54:23.847 MQT: stat/tasmota_sonoff/RESULT = {"Rule3":{"State":"OFF","Once":"OFF","StopOnError":"OFF","Length":0,"Free":511,"Rules":""}}
```
- [x] Provide the output of this command: `Status 0`:
```lua
  STATUS 0 output here:
22:56:29.089 MQT: stat/tasmota_sonoff/STATUS = {"Status":{"Module":25,"DeviceName":"Tasmota","FriendlyName":["Tasmota"],"Topic":"tasmota_sonoff","ButtonTopic":"0","Power":"0","PowerLock":"0","PowerOnState":3,"LedState":1,"LedMask":"FFFF","SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0,"InfoRetain":0,"StateRetain":0,"StatusRetain":0}}
22:56:29.097 MQT: stat/tasmota_sonoff/STATUS1 = {"StatusPRM":{"Baudrate":19200,"SerialConfig":"8N1","GroupTopic":"tasmotas","OtaUrl":"http://ota.tasmota.com/tasmota/tasmota.bin.gz","RestartReason":"Software/System restart","Uptime":"0T02:48:56","StartupUTC":"2024-11-29T19:07:33","Sleep":50,"CfgHolder":4617,"BootCount":66,"BCResetTime":"2024-04-24T18:06:21","SaveCount":130,"SaveAddress":"F6000"}}
22:56:29.104 MQT: stat/tasmota_sonoff/STATUS2 = {"StatusFWR":{"Version":"14.3.0.7(038e62b-tasmota)","BuildDateTime":"2024-11-29T16:01:51","Boot":31,"Core":"2_7_8","SDK":"2.2.2-dev(38a443e)","CpuFrequency":80,"Hardware":"ESP8285N08","CR":"380/699"}}
22:56:29.111 MQT: stat/tasmota_sonoff/STATUS3 = {"StatusLOG":{"SerialLog":0,"WebLog":3,"MqttLog":2,"SysLog":1,"LogHost":"","LogPort":514,"SSId":["xxx",""],"TelePeriod":300,"Resolution":"558180C0","SetOption":["00008009","2805C80001000600003C5A0A192800000000","00000080","00006000","00004000","00000000"]}}
22:56:29.125 MQT: stat/tasmota_sonoff/STATUS4 = {"StatusMEM":{"ProgramSize":652,"Free":348,"Heap":22,"ProgramFlashSize":1024,"FlashSize":1024,"FlashChipId":"14325E","FlashFrequency":40,"FlashMode":"DOUT","Features":["0809","8F9AC787","04368001","000000CF","010013C0","C000F981","00004004","00001000","54000020","00000080","00000000"],"Drivers":"1,2,!3,!4,!5,6,7,!8,9,10,12,!16,!18,!19,!20,!21,!22,!24,26,!27,29,!30,!35,!37,!45,62,!68","Sensors":"1,2,3,4,5,6","I2CDriver":"7"}}
22:56:29.133 MQT: stat/tasmota_sonoff/STATUS5 = {"StatusNET":{"Hostname":"tasmota-sonoff-2476","IPAddress":"192.168.x.x","Gateway":"192.168.x.x","Subnetmask":"255.255.255.0","DNSServer1":"192.168.x.x","DNSServer2":"0.0.0.0","Mac":"xx","Webserver":2,"HTTP_API":1,"WifiConfig":4,"WifiPower":17.0}}
22:56:29.140 MQT: stat/tasmota_sonoff/STATUS6 = {"StatusMQT":{"MqttHost":"192.168.x.x","MqttPort":xxx,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_C429AC","MqttUser":"xxx","MqttCount":1,"MAX_PACKET_SIZE":1200,"KEEPALIVE":30,"SOCKET_TIMEOUT":4}}
22:56:29.148 MQT: stat/tasmota_sonoff/STATUS7 = {"StatusTIM":{"UTC":"2024-11-29T21:56:29Z","Local":"2024-11-29T22:56:29","StartDST":"2024-03-31T02:00:00","EndDST":"2024-10-27T03:00:00","Timezone":"+01:00","Sunrise":"08:20","Sunset":"16:56"}}
22:56:29.155 MQT: stat/tasmota_sonoff/STATUS10 = {"StatusSNS":{"Time":"2024-11-29T22:56:29"}}
22:56:29.165 MQT: stat/tasmota_sonoff/STATUS11 = {"StatusSTS":{"Time":"2024-11-29T22:56:29","Uptime":"0T02:48:56","UptimeSec":10136,"Heap":22,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":1,"Wifi":{"AP":1,"SSId":"xxx","BSSId":"xxx","Channel":1,"Mode":"11n","RSSI":88,"Signal":-56,"LinkCount":1,"Downtime":"0T00:00:04"}}}

```
- [x] Set `weblog` to 4 and then, when you experience your issue, provide the output of the Console log:
```lua
  Console output here:
23:00:03.769 CMD: RfSend 0x7028DC, 24, 1, 10, 238
23:00:03.771 SRC: WebConsole from 192.168.1.99
23:00:03.774 CMD: Grp 0, Cmd 'RFSEND', Idx 1, Len 24, Pld 7350492, Data '0x7028DC, 24, 1, 10, 238'
23:00:03.781 MQT: stat/tasmota_sonoff/RESULT = {"Command":"Unknown","Input":"RFSEND 0x7028DC, 24, 1, 10, 238"}

23:00:40.170 CMD: RfSend {"Data":"0x7028DC","Bits":24,"Protocol":1,"Pulse":238}
23:00:40.172 SRC: WebConsole from 192.168.1.99
23:00:40.175 CMD: Grp 0, Cmd 'RFSEND', Idx 1, Len 54, Pld -99, Data '{"Data":"0x7028DC","Bits":24,"Protocol":1,"Pulse":238}'
23:00:40.182 MQT: stat/tasmota_sonoff/RESULT = {"Command":"Unknown","Input":"RFSEND {\"Data\":\"0x7028DC\",\"Bits\":24,\"Protocol\":1,\"Puls"}
```

### TO REPRODUCE
1. Open the console
2. enter `RfSend 0x7028DC, 24, 1, 10, 238` and press return
3. or enter `RfSend {"Data":"0x7028DC","Bits":24,"Protocol":1,"Pulse":238}` 

### EXPECTED BEHAVIOUR
The command `RfSend` should be available and working.

**(Please, remember to close the issue when the problem has been addressed)**

sfromis, 2024-11-30:
No issue here. The `RfSend` command is **only** when you have connected a RF transceiver chip *directly* to the ESP8266. You are instead using RF Bridge, which works differently, and the `RfSend` command is therefore invalid.
https://tasmota.github.io/docs/Commands/#rf-bridge

rhabacker, 2024-11-30:
Thank you for this clarifying information.

I then tried to save a recognized code with `RfKey1 2` and then call it, but this does not work. The device beeps and the red light flashes, but does not respond to the key press that is displayed before the learning process begins. After a while, the learning process is aborted.