arendst/Tasmota #22566: WebButton label does not support <br> (html newline) anymore

coolman111, 2024-11-30:
Tasmota Release: 14.3.0 (newest)
Hardware: D1Mini (ESP8266EX)

Problem: 
The Command WebButton<x> does not support HTML Break <br> anymore. In Tasmota releases before 14 the <br> forced a new line in the label of the button.
Now the <br> is displayed as normal text and no newline is performed.
I also tried </br>, \n, but it does not work.

How can we force a newline in the label in tasmota 14.3.0 ?

sfromis, 2024-11-30:
Well, it does no longer have the vulnerability of letting arbitrary html pass. The text is safely escaped when used in the web page. I'm not aware of holes in this injection prevention.

Jason2866, 2024-12-01:
Moving to Discussions since the new behaviour is not an fault.