arendst/Tasmota #22569: user_config_override.h can't override the config set in tasmota_configurations.h

Frogmore42, 2024-12-01:
### PROBLEM DESCRIPTION
_A clear and concise description of what the problem is._
As an example: I want to disable (`undef`) `USE_MPU_ACCEL`
This is defined in tasmota\include\tasmota_configurations_ESP32.h, which is included by tasmota\include\tasmota_configurations.h, which is included by tasmota_options.h, but only after tasmota/my_user_config.h which is what includes tasmota\user_config_override.h.

This means if I want to turn off the default implementation of xsns_85_mpu6886.ino I HAVE to modify a low level header file.

It seems like it should be possible to fix this by reversing the order of the includes in tasmoata_globals.h so my_user_config.h comes last and therefore wins.

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [ x] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [ x] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [x] Searched the problem in [discussions](https://github.com/arendst/Tasmota/discussions)
- [x ] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [ ] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [x ] Device used (e.g., Sonoff Basic): _____m5 Atom Matrix (esp32)
- [x ] Tasmota binary firmware version number used: _____14.3.0.7
  - [ ] Pre-compiled
  - [ x] Self-compiled
- [ x] Flashing tools used: _____
- [ ] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```lua
  Configuration output here:

```
- [ ] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```lua
  Rules output here:

```
- [ ] Provide the output of this command: `Status 0`:
```lua
  STATUS 0 output here:

```
- [ ] Set `weblog` to 4 and then, when you experience your issue, provide the output of the Console log:
```lua
  Console output here:

```

### TO REPRODUCE
_Steps to reproduce the behavior:_

### EXPECTED BEHAVIOUR
_A clear and concise description of what you expected to happen._
I can choose to create a binary that does not include the MPU code just by editing user_config_override.h

It looks like it might not be as simple as just reversing the order of the includes in tasmota_options.h (since when I did that I got a linker error for something unrelated in berry).

### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._

### ADDITIONAL CONTEXT
_Add any other context about the problem here._

**(Please, remember to close the issue when the problem has been addressed)**

Jason2866, 2024-12-01:
You need to define your own firmware setup. See https://tasmota.github.io/docs/Compile-your-build/#defining-multiple-custom-firmwares

Frogmore42, 2024-12-01:
@Jason2866 thanks for the tip on multiple custom firmware, but it too has the same issue.

Everything in user_config_override.h gets processed before everything in tasmota_configurations*.h, which means it is impossible to actually override what they (tasmota_configurations*.h) do.  You  can add things using user_config_override.h, but you can't remove something that tasmota_configurations*.h add, since they are processed later.

So it is fine to say won't fix for this issue.  I can think of several ways that I can get whatever behavior I want.  Since this is open source, I can change whatever I want to get it to do what I want.  I filed this issue, since I figured the whole point of user_config_override.h was to make it so I did NOT have to change the underlying code.  I also recall Sonoff-Tasmota from the days before user_config_override.h when my_user_config.h was the way to do it (of course, it still has the same problem).  Tasmota has gotten a lot more complex (because it does a LOT more than it used to) and the configuration is much more complicated.  As long as it generally works, it probably doesn't matter, but it is a lot harder to understand.

sfromis, 2024-12-01:
What you have in `tasmota_configurations_ESP32.h` does not have to be used at all, unless you build for one of the precomposed build environments with `#define` symbols for `FIRMWARE_xxxxx`. You can create a custom build environment without any of those, but then you have to add features you specifically want to have included.