arendst/Tasmota #22579: Active Power Total sometimes disappears in Shelly 3EM

jeroenst, 2024-12-02:
### PROBLEM DESCRIPTION
_A clear and concise description of what the problem is._
The Active Power Total field is dissapearing on the main webpage from time to time.

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [x] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [x] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [x] Searched the problem in [discussions](https://github.com/arendst/Tasmota/discussions)
- [x] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [x] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [x] Device used (e.g., Sonoff Basic): Shelly 3EM
- [x] Tasmota binary firmware version number used: 14.3.0.7 development commitid: 341cc87527be5e931bd69ce9720e9f3712302fa6 
  - [ ] Pre-compiled
  - [x] Self-compiled
- [x] Flashing tools used: webpage
- [x] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```lua
15:51:16.484 MQT: home/TASMOTA-SHELLY3EM-01/stat/RESULT = {"NAME":"Shelly 3EM","GPIO":[1,1,288,1,32,8065,0,0,640,8064,608,224,8096,0],"FLAG":0,"BASE":18}
15:51:16.702 MQT: home/TASMOTA-SHELLY3EM-01/stat/RESULT = {"Module":{"0":"Shelly 3EM"}}
15:51:16.963 MQT: home/TASMOTA-SHELLY3EM-01/stat/RESULT = {"GPIO0":{"0":"None"},"GPIO1":{"0":"None"},"GPIO2":{"288":"Led1"},"GPIO3":{"0":"None"},"GPIO4":{"32":"Button1"},"GPIO5":{"8065":"ADE7880 IRQ2"},"GPIO9":{"0":"None"},"GPIO10":{"0":"None"},"GPIO12":{"640":"I2C SDA"},"GPIO13":{"8064":"ADE7880 IRQ1"},"GPIO14":{"608":"I2C SCL"},"GPIO15":{"224":"Relay1"},"GPIO16":{"8096":"Reset"},"GPIO17":{"0":"None"}}

```
- [x] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```lua
15:51:58.537 MQT: home/TASMOTA-SHELLY3EM-01/stat/RESULT = {"Rule1":{"State":"OFF","Once":"OFF","StopOnError":"OFF","Length":0,"Free":511,"Rules":""}}
15:51:58.772 MQT: home/TASMOTA-SHELLY3EM-01/stat/RESULT = {"Rule2":{"State":"OFF","Once":"OFF","StopOnError":"OFF","Length":0,"Free":511,"Rules":""}}
15:51:59.026 MQT: home/TASMOTA-SHELLY3EM-01/stat/RESULT = {"Rule3":{"State":"ON","Once":"OFF","StopOnError":"OFF","Length":467,"Free":44,"Rules":"on file#calib.dat do {\"state\":0,\"rms\":{\"current_a\":3202113,\"current_b\":3203075,\"current_c\":3216489,\"current_n\":55261950,\"current_s\":1325446,\"voltage_a\":-734868,\"voltage_b\":-706252,\"voltage_c\":-725460},\"angles\":{\"angle0\":160,\"angle1\":152,\"angle2\":144},\"powers\":{\"totactive\":{\"a\":-1399538,\"b\":-1395614,\"c\":-1392329},\"apparent\":{\"a\":630401,\"b\":630366,\"c\":630398}},\"energies\":{\"totactive\":{\"a\":25697,\"b\":25697,\"c\":25696},\"apparent\":{\"a\":64209,\"b\":64256,\"c\":64291}}} endon"}}

```
- [x] Provide the output of this command: `Status 0`:
```lua
15:52:32.610 MQT: home/TASMOTA-SHELLY3EM-01/stat/STATUS0 = {"Status":{"Module":0,"DeviceName":"Tasmota","FriendlyName":["Tasmota"],"Topic":"home/TASMOTA-SHELLY3EM-01","ButtonTopic":"0","Power":"0","PowerLock":"0","PowerOnState":3,"LedState":1,"LedMask":"FFFF","SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0],"ButtonRetain":1,"SwitchRetain":1,"SensorRetain":1,"PowerRetain":1,"InfoRetain":1,"StateRetain":1,"StatusRetain":1},"StatusPRM":{"Baudrate":115200,"SerialConfig":"8N1","GroupTopic":"tasmotas","OtaUrl":"http://ota.tasmota.com/tasmota/release/tasmota-minimal.bin.gz","RestartReason":"Software/System restart","Uptime":"0T00:31:49","StartupUTC":"2024-12-02T14:20:43","Sleep":50,"CfgHolder":4617,"BootCount":21,"BCResetTime":"2024-11-30T11:22:30","SaveCount":91,"SaveAddress":"F9000"},"StatusFWR":{"Version":"14.3.0.7(tasmota)","BuildDateTime":"2024-12-02T15:14:34","Boot":31,"Core":"2_7_8","SDK":"2.2.2-dev(38a443e)","CpuFrequency":80,"Hardware":"ESP8266EX","CR":"384/699"},"StatusLOG":{"SerialLog":2,"WebLog":2,"MqttLog":0,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["J&M IOT",""],"TelePeriod":60,"Resolution":"558980C0","SetOption":["002082E9","2805C80001000600003C5A0A192800000000","004000E8","00206000","8004C180","00000000"]},"StatusMEM":{"ProgramSize":712,"Free":288,"Heap":16,"ProgramFlashSize":1024,"FlashSize":4096,"FlashChipId":"16301C","FlashFrequency":40,"FlashMode":"DOUT","Features":["0809","8F9AC7C7","04368001","000000CF","010013C0","C000F981","00004004","00001000","54800020","00000080","00000000"],"Drivers":"1,2,3,!4,!5,!6,7,!8,9,10,12,!16,!18,!19,!20,!21,!22,!24,26,!27,29,!30,!35,!37,!45,62,!68","Sensors":"1,2,3,4,5,6","I2CDriver":"7,65"},"StatusNET":{"Hostname":"TASMOTA-SHELLY3EM-01","IPAddress":"172.30.31.15","Gateway":"172.30.30.254","Subnetmask":"255.255.254.0","DNSServer1":"172.30.30.254","DNSServer2":"0.0.0.0","Mac":"34:94:54:75:F0:F5","Webserver":2,"HTTP_API":1,"WifiConfig":4,"WifiPower":17.0},"StatusMQT":{"MqttHost":"mqtt.jst-it.nl","MqttPort":8883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_75F0F5","MqttUser":"jeroen","MqttCount":1,"MAX_PACKET_SIZE":1200,"KEEPALIVE":30,"SOCKET_TIMEOUT":4},"StatusTIM":{"UTC":"2024-12-02T14:52:32Z","Local":"2024-12-02T15:52:32","StartDST":"2024-03-31T02:00:00","EndDST":"2024-10-27T03:00:00","Timezone":99,"Sunrise":"08:24","Sunset":"16:54"},"StatusPTH":{"PowerDelta":[1,0,0],"PowerLow":0,"PowerHigh":0,"VoltageLow":0,"VoltageHigh":0,"CurrentLow":0,"CurrentHigh":0,"MaxPower":0,"MaxPowerHold":10,"MaxPowerWindow":30,"MaxEnergy":0,"MaxEnergyStart":0},"StatusSNS":{"Time":"2024-12-02T15:52:32","ENERGY":{"TotalStartTime":"2024-11-30T13:18:05","Total":[9.584,22.727,0.005],"Yesterday":[1.513,10.905,-0.002],"Today":[5.281,7.897,0.005],"TodaySumImport":0.155,"TodaySumExport":0.002,"ExportActive":[8.196,0.000,0.012],"Power":[165.3,229.6,-0.1],"ApparentPower":[548.5,423.7,3.4],"ReactivePower":[523.0,356.1,3.4],"Factor":[0.30,0.54,-0.02],"Frequency":[50,50,50],"Voltage":[249,249,249],"Current":[2.217,1.712,0.014],"CurrentNeutral":0.011}},"StatusSTS":{"Time":"2024-12-02T15:52:32","Uptime":"0T00:31:49","UptimeSec":1909,"Heap":14,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":20,"MqttCount":1,"POWER":"OFF","Wifi":{"AP":1,"SSId":"J&M IOT","BSSId":"FA:92:BF:17:DA:AE","Channel":1,"Mode":"11n","RSSI":100,"Signal":-42,"LinkCount":1,"Downtime":"0T00:00:07"}}} (retained)

```
- [ ] Set `weblog` to 4 and then, when you experience your issue, provide the output of the Console log:
```lua
15:53:56.312 MQT: home/TASMOTA-SHELLY3EM-01/tele/STATE = {"Time":"2024-12-02T15:53:56","Uptime":"0T00:33:13","UptimeSec":1993,"Heap":17,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":1,"POWER":"OFF","Wifi":{"AP":1,"SSId":"J&M IOT","BSSId":"FA:92:BF:17:DA:AE","Channel":1,"Mode":"11n","RSSI":100,"Signal":-42,"LinkCount":1,"Downtime":"0T00:00:07"}} (retained)
15:53:56.348 MQT: home/TASMOTA-SHELLY3EM-01/tele/SENSOR = {"Time":"2024-12-02T15:53:56","ENERGY":{"TotalStartTime":"2024-11-30T13:18:05","Total":[9.588,22.732,0.005],"Yesterday":[1.513,10.905,-0.002],"Today":[5.285,7.902,0.005],"TodaySumImport":0.165,"TodaySumExport":0.002,"ExportActive":[8.196,0.000,0.012],"Period":[2.9,3.9,0.0],"Power":[176.0,232.1,-0.1],"ApparentPower":[550.6,426.4,3.5],"ReactivePower":[521.7,357.7,3.4],"Factor":[0.32,0.54,-0.01],"Frequency":[50,50,50],"Voltage":[251,251,251],"Current":[2.211,1.711,0.014],"CurrentNeutral":0.011}} (retained)
15:53:56.388 MQT: home/TASMOTA-SHELLY3EM-01/tele/SENSOR = {"Time":"2024-12-02T15:53:56","ENERGY":{"TotalStartTime":"2024-11-30T13:18:05","Total":[9.588,22.732,0.005],"Yesterday":[1.513,10.905,-0.002],"Today":[5.285,7.902,0.005],"TodaySumImport":0.165,"TodaySumExport":0.002,"ExportActive":[8.196,0.000,0.012],"Power":[176.0,232.1,-0.1],"ApparentPower":[550.6,426.4,3.5],"ReactivePower":[521.7,357.7,3.4],"Factor":[0.32,0.54,-0.01],"Frequency":[50,50,50],"Voltage":[251,251,251],"Current":[2.211,1.711,0.014],"CurrentNeutral":0.011}} (retained)
15:54:00.306 MQT: home/TASMOTA-SHELLY3EM-01/tele/MARGINS = {"MARGINS":{"PowerDelta":[6,0,0]}} (retained)
15:54:00.339 MQT: home/TASMOTA-SHELLY3EM-01/tele/SENSOR = {"Time":"2024-12-02T15:54:00","ENERGY":{"TotalStartTime":"2024-11-30T13:18:05","Total":[9.588,22.732,0.005],"Yesterday":[1.513,10.905,-0.002],"Today":[5.285,7.902,0.005],"TodaySumImport":0.165,"TodaySumExport":0.002,"ExportActive":[8.196,0.000,0.012],"Power":[181.6,237.1,-0.0],"ApparentPower":[554.3,431.5,3.5],"ReactivePower":[523.7,360. [... cut ...]
Just flash a Shelly 3EM and watch the Active Power Total dissapear.

### EXPECTED BEHAVIOUR
_A clear and concise description of what you expected to happen._
Active Power Total is always visible

### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._

https://github.com/user-attachments/assets/a2604c1f-f4e2-4a9f-81c4-8b22ba900a70

### ADDITIONAL CONTEXT
_Add any other context about the problem here._

**(Please, remember to close the issue when the problem has been addressed)**

arendst, 2024-12-02:
As designed. As your L3 is toggling between negative (export) and positive (import) you'll see this.

See code snippit:
```
    if (abs(negative_phases) != Energy->phase_count) {  // Provide total power if producing power (PV) and multi phase
       WSContentSend_PD(HTTP_SNS_POWER_TOTAL, WebEnergyFmt(Energy->active_power, Settings->flag2.wattage_resolution, 3));
    }
```

The idea behind this is that in normal situations (no export) the backend can/will provide calculations over power. Un case of export one felt the need for an Active Power Total display showing the sum of imported and exported power.

jeroenst, 2024-12-02:
Can we maybe just show Power Total Active always?

jeroenst, 2024-12-03:
Tested and works okay! Thanx!

closed 2024-12-03 by jeroenst