arendst/Tasmota #22581: SSD1306 Display does not work anymore on Tasmota 14.3

Hennyr, 2024-12-02:
### PROBLEM DESCRIPTION
After upgrading a ESP12n_NODEMCU the display did not work any more. Have reflash new version and configurated again. It all didn't helped. after flashnig back to version 13.3 it worked again

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [X] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [X] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [X Searched the problem in [discussions](https://github.com/arendst/Tasmota/discussions)
- [X] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [ ] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [X] Device used (e.g., Sonoff Basic): _____
- [X] Tasmota binary firmware version number used: Tasmota_display 14.3 timestamp 15102024
  - [X] Pre-compiled
  - [ ] Self-compiled
- [ ] Flashing tools used: _____
- [ ] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```lua
  Configuration output here:
{"NAME":"ESP12n_NodeMCU","GPIO":[1,1,1,1,1,1,1,1,1,1,1,1,1,1],"FLAG":0,"BASE":18}

```
- [ ] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```lua
  Rules output here:
Rule1	on tele-DS18B20#Temperature do Displaytext [zs1l1c1]Temperatuur [s2l2c2]%value% ~f8C endon

```
- [ ] Provide the output of this command: `Status 0`:
```lua
  STATUS 0 output here: Version 13.3
stat/tasmota_ESP12n_temp/STATUS = {"Status":{"Module":0,"DeviceName":"Tasmota-ESP12n_NodeMCU","FriendlyName":["Tasmota-esp12n"],"Topic":"tasmota_ESP12n_temp","ButtonTopic":"0","Power":1,"PowerOnState":3,"LedState":1,"LedMask":"FFFF","SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0,"InfoRetain":0,"StateRetain":0,"StatusRetain":0}}
17:12:49.087 MQT: stat/tasmota_ESP12n_temp/STATUS1 = {"StatusPRM":{"Baudrate":115200,"SerialConfig":"8N1","GroupTopic":"tasmotas","OtaUrl":"http://ota.tasmota.com/tasmota/release/tasmota32.bin.gz","RestartReason":"External System","Uptime":"0T00:00:30","StartupUTC":"2024-12-02T16:12:19","Sleep":50,"CfgHolder":4617,"BootCount":7,"BCResetTime":"2023-02-28T12:20:39","SaveCount":26,"SaveAddress":"F8000"}}
17:12:49.125 MQT: stat/tasmota_ESP12n_temp/STATUS2 = {"StatusFWR":{"Version":"13.3.0.2(display)","BuildDateTime":"2024-01-10T08:23:06","Boot":31,"Core":"2_7_4_9","SDK":"2.2.2-dev(38a443e)","CpuFrequency":80,"Hardware":"ESP8266EX","CR":"383/699"}}
17:12:49.146 MQT: stat/tasmota_ESP12n_temp/STATUS3 = {"StatusLOG":{"SerialLog":2,"WebLog":2,"MqttLog":3,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["deco-m9-hr",""],"TelePeriod":10,"Resolution":"558180C0","SetOption":["00008009","2805C80001000600003C5A0A192800000000","00000080","00006000","00004000","00000000"]}}
17:12:49.181 MQT: stat/tasmota_ESP12n_temp/STATUS4 = {"StatusMEM":{"ProgramSize":616,"Free":384,"Heap":22,"ProgramFlashSize":1024,"FlashSize":4096,"FlashChipId":"164020","FlashFrequency":40,"FlashMode":"DOUT","Features":["0809","0FAA858E","04049FA1","000000C3","00000000","00004080","00000020","401E8000","54000000","00000080","00000000"],"Drivers":"1,2,!4,!5,!8,9,10,12,13,!16,62,!68","Sensors":"1,2,5,6","I2CDriver":"3,4,5,6,38,39,47"}}
17:12:49.222 MQT: stat/tasmota_ESP12n_temp/STATUS5 = {"StatusNET":{"Hostname":"tasmota-ESP12n-temp-7464","IPAddress":"192.168.178.175","Gateway":"192.168.178.1","Subnetmask":"255.255.255.0","DNSServer1":"192.168.178.1","DNSServer2":"0.0.0.0","Mac":"60:01:94:75:1D:28","Webserver":2,"HTTP_API":1,"WifiConfig":4,"WifiPower":17.0}}
17:12:49.251 MQT: stat/tasmota_ESP12n_temp/STATUS6 = {"StatusMQT":{"MqttHost":"192.168.178.65","MqttPort":1883,"MqttClientMask":"Esp12n","MqttClient":"Esp12n","MqttUser":"mqttHR","MqttCount":1,"MAX_PACKET_SIZE":1200,"KEEPALIVE":30,"SOCKET_TIMEOUT":4}}
17:12:49.273 MQT: stat/tasmota_ESP12n_temp/STATUS7 = {"StatusTIM":{"UTC":"2024-12-02T16:12:49Z","Local":"2024-12-02T17:12:49","StartDST":"2024-03-31T02:00:00","EndDST":"2024-10-27T03:00:00","Timezone":"+01:00","Sunrise":"08:24","Sunset":"16:54"}}
17:12:49.295 MQT: stat/tasmota_ESP12n_temp/STATUS10 = {"StatusSNS":{"Time":"2024-12-02T17:12:49","DS18B20":{"Id":"01215CBAE420","Temperature":18.1},"TempUnit":"C"}}
17:12:49.308 MQT: stat/tasmota_ESP12n_temp/STATUS11 = {"StatusSTS":{"Time":"2024-12-02T17:12:49","Uptime":"0T00:00:30","UptimeSec":30,"Heap":21,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":20,"MqttCount":1,"POWER":"ON","Wifi":{"AP":1,"SSId":"deco-m9-hr","BSSId":"42:84:6A:86:D5:05","Channel":10,"Mode":"11n","RSSI":50,"Signal":-75,"LinkCount":1,"Downtime":"0T00:00:05"}}}
17:12:50.223 MQT: tele/tasmota_ESP12n_temp/STATE = {"Time":"2024-12-02T17:12:50","Uptime":"0T00:00:31","UptimeSec":31,"Heap":25,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":36,"MqttCount":1,"POWER":"ON","Wifi":{"AP":1,"SSId":"deco-m9-hr","BSSId":"42:84:6A:86:D5:05","Channel":10,"Mode":"11n","RSSI":48,"Signal":-76,"LinkCount":1,"Downtime":"0T00:00:05"}}
17:12:50.259 MQT: tele/tasmota_ESP12n_temp/SENSOR = {"Time":"2024-12-02T17:12:50","DS18B20":{"Id":"01215CBAE420","Temperature":18.1},"TempUnit":"C"}

```
- [ ] Set `weblog` to 4 and then, when you experience your issue, provide the output of the Console log:
```lua
  Console output here:

```

### TO REPRODUCE
_Steps to reproduce the behavior:_
Flash or upgrade to version 14,3 and restore dump

### EXPECTED BEHAVIOUR
_A clear and concise description of what you expected to happen._

### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._

### ADDITIONAL CONTEXT
_Add any other context about the problem here._
The displaymodel could not set to 2 the second time i flashed to 14.3 and no restore was done.

**(Please, remember to close the issue when the problem has been addressed)**

Romik4x4, 2024-12-02:
#define USE_UNIVERSAL_DISPLAY                   // New universal display driver for both I2C and SPI
if define this - all is working.

Jason2866, 2024-12-02:
Please check release change entrys before opening an issue.

closed 2024-12-02 by Jason2866

Hennyr, 2024-12-02:
Should i use this in the console?

And is it now possible to use the display in the standard Tasmota?

---
Met vriendelijke groet,

Henny Ribbink

-------- Oorspronkelijke bericht --------

  		ONDERWERP:
  		Re: [arendst/Tasmota] SSD1306 Display does not work anymore on 
Tasmota 14.3 (Issue #22581)

  		DATUM:
  		02-12-2024 18:05

  		AFZENDER:
  		Roma ***@***.***>

  		ONTVANGER:
  		arendst/Tasmota ***@***.***>

#define USE_UNIVERSAL_DISPLAY // New universal display driver for both 
I2C and SPI
if define this - all is working.

Jason2866, 2024-12-02:
See https://tasmota.github.io/docs/Universal-Display-Driver/#migrating-to-udisplay

Hennyr, 2024-12-02:
Dear Jason2866,

I have checked the releasenotes, did not found this. Also google did not 
gave this.

But thanks for your help.

---
Met vriendelijke groet,

Henny Ribbink

-------- Oorspronkelijke bericht --------

  		ONDERWERP:
  		Re: [arendst/Tasmota] SSD1306 Display does not work anymore on 
Tasmota 14.3 (Issue #22581)

  		DATUM:
  		02-12-2024 20:08

  		AFZENDER:
  		Jason2866 ***@***.***>

  		ONTVANGER:
  		arendst/Tasmota ***@***.***>

Please check release change entrys before opening an issue.

iddq, 2024-12-30:
I am also having trouble with the SSD1306. Nothing appears, the screen is dark. Before I go into details, I would like to ask if the device can receive commands through the serial line used for firmware uploading? Unfortunately, it does not respond to anything, as if it is not receiving the commands. I am trying with the command screen /dev/ttyUSB0 115200. Debug messages appear, but it does not respond to anything through this.

sfromis, 2024-12-30:
Yes, you can get a serial console that way, as long as your `SerialLog` level allows things through (make sure to have at least log level 2), you recognize that Tasmota does not echo whats typed (operating in half duplex mode), and you may have to send each line with ctrl-J instead of an "enter/return" key. Details depends on config of your terminal program.

When it comes to SSD1306, it is important that you fully follow the config steps from the link provided by Jason2866.

iddq, 2024-12-30:
thanks for the ^J

iddq, 2024-12-30:
I haven't been able to make progress so far. Could the issue at the beginning, where the display is found at multiple addresses, be a problem?

00:00:00.065 I2C: SevenSeg found at 0x3c
00:00:00.066 I2C: SevenSeg found at 0x74
00:00:00.067 I2C: SevenSeg found at 0x75
00:00:00.068 I2C: SevenSeg found at 0x72
00:00:00.069 I2C: SevenSeg found at 0x73
00:00:00.073 I2C: SevenSeg found at 0x76
00:00:00.081 DSP: Rule3 descriptor used
00:00:00.212 DSP: SSD1306 initialized

00:28:05.834 CMD: i2cscan
00:28:05.835 SRC: Serial
00:28:05.836 CMD: Grp 0, Cmd 'I2CSCAN', Idx 1, Len 0, Pld -99, Data ''
00:28:05.854 RSL: RESULT = {"I2CScan":"Device(s) found at 0x3c"}

00:28:31.418 CMD: rule3
00:28:31.419 SRC: Serial
00:28:31.421 CMD: Grp 0, Cmd 'RULE', Idx 3, Len 0, Pld -99, Data ''
00:28:31.424 RSL: RESULT = {"Rule3":{"State":"OFF","Once":"OFF","StopOnError":"OFF","Length":166,"Free":345,"Rules":":H,SSD1306,128,64,1,I2C,3c,,,* :S,0,2,1,0,30,20 :I AE D5,80 A8,3F D3,00 40 8D,14 20,00 A1 C8 DA,12 81,9F D9,F1 DB,40 A4 A6 AF :o,AE :O,AF :A,00,10,40,00,00 :i,A6,A7 #"}}

rule3 from here: https://github.com/arendst/Tasmota/discussions/21412

00:29:20.472 WIF: Checking connection...
00:29:36.522 CMD: display
00:29:36.523 SRC: Serial
00:29:36.525 CMD: Grp 0, Cmd 'DISPLAY', Idx 1, Len 0, Pld -99, Data ''
00:29:36.528 RSL: RESULT = {"Display":{"Model":11,"Type":0,"Width":128,"Height":64,"Mode":1,"Dimmer":100,"Size":1,"Font":1,"Rotate":0,"Invert":0,"Refresh":2,"Cols":[16,8],"Rows":2}}

00:29:56.349 CMD: displaymode
00:29:56.350 SRC: Serial
00:29:56.352 CMD: Grp 0, Cmd 'DISPLAYMODE', Idx 1, Len 0, Pld -99, Data ''
00:29:56.354 RSL: RESULT = {"DisplayMode":1}

sfromis, 2024-12-31:
I suspect that `DisplayModel` 11 (which is seven-segment), leads to it trying to detect the wrong display type, especially if `DisplayAddress` was set to 60 - this is no longer relevant to the display types handled by the Universal Display Driver.

Try changing `DisplayModel` to something else, like 17.