arendst/Tasmota #22587: No MQTT connection attempt in Safeboot

ajaypala, 2024-12-04:
### PROBLEM DESCRIPTION
_A clear and concise description of what the problem is._
MQTT does not even attempt to make connection in safeboot. Note I'm using TLS for MQTT. Works fine in normal boot.

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [x] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [x] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [x] Searched the problem in [discussions](https://github.com/arendst/Tasmota/discussions)
- [x] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [ ] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [x] Device used (e.g., Sonoff Basic): Sonoff POWR320D
- [x] Tasmota binary firmware version number used: 14.3.0
  - [x] Pre-compiled
  - [ ] Self-compiled
- [x] Flashing tools used: OTA
- [x] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```lua
  Configuration output here:
08:03:46.459 RSL: RESULT = {"NAME":"Sonoff POWR320D","GPIO":[32,0,9313,0,9312,576,0,0,0,0,9280,0,3104,0,320,0,0,0,0,0,0,9184,9248,9216,0,0,0,0,0,0,0,0,0,0,0,0],"FLAG":0,"BASE":1}
08:03:46.673 RSL: RESULT = {"Module":{"0":"Sonoff POWR320D"}}
08:03:46.929 RSL: RESULT = {"GPIO0":{"32":"Button1"},"GPIO1":{"0":"None"},"GPIO2":{"9313":"Relay_b2"},"GPIO3":{"0":"None"},"GPIO4":{"9312":"Relay_b1"},"GPIO5":{"576":"LedLink_i"},"GPIO6":{"0":"None"},"GPIO7":{"0":"None"},"GPIO8":{"0":"None"},"GPIO9":{"0":"None"},"GPIO10":{"0":"None"},"GPIO11":{"0":"None"},"GPIO12":{"0":"None"},"GPIO13":{"0":"None"},"GPIO14":{"9280":"TM1621 DAT"},"GPIO15":{"0":"None"},"GPIO16":{"3104":"CSE7766 Rx"},"GPIO17":{"0":"None"},"GPIO18":{"320":"Led_i1"},"GPIO19":{"0":"None"},"GPIO20":{"0":"None"},"GPIO21":{"0":"None"},"GPIO22":{"0":"None"},"GPIO23":{"0":"None"},"GPIO24":{"0":"None"},"GPIO25":{"9184":"TM1621 CS"},"GPIO26":{"9248":"TM1621 RD"},"GPIO27":{"9216":"TM1621 WR"},"GPIO32":{"0":"None"},"GPIO33":{"0":"None"},"GPIO34":{"0":"None"},"GPIO35":{"0":"None"},"GPIO36":{"0":"None"},"GPIO37":{"0":"None"},"GPIO38":{"0":"None"},"GPIO39":{"0":"None"}}

```
- [ ] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```lua
  Rules output here:

```
- [x] Provide the output of this command: `Status 0`:
```lua
  STATUS 0 output here:
08:01:45.093 CMD: status 0
08:01:45.096 RSL: STATUS = {"Status":{"Module":0,"DeviceName":"Hot Water","FriendlyName":["Hot Water"],"Topic":"tasmota_112C94","ButtonTopic":"0","Power":"0","PowerLock":"0","PowerOnState":3,"LedState":1,"LedMask":"FFFF","SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0,"InfoRetain":0,"StateRetain":0,"StatusRetain":0}}
08:01:45.136 RSL: STATUS1 = {"StatusPRM":{"Baudrate":115200,"SerialConfig":"8N1","GroupTopic":"tasmotas","OtaUrl":"http://ota.tasmota.com/tasmota32/release/tasmota32.bin","RestartReason":"Software reset CPU","Uptime":"0T00:00:15","StartupUTC":"2024-12-04T19:01:30","Sleep":50,"CfgHolder":4617,"BootCount":36,"BCResetTime":"2022-08-11T21:11:33","SaveCount":3023}}
08:01:45.159 RSL: STATUS2 = {"StatusFWR":{"Version":"14.3.0(release-safeboot)single-core","BuildDateTime":"2024-10-15T08:18:18","Core":"3_1_0","SDK":"5.3.1.240924","CpuFrequency":240,"Hardware":"ESP32-D0WD-V3 v3.0","CR":"533/699"}}
08:01:45.181 RSL: STATUS3 = {"StatusLOG":{"SerialLog":2,"WebLog":2,"MqttLog":0,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["TP-Link_7859",""],"TelePeriod":300,"Resolution":"558080C0","SetOption":["00008009","2805C80001000680003C5A0A192800000000","00000081","00206000","00004000","00000000"]}}
08:01:45.215 RSL: STATUS4 = {"StatusMEM":{"ProgramSize":805,"Free":26,"Heap":211,"StackLowMark":4,"PsrMax":0,"PsrFree":0,"ProgramFlashSize":4096,"FlashSize":4096,"FlashChipId":"16405E","FlashFrequency":40,"FlashMode":"DIO","Features":["0809","00000588","00000003","00000002","00000000","00000000","40000000","00000000","00000000","00000000","00000000"],"Drivers":"1,2,82","Sensors":""}}
08:01:45.252 RSL: STATUS5 = {"StatusNET":{"Hostname":"tasmota-112C94-3220","IPAddress":"192.168.5.12","Gateway":"192.168.5.1","Subnetmask":"255.255.255.0","DNSServer1":"192.168.5.1","DNSServer2":"0.0.0.0","Mac":"C4:DE:E2:11:2C:94","IP6Global":"","IP6Local":"fe80::c6de:e2ff:fe11:2c94%st1","Ethernet":{"Hostname":"","IPAddress":"0.0.0.0","Gateway":"0.0.0.0","Subnetmask":"0.0.0.0","DNSServer1":"192.168.5.1","DNSServer2":"0.0.0.0","Mac":"00:00:00:00:00:00","IP6Global":"","IP6Local":""},"Webserver":2,"HTTP_API":1,"WifiConfig":4,"WifiPower":17.0}}
08:01:45.295 RSL: STATUS6 = {"StatusMQT":{"MqttHost":"mqtt.xxxx.org","MqttPort":8883,"MqttClientMask":"tasmota_%06X","MqttClient":"tasmota_112C94","MqttUser":"tasmota_112C94","MqttCount":0,"MAX_PACKET_SIZE":1200,"KEEPALIVE":30,"SOCKET_TIMEOUT":4}}
08:01:45.317 RSL: STATUS7 = {"StatusTIM":{"UTC":"2024-12-04T19:01:45Z","Local":"2024-12-05T08:01:45","StartDST":"2024-09-29T02:00:00","EndDST":"2024-04-07T03:00:00","Timezone":99}}
08:01:45.328 RSL: STATUS10 = {"StatusSNS":{"Time":"2024-12-05T08:01:45"}}
08:01:45.347 RSL: STATUS11 = {"StatusSTS":{"Time":"2024-12-05T08:01:45","Uptime":"0T00:00:15","UptimeSec":15,"Heap":211,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":0,"POWER":"OFF","Wifi":{"AP":1,"SSId":"TP-Link_7859","BSSId":"B0:BE:76:58:78:59","Channel":10,"Mode":"HE20","RSSI":84,"Signal":-58,"LinkCount":1,"Downtime":"0T00:00:04"}}}

```
- [x] Set `weblog` to 4 and then, when you experience your issue, provide the output of the Console log:
```lua
  Console output here:

```

### TO REPRODUCE
_Steps to reproduce the behavior:_
Restart 3 into safeboot.

### EXPECTED BEHAVIOUR
_A clear and concise description of what you expected to happen._
Once in safeboot, MQTT connection is established.

### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._

### ADDITIONAL CONTEXT
_Add any other context about the problem here._
MQTT TLS connection was being attempted in V12.something of safeboot on the same device with same settings. A OTA firmware update restarted into safeboot without actually updating any firmware. I then updated the safeboot firmware using upgrade 2. Noticed no MQTT connection in web console, meaning I will not be able to upgrade my remote devices. Used web console restart 3 back to normal and completed firmware upgrade using upgrade 1. 

**(Please, remember to close the issue when the problem has been addressed)**

Jason2866, 2024-12-05:
Not an issue. Behaviour by design. An OTA update does **not** update safeboot. So old safeboot version may missing features of actual safeboot version.
When being unsure which safeboot version is "on" the device, better OTA safeboot before updating Tasmota via OTA.

closed 2024-12-05 by Jason2866

ajaypala, 2024-12-05:
Is MQTT meant to work in safeboot? 

> Note: the Safeboot firmware is a reduced version of Tasmota containing only what's required for OTA updates (Web UI, MQTT, TLS...).
https://tasmota.github.io/docs/Safeboot/#introducing-safeboot

ajaypala, 2024-12-05:
If I turn off SO103 MQTT TLS before restart 3 into safeboot, MQTT will work. Seems like MQTT TLS is no longer support in safeboot. Is this by design? If so, I'll create a PR to update the documentation.

sfromis, 2024-12-05:
What is your Tasmota version in safeboot?

ajaypala, 2024-12-05:
> What is your Tasmota version in safeboot?

14.3.0

s-hadinger, 2024-12-06:
MQTT TLS is supported in safeboot