arendst/Tasmota #22605: Wrong Core Version in current development and prerelease 14.40

helmar74, 2024-12-08:
### PROBLEM DESCRIPTION
_A clear and concise description of what the problem is._

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [ ] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [ ] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [ ] Searched the problem in [discussions](https://github.com/arendst/Tasmota/discussions)
- [ ] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [ ] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [ ] Device used (e.g., Sonoff Basic): _____
- [ ] Tasmota binary firmware version number used: _____
  - [ ] Pre-compiled
  - [ ] Self-compiled
- [ ] Flashing tools used: _____
- [ ] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```lua
  Configuration output here:

```
- [ ] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```lua
  Rules output here:

```
- [ ] Provide the output of this command: `Status 0`:
```lua
  STATUS 0 output here:

```
- [ ] Set `weblog` to 4 and then, when you experience your issue, provide the output of the Console log:
```lua
  Console output here:

```

### TO REPRODUCE
_Steps to reproduce the behavior:_

### EXPECTED BEHAVIOUR
_A clear and concise description of what you expected to happen._

### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._

### ADDITIONAL CONTEXT
_Add any other context about the problem here._

**(Please, remember to close the issue when the problem has been addressed)**

arendst, 2024-12-08:
What do you see as core version?

helmar74, 2024-12-08:
> What do you see as core version?

```
16:35:11.232 CMD: status 2
16:35:11.234 SRC: WebConsole from 192.168.1.31
16:35:11.235 CMD: Grp 0, Cmd 'STATUS', Idx 1, Len 1, Pld 2, Data '2'
16:35:11.239 MQT: stat/powerplug_6/STATUS2 = {"StatusFWR":{"Version":"14.4.0(tasmota)","BuildDateTime":"2024.12.08 16:30:41","Boot":31,"Core":"2_7_7","SDK":"2.2.2-dev(38a443e)","CpuFrequency":160,"Hardware":"ESP8285N08","CR":"384/699"}}
```

arendst, 2024-12-08:
Strange. I see 2.7.8

```
{"StatusFWR":{"Version":"14.3.0.7(theo)","BuildDateTime":"2024-12-08T16:37:59","Boot":31,"Core":"2_7_8","SDK":"2.2.2-dev(38a443e)","CpuFrequency":80,"Hardware":"ESP8266EX","CR":"410/699"}}
```

helmar74, 2024-12-08:
> Strange. I see 2.7.8
> 
> ```
> {"StatusFWR":{"Version":"14.3.0.7(theo)","BuildDateTime":"2024-12-08T16:37:59","Boot":31,"Core":"2_7_8","SDK":"2.2.2-dev(38a443e)","CpuFrequency":80,"Hardware":"ESP8266EX","CR":"410/699"}}
> ```

weird... just for trying please use preRelease 14.4.0

helmar74, 2024-12-08:
> > Strange. I see 2.7.8
> > ```
> > {"StatusFWR":{"Version":"14.3.0.7(theo)","BuildDateTime":"2024-12-08T16:37:59","Boot":31,"Core":"2_7_8","SDK":"2.2.2-dev(38a443e)","CpuFrequency":80,"Hardware":"ESP8266EX","CR":"410/699"}}
> > ```
> 
> weird... just for trying please use preRelease 14.4.0

During compilation the log says, the core 2.7.8 is used:

```
*** use provided platformio_tasmota_cenv.ini as planned ***
CONFIGURATION: https://docs.platformio.org/page/boards/espressif8266/esp8266_1M.html
PLATFORM: Espressif 8266 (2024.9.0) > Espressif Generic ESP8266 Tasmota 1M sketch NO FS
HARDWARE: ESP8266 160MHz, 80KB RAM, 972.00KB Flash
PACKAGES: 
 - framework-arduinoespressif8266 @ **2.7.8** 
 - tool-esptoolpy @ 4.7.6 
```

helmar74, 2024-12-08:
I removed ~./platformio directory, now it's working

closed 2024-12-08 by helmar74

helmar74, 2024-12-08:
```
17:39:57.201 CMD: status 2
17:39:57.202 SRC: WebConsole from 192.168.1.31
17:39:57.203 CMD: Grp 0, Cmd 'STATUS', Idx 1, Len 1, Pld 2, Data '2'
17:39:57.207 MQT: stat/powerplug_6/STATUS2 = {"StatusFWR":{"Version":"14.4.0(tasmota)","BuildDateTime":"2024.12.08 17:35:44","Boot":31,"Core":"2_7_8","SDK":"2.2.2-dev(38a443e)","CpuFrequency":160,"Hardware":"ESP8285N08","CR":"384/699"}}
```