arendst/Tasmota #22613: Teleinfo plugin not working on GPIO13

azertylr, 2024-12-09:
### PROBLEM DESCRIPTION
Plugin Teleinfo https://tasmota.github.io/docs/Teleinfo/ is not working/stable

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [X] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [X] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [X] Searched the problem in [discussions](https://github.com/arendst/Tasmota/discussions)
- [X] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [X] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [X] Device used (e.g., Sonoff Basic): Wemos D1 mini
- [X] Tasmota binary firmware version number used: 14.3.0
  - [ ] Pre-compiled
  - [X] Self-compiled with https://github.com/benzino77/tasmocompiler/
- [X] Flashing tools used: [_____](https://tasmota.github.io/docs/Teleinfo/)

### TO REPRODUCE
_Steps to reproduce the behavior:_

When using GPIO 3 - Rx as indicated in the doc, power information doesn't show up in the device. It doesn't read any data.

When using GPIO13 - D7, it is working fine.

### EXPECTED BEHAVIOUR
_A clear and concise description of what you expected to happen._

the plugin should work on pin GPIO3 as indicated in the doc or update the doc

Jason2866, 2024-12-09:
@hallard  Charles can you have a look?

hallard, 2024-12-09:
Sure, WeMos D1 Mini does not have any hardware to receive teleinfo, so I imagine external hardware has been used. But without knowing which hardware is used and how it's connected, moreover which documentation is used, it's quite hard to find a solution.

barbudor, 2024-12-09:
Gpio 3 is standard RX connected to the USB chip on the mini D1
If the board is connected to a PC, or simply powered by USB, the USB chip is in control of the serial port on gpio 1 & 3

So forget if you absolutely want to use gpio3, power the board through Vin and do not connect anything to USB

This being said, using GPIO13 is perfect for Teleinfo

closed 2024-12-09 by azertylr