arendst/Tasmota #22624: Release firmware image 14.4.0 for ESP32 ZBB seems defective

meyergru, 2024-12-11:
### PROBLEM DESCRIPTION
The file https://ota.tasmota.com/tasmota32/release/tasmota32-zbbrdgpro.bin seems defective.

When I update the firmware on my device, I do not get the "upload successful" message and the device stays in safeboot mode.

This worked up to version 14.3.0, although I once had this problem before.

The image seems a little small, usually images get bigger from release to release. This time, the image is somewhat smaller than the previous one.

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [x] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [x] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [x] Searched the problem in [discussions](https://github.com/arendst/Tasmota/discussions)
- [x] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [x] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [x] Device used (e.g., Sonoff Basic): Sonoff ZBBrige Pro
- [x] Tasmota binary firmware version number used: 14.4.0
  - [x] Pre-compiled
  - [ ] Self-compiled
- [x] Flashing tools used: Web upgrade
- [x] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```lua
  Configuration output here:
16:05:43.637 MQT: stat/zigbeepro/RESULT = {"NAME":"TCP ZBBridge Pro","GPIO":[0,0,576,0,480,0,0,0,0,1,1,5792,0,0,0,5472,0,320,5793,5504,0,640,608,32,0,0,0,0,0,1,0,0,0,0,0,0],"FLAG":0,"BASE":1}
16:05:43.861 MQT: stat/zigbeepro/RESULT = {"Module":{"0":"TCP ZBBridge Pro"}}
16:05:44.121 MQT: stat/zigbeepro/RESULT = {"GPIO0":{"0":"None"},"GPIO1":{"0":"None"},"GPIO2":{"576":"LedLink_i"},"GPIO3":{"0":"None"},"GPIO4":{"480":"Buzzer"},"GPIO5":{"0":"None"},"GPIO6":{"0":"None"},"GPIO7":{"0":"None"},"GPIO8":{"0":"None"},"GPIO9":{"0":"None"},"GPIO10":{"0":"None"},"GPIO11":{"0":"None"},"GPIO12":{"0":"None"},"GPIO13":{"0":"None"},"GPIO14":{"0":"None"},"GPIO15":{"5792":"Zigbee Rst1"},"GPIO16":{"0":"None"},"GPIO17":{"0":"None"},"GPIO18":{"0":"None"},"GPIO19":{"5472":"TCP Tx"},"GPIO20":{"0":"None"},"GPIO21":{"320":"Led_i1"},"GPIO22":{"5793":"Zigbee Rst2"},"GPIO23":{"5504":"TCP Rx"},"GPIO24":{"0":"None"},"GPIO25":{"640":"I2C SDA1"},"GPIO26":{"608":"I2C SCL1"},"GPIO27":{"32":"Button1"},"GPIO32":{"0":"None"},"GPIO33":{"0":"None"},"GPIO34":{"0":"None"},"GPIO35":{"0":"None"},"GPIO36":{"0":"None"},"GPIO37":{"0":"None"},"GPIO38":{"0":"None"},"GPIO39":{"0":"None"}}
```
- [ ] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```lua
  Rules output here:
16:06:29.501 MQT: stat/zigbeepro/RESULT = {"Rule1":{"State":"ON","Once":"OFF","StopOnError":"OFF","Length":37,"Free":474,"Rules":"on system#boot do TCPStart 8888 endon"}}
16:06:29.726 MQT: stat/zigbeepro/RESULT = {"Rule2":{"State":"OFF","Once":"OFF","StopOnError":"OFF","Length":0,"Free":511,"Rules":""}}
16:06:29.928 MQT: stat/zigbeepro/RESULT = {"Rule3":{"State":"OFF","Once":"OFF","StopOnError":"OFF","Length":0,"Free":511,"Rules":""}}
```
- [ ] Provide the output of this command: `Status 0`:
```lua
  STATUS 0 output here:
16:07:05.041 MQT: stat/zigbeepro/STATUS = {"Status":{"Module":0,"DeviceName":"zigbeepro","FriendlyName":["zigbeepro"],"Topic":"zigbeepro","ButtonTopic":"0","Power":"0","PowerLock":"0","PowerOnState":3,"LedState":1,"LedMask":"FFFF","SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0,"InfoRetain":0,"StateRetain":0,"StatusRetain":0}}
16:07:05.084 MQT: stat/zigbeepro/STATUS1 = {"StatusPRM":{"Baudrate":115200,"SerialConfig":"8N1","GroupTopic":"tasmotas","OtaUrl":"http://ota.tasmota.com/tasmota32/release/tasmota32-zbbrdgpro.bin","RestartReason":"Software reset CPU","Uptime":"0T00:06:33","StartupUTC":"2024-12-11T15:00:32","Sleep":50,"CfgHolder":4617,"BootCount":49,"BCResetTime":"2023-01-02T20:29:53","SaveCount":342}}
16:07:05.123 MQT: stat/zigbeepro/STATUS2 = {"StatusFWR":{"Version":"14.3.0(release-zbbrdgpro)","BuildDateTime":"2024-10-15T08:20:56","Core":"3_1_0","SDK":"5.3.1.240924","CpuFrequency":160,"Hardware":"ESP32-D0WD-V3 v3.0","CR":"386/699"}}
16:07:05.141 MQT: stat/zigbeepro/STATUS3 = {"StatusLOG":{"SerialLog":2,"WebLog":2,"MqttLog":0,"SysLog":0,"LogHost":"syslog.mgsoft","LogPort":514,"SSId":["IdIoT",""],"TelePeriod":60,"Resolution":"558180C0","SetOption":["00008009","2805C80001000600003C5A0A192800000000","00000088","00006000","00004000","00000000"]}}
16:07:05.178 MQT: stat/zigbeepro/STATUS4 = {"StatusMEM":{"ProgramSize":1567,"Free":1120,"Heap":155,"StackLowMark":3,"PsrMax":2048,"PsrFree":2022,"ProgramFlashSize":4096,"FlashSize":4096,"FlashChipId":"16405E","FlashFrequency":40,"FlashMode":"DIO","Features":["0809","8F8A87CF","00548001","000000CF","010013C0","C0000981","40024080","00200000","5400082C","00000080","00000004"],"Drivers":"1,2,!3,!4,!5,!8,9,10,12,!16,!20,!21,!23,24,!27,29,!35,38,41,50,52,!59,62,!68,82,!121","Sensors":"1,2,3,5,6,127","I2CDriver":"7"}}
16:07:05.228 MQT: stat/zigbeepro/STATUS5 = {"StatusNET":{"Hostname":"zigbeepro","IPAddress":"192.168.7.30","Gateway":"192.168.7.1","Subnetmask":"255.255.255.0","DNSServer1":"192.168.7.1","DNSServer2":"0.0.0.0","Mac":"0C:DC:7E:32:BE:9C","IP6Global":"2001:xxx:xxx:xxx:edc:7eff:fe32:be9c","IP6Local":"fe80::edc:7eff:fe32:be9c%st1","Ethernet":{"Hostname":"","IPAddress":"0.0.0.0","Gateway":"0.0.0.0","Subnetmask":"0.0.0.0","DNSServer1":"192.168.7.1","DNSServer2":"0.0.0.0","Mac":"00:00:00:00:00:00","IP6Global":"","IP6Local":""},"Webserver":2,"HTTP_API":1,"WifiConfig":4,"WifiPower":17.0}}
16:07:05.278 MQT: stat/zigbeepro/STATUS6 = {"StatusMQT":{"MqttHost":"mqtt.iot","MqttPort":1883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_32BE9C","MqttUser":"DVES_USER","MqttCount":1,"MAX_PACKET_SIZE":1200,"KEEPALIVE":30,"SOCKET_TIMEOUT":4}}
16:07:05.306 MQT: stat/zigbeepro/STATUS7 = {"StatusTIM":{"UTC":"2024-12-11T15:07:05Z","Local":"2024-12-11T16:07:05","StartDST":"2024-03-31T02:00:00","EndDST":"2024-10-27T03:00:00","Timezone":"+01:00","Sunrise":"08:34","Sunset":"16:52"}}
16:07:05.326 MQT: stat/zigbeepro/STATUS10 = {"StatusSNS":{"Time":"2024-12-11T16:07:05"}}
16:07:05.339 MQT: stat/zigbeepro/STATUS11 = {"StatusSTS":{"Time":"2024-12-11T16:07:05","Uptime":"0T00:06:33","UptimeSec":393,"Heap":155,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":1,"Berry":{"HeapUsed":22,"Objects":296},"Wifi":{"AP":1,"SSId":"IdIoT","BSSId":"26:5A:4C:B9:EA:FA","Channel":6,"Mode":"HT40","RSSI":100,"Signal":-48,"LinkCount":1,"Downtime":"0T00:00:04"}}}
```
- [ ] Set `weblog` to 4 and then, when you experience your issue, provide the output of the Console log:
```lua
  Console output here:

```

### TO REPRODUCE
_Steps to reproduce the behavior:_

Try to update the firmware via web UI.

### EXPECTED BEHAVIOUR
_A clear and concise description of what you expected to happen._

Should upgrade firmware.

### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._

### ADDITIONAL CONTEXT
_Add any other context about the problem here._

**(Please, remember to close the issue when the problem has been addressed)**

helmar74, 2024-12-11:
Have tried uploading it again?
Happened on 1-2 ESP32 devices for me also, but after uploading (in safeboot) again, it worked...

meyergru, 2024-12-11:
I tried it three times, even downloaded it and updated from my local copy to no avail. Obviously, second and third tries were from safeboot. In contrast, the 14.3.0 image works every time.

closed 2024-12-11 by meyergru