arendst/Tasmota #22628: Connect failed to mqtt rc -5

flogehring, 2024-12-11:
### PROBLEM DESCRIPTION

I just want to log this here, in case somebody else has the same problem.
After I've spent an hour trying to figure out why my device wont connect to my broker.
The host address may NOT contain `mqtt://`

This fails:
[image]

This works
[image]

Feel free to close the issue
OR 
add a hint to the UI
OR
parse the input

sfromis, 2024-12-11:
The hint is already in the UI. "Host" does not mean "URL".

barbudor, 2024-12-12:
You are free to contribute a PR on the doc repo if you feel that what you read there was not clear enough
Thanks in advance for your contribution