arendst/Tasmota #22652: Energy calculation not proper

rohanpatil-rc, 2024-12-15:
I Have [Wipro 16A Power Monitoring Plug](https://templates.blakadder.com/wipro_DS11160.html)
Monitoring refrigerator it has BL0937 with tuya chip which i replace with esp8266(tasmota 14.3.0). the reading for 14dec are weird 
screenshots  to explain the issues i have install it 2days ago and those reading are perfect.
[image]
[image]
[image]

sfromis, 2024-12-15:
I see nothing "weird", just signs that your device experiences loss of input power. Tasmota does not wear down flash by continuously writing persistent data.

rohanpatil-rc, 2024-12-15:
> I see nothing "weird", just signs that your device experiences loss of input power. Tasmota does not wear down flash by continuously writing persistent data.

ok i will try with different appliance and resolder the esp8266 conations