arendst/Tasmota #22704: Matter Air Quality Sensor with MHZ19B not showing values in Apple Home

stony2k, 2024-12-22:
### PROBLEM DESCRIPTION
### What I've tried
1. Configured Matter endpoint for Air Quality Sensor with `MHZ19B#CarbonDioxide` and `MHZ19B`
2. Sensor delivers correct values:
14:03:05.045 MQT: tele/tasmota_B8CCBC/SENSOR = {"Time":"2024-12-22T14:03:05","MHZ19B":{"Model":"B","CarbonDioxide":908,"Temperature":32.0},"TempUnit":"C"}

### Troubleshooting Steps
- Factory reset Matter
- Removing and re-adding the device in Apple Home
- Different Matter endpoint configurations
- Restarting Tasmota

### Related Code
- [Matter Air Quality Sensor Implementation](https://github.com/arendst/Tasmota/blob/master/lib/libesp32/berry_matter/src/embedded/Matter_Plugin_2_Sensor_Air_Quality.be)
- [Pull Request that added Air Quality support](https://github.com/arendst/Tasmota/pull/21559/commits/28fd3bbbf99e5c2d3d0b918a71e38b8c1011e38c)

### Question
What am I missing to get the CO2 values showing up correctly in Apple Home? The sensor works, Matter is configured, but the values are not being transmitted correctly. According to the berry script implementation, the values should be transmitted in a correct format to Apple Home.

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [x] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [x] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [x] Searched the problem in [discussions](https://github.com/arendst/Tasmota/discussions)
- [x] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [x] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [x] Device used (e.g., Sonoff Basic): Nodemcu ESP32
- [x] Tasmota binary firmware version number used: 14.4.1.1 (3412761-tasmota32)
  - [x] Pre-compiled
  - [ ] Self-compiled
- [x] Flashing tools used: https://tasmota.github.io/install/
- [x ] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```lua
18:21:03.587 CMD: Backlog Template; Module; GPIO 255
18:21:03.589 SRC: WebConsole from 10.0.2.1
18:21:03.590 CMD: Grp 0, Cmd 'BACKLOG', Idx 1, Len 26, Pld -99, Data 'Template; Module; GPIO 255'
```
- [ ] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```lua
  Rules output here:

```
- [x] Provide the output of this command: `Status 0`:
```lua
18:23:07.811 CMD: status 0
18:23:07.815 RSL: STATUS = {"Status":{"Module":1,"DeviceName":"Tasmota","FriendlyName":["Tasmota"],"Topic":"tasmota_B8CCBC","ButtonTopic":"0","Power":"0","PowerLock":"0","PowerOnState":3,"LedState":1,"LedMask":"FFFF","SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0,"InfoRetain":0,"StateRetain":0,"StatusRetain":0}}
18:23:07.855 RSL: STATUS1 = {"StatusPRM":{"Baudrate":115200,"SerialConfig":"8N1","GroupTopic":"tasmotas","OtaUrl":"http://ota.tasmota.com/tasmota32/release/tasmota32-DE.bin","RestartReason":"Software reset CPU","Uptime":"0T00:06:03","StartupUTC":"2024-12-22T17:17:04","Sleep":50,"CfgHolder":4617,"BootCount":11,"BCResetTime":"2024-12-22T18:00:19","SaveCount":25}}
18:23:07.893 RSL: STATUS2 = {"StatusFWR":{"Version":"14.4.1.1(3412761-tasmota32)","BuildDateTime":"2024.12.22 14:39:18","Core":"3_1_0","SDK":"5.3.2","CpuFrequency":160,"Hardware":"ESP32-D0WDQ6 v1.0","CR":"412/699"}}
18:23:07.911 RSL: STATUS3 = {"StatusLOG":{"SerialLog":2,"WebLog":2,"MqttLog":0,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["Stony",""],"TelePeriod":300,"Resolution":"558180C0","SetOption":["00008009","2805C80001000600003C5A0A192800000000","00000080","00006000","00004000","00000020"]}}
18:23:07.952 RSL: STATUS4 = {"StatusMEM":{"ProgramSize":2019,"Free":860,"Heap":104,"StackLowMark":2,"PsrMax":0,"PsrFree":0,"ProgramFlashSize":4096,"FlashSize":4096,"FlashChipId":"16405E","FlashFrequency":40,"FlashMode":"DIO","Features":["0407","9F9AD7DF","0015A001","B7F7BFCF","05DA9BC4","E0360DC7","480840D2","20200000","D4BC482D","810A80F1","00000014"],"Drivers":"1,2,!3,!4,!5,7,!8,9,10,11,12,!14,!16,!17,!20,!21,!24,26,!27,29,!34,!35,38,50,52,!59,!60,62,!63,!66,!67,!68,!73,!75,82,!86,!87,!88,!121","Sensors":"1,2,3,5,6,7,8,9,10,11,12,13,14,15,17,18,19,20,21,22,26,31,34,37,39,40,42,43,45,51,52,55,56,58,59,64,66,67,74,85,92,95,98,103,105,109,127","I2CDriver":"7,8,9,10,11,12,13,14,15,17,18,20,24,29,31,36,41,42,44,46,48,58,62,65,69,76,77,82,89"}}
18:23:08.023 RSL: STATUS5 = {"StatusNET":{"Hostname":"tasmota-B8CCBC-3260","IPAddress":"10.0.0.121","Gateway":"10.0.0.1","Subnetmask":"255.255.0.0","DNSServer1":"10.0.1.22","DNSServer2":"0.0.0.0","Mac":"08:3A:F2:B8:CC:BC","IP6Global":"fdfe:9a7f:f9e:2347:a3a:f2ff:feb8:ccbc","IP6Local":"fe80::a3a:f2ff:feb8:ccbc%st1","Ethernet":{"Hostname":"","IPAddress":"0.0.0.0","Gateway":"0.0.0.0","Subnetmask":"0.0.0.0","DNSServer1":"10.0.1.22","DNSServer2":"0.0.0.0","Mac":"00:00:00:00:00:00","IP6Global":"","IP6Local":""},"Webserver":2,"HTTP_API":1,"WifiConfig":4,"WifiPower":16.0}}
18:23:08.073 RSL: STATUS6 = {"StatusMQT":{"MqttHost":"","MqttPort":1883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_B8CCBC","MqttUser":"DVES_USER","MqttCount":0,"MAX_PACKET_SIZE":1200,"KEEPALIVE":30,"SOCKET_TIMEOUT":4}}
18:23:08.090 RSL: STATUS7 = {"StatusTIM":{"UTC":"2024-12-22T17:23:08Z","Local":"2024-12-22T18:23:08","StartDST":"2024-03-31T02:00:00","EndDST":"2024-10-27T03:00:00","Timezone":"+01:00","Sunrise":"08:41","Sunset":"16:55"}}
18:23:08.120 RSL: STATUS10 = {"StatusSNS":{"Time":"2024-12-22T18:23:08","MHZ19B":{"Model":"B","CarbonDioxide":939,"Temperature":27.0},"TempUnit":"C"}}
18:23:08.142 RSL: STATUS11 = {"StatusSTS":{"Time":"2024-12-22T18:23:08","Uptime":"0T00:06:04","UptimeSec":364,"Heap":101,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":0,"Berry":{"HeapUsed":22,"Objects":278},"Wifi":{"AP":1,"SSId":"Stony","BSSId":"1C:ED:6F:F8:47:81","Channel":11,"Mode":"HT40","RSSI":54,"Signal":-73,"LinkCount":1,"Downtime":"0T00:00:04"}}}
```
- [ ] Set `weblog` to 4 and then, when you experience your issue, provide the output of the Console log:
```lua
  Console output here:

```

### TO REPRODUCE
1. Connect the MHZ19B CO2 sensor to Tasmota.
2. Enable Matter and pair it with Apple Home.
3. Configure the Matter endpoint for Air Quality Sensor using MHZ19B#CarbonDioxide.
4. Observe the Air Quality status in Apple Home.

### EXPECTED BEHAVIOUR
The CO2 values from the MHZ19B sensor should be displayed correctly in Apple Home's Air Quality widget.

### SCREENSHOTS
[image]
[image]
[image]

### ADDITIONAL CONTEXT

s-hadinger, 2024-12-22:
I'm not sure what Apple supports with Air Quality Sensor. CO2 level and "Air Quality" are different values.

stony2k, 2024-12-22:
Apple supports CO2 sensors in general but I am not sure if they are supported and implemented in Matter. Here is a, non Matter, device which I implemented via ioBroker and YAHKA for comparison. For air quality 2 datapoints are used:
[image]

CarbonDioxideLevel and AirQuality. For AirQuality I mapped CarbonDioxideLevel to different numbers for the quality level:
`if (value <= 750)
return 1;
else if (value <= 1000)
return 2;
else if (value <= 1250)
return 3;
else if (value <= 1500)
return 4;
else if (value <= 1750)
return 5;`

s-hadinger, 2024-12-22:
Ok, I'll do some testing.

stony2k, 2024-12-22:
Home Assistant recognises this device via matter as well, but also no values are shown: [image]

s-hadinger, 2024-12-23:
It should be fixed now. Please try latest in the `development` branch.

Note: the filter in Matter configuration should be only the overall key `I9B`

closed 2024-12-23 by s-hadinger

stony2k, 2024-12-24:
Air Quality is now shown in Apple Home, yes. Can you have a look if it is possible to show the CO2 value as well? I should work since it is implemented for Homekit devices? https://developer.apple.com/documentation/homekit/hmcharacteristictypecarbondioxidelevel
I thinks there should be a way to display the value as well.

s-hadinger, 2024-12-24:
Can you please send logs with `weblog 3` during reporting of values ?

stony2k, 2024-12-24:
Sure, do you need more information?
[weblog_3.txt](https://github.com/user-attachments/files/18237830/weblog_3.txt)

s-hadinger, 2024-12-26:
Everything is correct in the logs, the values for CO2 are reported to the subscriptions

PR #22708 (2024-12-23): Matter fix air quality
## Description:

Matter fix air quality, and compute AirQuality from CO2:
- Good (1): up to 750 ppm
- Fair (2): up to 1000 ppm
- Moderate (3): up to 1250 ppm
- Poor (4): up to 1500 ppm
- VeryPoor (5): up to 1750 ppm
- ExtremelyPoor (6): above 1750 ppm

**Related issue (if applicable):** fixes #22704

## Checklist:
  - [x] The pull request is done against the latest development branch
  - [x] Only relevant files were touched
  - [x] Only one feature/fix was added per PR and the code change compiles without warnings
  - [ ] The code change is tested and works with Tasmota  [... cut]