arendst/Tasmota #22714: KNX reception of telegrams no longer works 14.4.x

skynet74, 2024-12-25:
Since versiob 14.4. KNX reception of telegrams no longer works.
Sending telegrams still works perfectly.

I have not made any further configuration changes. Other Tasmota Shellys with 14.2 run without problems.

I tried downgrading to 14.2, unfortunately it doesn't work.

Device: Shelly 1 V3 with knx firmware 14.4.1 

[image]

Noschvie, 2024-12-25:
Did you have a look to #22242 ?

krishy2, 2025-01-01:
Same issue here after upgrading from v12.

krishy2, 2025-01-01:
Config save, config reset, then config restore fix it.
However, not elegant.

Also all the sensors should be included, not just the first.

krishy2, 2025-01-06:
After several days the KNX reception stopped.

clkempe, 2025-01-19:
I have the same issue. After a reboot it works for a time period and after a while it doesn't work any more. About 10 devices affected; different models.