arendst/Tasmota #22729: Gosund P1 Analog button error in release 14.4.X

gerliczky, 2024-12-28:
### PROBLEM DESCRIPTION
Analog buttons in Gosund P1 can be operated with a rule:
```
Rule2 
  ON Analog#A0div10<30 DO Power3 2 BREAK 
  ON Analog#A0div10<60 DO Power2 2 BREAK 
  ON Analog#A0div10<80 DO Power1 2 ENDON
``` 
as written here: https://templates.blakadder.com/gosund_P1.html

In release 14..0.0 it works perfectly, but in release 14.4.0 it is stopped working. ( I didn't tested the in between releases.)

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [x] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [x] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [x] Searched the problem in [discussions](https://github.com/arendst/Tasmota/discussions)
- [x] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [x] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [x] Device used (e.g., Sonoff Basic): Gosund P1
- [x] Tasmota binary firmware version number used: 14.4.0
  - [x] Pre-compiled
  - [ ] Self-compiled

arendst, 2024-12-28:
Checking...

gerliczky, 2024-12-28:
Result in 14.0.0
```
{"AdcParam1":[1,0,0,0.0005,0]}
```

closed 2025-01-05 by arendst