arendst/Tasmota #22730: After a firmware update "tasmota/discovery/..../sensors" sends an incomplete data set (ESP8285N08)

Rennpferd, 2024-12-28:
### PROBLEM DESCRIPTION

After a firmware update     "tasmota/discovery/..../sensors"   sends an incomplete data set

I use a "Nous L12T WiFi Smart Curtain Module with Tasmota" with ESP8285N08 and firmware version 13.2.0. Everything works great.

I use it in "MQT: stat/tasmota_8F447F/RESULT = {"ShutterMode":1}"

Then I do an OTA update to the current version firmware 14.4.1.

The update runs without errors. Then I see that the position of the roller shutter is no longer displayed in % in the Home Assistant. This and other sensors are no longer defined and should no longer exist!

These sensors are missing in "tasmota/discovery/..../sensors" after the update:

"Position"

"Direction"

"Target"

"Tilt"

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [x] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [x] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [x] Searched the problem in [discussions](https://github.com/arendst/Tasmota/discussions)
- [x] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [x] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [x] Device used (e.g., Sonoff Basic): Nous L12T WiFi Smart Curtain Module with Tasmota
- [x] Tasmota binary firmware version number used: 14.4.1
  - [x] Pre-compiled
  - [ ] Self-compiled
- [x] Flashing tools used: OTA Url http://ota.tasmota.com/tasmota/tasmota.bin.gz
- [x] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```lua
  Configuration output here:

Firmware Version 13.2.0

15:13:02.826 CMD: Backlog Template; Module; GPIO 255
15:13:02.859 MQT: stat/tasmota_8F447F/RESULT = {"NAME":" Smart WiFi Curtain Module L12T","GPIO":[1,160,1,161,225,224,1,1,544,1,32,1,1,1],"FLAG":0,"BASE":18}
15:13:03.081 MQT: stat/tasmota_8F447F/RESULT = {"Module":{"0":" Smart WiFi Curtain Module L12T"}}
15:13:03.340 MQT: stat/tasmota_8F447F/RESULT = {"GPIO0":{"0":"None"},"GPIO1":{"160":"Switch1"},"GPIO2":{"0":"None"},"GPIO3":{"161":"Switch2"},"GPIO4":{"225":"Relay2"},"GPIO5":{"224":"Relay1"},"GPIO9":{"0":"None"},"GPIO10":{"0":"None"},"GPIO12":{"544":"LedLink"},"GPIO13":{"0":"None"},"GPIO14":{"32":"Button1"},"GPIO15":{"0":"None"},"GPIO16":{"0":"None"},"GPIO17":{"0":"None"}}

```

```
- [ ] Provide the output of this command: `Status 0`:
```lua
  STATUS 0 output here:

Firmware Version 13.2.0

15:14:19.477 MQT: stat/tasmota_8F447F/STATUS = {"Status":{"Module":0,"DeviceName":"Rolladen Schlafzimmer nous Tasmota","FriendlyName":["Rolladen Schlafzimmer nous Tasmota DN","Rolladen Schlafzimmer nous Tasmota FN1"],"Topic":"tasmota_8F447F","ButtonTopic":"0","Power":0,"PowerOnState":3,"LedState":1,"LedMask":"FFFF","SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[4,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0,"InfoRetain":0,"StateRetain":0,"StatusRetain":0}}
15:14:19.524 MQT: stat/tasmota_8F447F/STATUS1 = {"StatusPRM":{"Baudrate":115200,"SerialConfig":"8N1","GroupTopic":"tasmotas","OtaUrl":"http://ota.tasmota.com/tasmota/release-13.2.0/tasmota.bin.gz","RestartReason":"Software/System restart","Uptime":"0T00:03:02","StartupUTC":"2024-12-28T14:11:17","Sleep":50,"CfgHolder":4617,"BootCount":11,"BCResetTime":"2023-05-28T10:43:47","SaveCount":30,"SaveAddress":"F4000"}}
15:14:19.558 MQT: stat/tasmota_8F447F/STATUS2 = {"StatusFWR":{"Version":"13.2.0(tasmota)","BuildDateTime":"2023-10-19T09:02:07","Boot":31,"Core":"2_7_4_9","SDK":"2.2.2-dev(38a443e)","CpuFrequency":80,"Hardware":"ESP8285N08","CR":"544/699"}}
15:14:19.579 MQT: stat/tasmota_8F447F/STATUS3 = {"StatusLOG":{"SerialLog":2,"WebLog":2,"MqttLog":4,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["caupona_24",""],"TelePeriod":300,"Resolution":"558180C0","SetOption":["0000C009","2805C80001000600003C5A0A192800000000","40000080","00006000","00004000","00000000"]}}
15:14:19.613 MQT: stat/tasmota_8F447F/STATUS4 = {"StatusMEM":{"ProgramSize":635,"Free":368,"Heap":22,"ProgramFlashSize":1024,"FlashSize":1024,"FlashChipId":"144051","FlashFrequency":40,"FlashMode":"DOUT","Features":["00000809","8F9AC787","04368001","000000CF","010013C0","C000F981","00004004","00001000","54000020","00000080"],"Drivers":"1,2,3,4,5,6,7,8,9,10,12,16,18,19,20,21,22,24,26,27,29,30,35,37,45,62,68","Sensors":"1,2,3,4,5,6","I2CDriver":"7"}}
15:14:19.650 MQT: stat/tasmota_8F447F/STATUS5 = {"StatusNET":{"Hostname":"tasmota-8F447F-1151","IPAddress":"192.168.178.190","Gateway":"192.168.178.1","Subnetmask":"255.255.255.0","DNSServer1":"192.168.178.1","DNSServer2":"0.0.0.0","Mac":"7C:87:CE:8F:44:7F","Webserver":2,"HTTP_API":1,"WifiConfig":4,"WifiPower":17.0}}
15:14:19.677 MQT: stat/tasmota_8F447F/STATUS6 = {"StatusMQT":{"MqttHost":"192.168.178.46","MqttPort":1883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_8F447F","MqttUser":"homeassistant","MqttCount":1,"MAX_PACKET_SIZE":1200,"KEEPALIVE":30,"SOCKET_TIMEOUT":4}}
15:14:19.705 MQT: stat/tasmota_8F447F/STATUS7 = {"StatusTIM":{"UTC":"2024-12-28T14:14:19","Local":"2024-12-28T15:14:19","StartDST":"2024-03-31T02:00:00","EndDST":"2024-10-27T03:00:00","Timezone":"+01:00","Sunrise":"08:43","Sunset":"17:00"}}
15:14:19.725 MQT: stat/tasmota_8F447F/STATUS10 = {"StatusSNS":{"Time":"2024-12-28T15:14:19","Switch1":"OFF","Switch2":"ON","Shutter1":{"Position":100,"Direction":0,"Target":100,"Tilt":0}}}
15:14:19.740 MQT: stat/tasmota_8F447F/STATUS11 = {"StatusSTS":{"Time":"2024-12-28T15:14:19","Uptime":"0T00:03:02","UptimeSec":182,"Heap":22,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":1,"POWER1":"OFF","POWER2":"OFF","Wifi":{"AP":1,"SSId":"caupona_24","BSSId":"1C:ED:6F:7B:46:60","Channel":11,"Mode":"11n","RSSI":86,"Signal":-57,"LinkCount":1,"Downtime":"0T00:00:05"}}}
15:14:19.773 MQT: stat/tasmota_8F447F/STATUS13 = {"StatusSHT":{"SHT0":{"Relay1":1,"Relay2":2,"Open":260,"Close":260,"50perc":65,"Delay":0,"Opt":"0000","Calib":[1560,364,3484,900,1000],"Mode":"1"}}}

```

### TO REPRODUCE
[... 22 lines cut ...]

### EXPECTED BEHAVIOUR

I would like the sensors to be available again like they were in the old version.

### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._

Firmware Version 14.4.1

[image]

[image]

Firmware Version 13.2.0

[image]

[image]

### ADDITIONAL CONTEXT

I can't find any other solution than downgrading to the old firmware version. Then it worked as usual again! I won't update until there is a solution

**(Please, remember to close the issue when the problem has been addressed)**

GerdRuetten, 2025-01-01:
I can confirm this issue with Shelly 2.5 and Shelly 2pm.

Position, Target, Tilt and Direction are missing unter 14.4.1
Rollback to 14.3.0 the entitys are back

I can't read any hint in the changelogs and/ore as breaking changed.

Please can somebody support it.

TIA!
Bets regards
Gerd

arendst, 2025-01-01:
See cause and solution https://github.com/arendst/Tasmota/pull/22353

stefanbode, 2025-01-02:
Happy new year and thx for bringing it up. We did some changes in the way rules work with shutters. Interestingly I operate my shutters also with HA and did not see an issue. Mine still work fine. Anyhow the discovery message is wrong. Lets fix it....

arendst, 2025-01-02:
Ah I think I know the cause of the issue. I changed the init order delaying shutter init. This may be too late for discovery. I’ll take a look how to fix this without reverting the delay (as I did it for some reason I’ve already dorgotten)

stefanbode, 2025-01-02:
I can also make a fix. I do see that TasmotaGlobal.tele_period does not count anymore from 0...teleperiod. The 0 is still working. In all other cases I got a 2. And I do see that on discovery I get a 5. Maybe something I can rely on and just add the sensors also on TasmotaGlobal.tele_period==5. In my test this recovers the issue

stefanbode, 2025-01-02:
in the esp32 line 2386 to 
        if (!ShutterGlobal.sensor_data_reported || TasmotaGlobal.tele_period != 2) {
        
 recovers it on my side. would do similar on esp8266

arendst, 2025-01-02:
Thx. I'll make the change in both files now and push to dev branch.

closed 2025-01-05 by arendst

simowilso, 2025-01-28:
Is this only for shutters? I have many Tasmota devices (mostly power monitoring plugs) that now no longer show up after restarting home assistant/mqtt. They are all running 14.4.1, I have to restart each one in order to sow up in HA.

stefanbode, 2025-01-28:
This is already fixed

simowilso, 2025-01-28:
Thanks for the response - I know the issue as been fixed (but not released) for shutters but my question is would the fix apply to other devices? If not I need to create a new bug/issue because none of my devices are being discovered after a HA/MQTT restart. (Server reboot)

stefanbode, 2025-01-29:
This was a shutter only issue because i changed the way sensor data is published and Theo changed the timing. The shutter missing in HA discovery was only for shutters. The device register correctly but the shutter was missing. This bug was never in a released version as far as I know.