arendst/Tasmota #22741: error: 'esp_gpio_revoke' was not declared in this scope

pmendiuk, 2024-12-29:
### PROBLEM DESCRIPTION
_A clear and concise description of what the problem is._
I am getting compile errors with tasmota32-base, tasmota32-zbbrdgpro, tasmota32solo1

C:/Projects/tasmota/tasmota.ino: In function 'void setup()':
C:/Projects/tasmota/tasmota.ino:421:7: error: 'esp_gpio_revoke' was not declared in th
is scope
  421 |       esp_gpio_revoke(BIT64(CONFIG_D0WD_PSRAM_CS_IO) | BIT64(CONFIG_D0WD_PSRAM_CLK_IO));
      |       ^~~~~~~~~~~~~~~

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [ x] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [x ] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [x ] Searched the problem in [discussions](https://github.com/arendst/Tasmota/discussions)
- [x ] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [x ] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [ ] Device used (e.g., Sonoff Basic): esp32dev
- [ ] Tasmota binary firmware version number used: 14.4.1
  - [ ] Pre-compiled
  - [x ] Self-compiled
- [ ] Flashing tools used: _____
- [ ] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```lua
  Configuration output here:

```
- [ ] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```lua
  Rules output here:

```
- [ ] Provide the output of this command: `Status 0`:
```lua
  STATUS 0 output here:

```
- [ ] Set `weblog` to 4 and then, when you experience your issue, provide the output of the Console log:
```lua
  Console output here:

```

### TO REPRODUCE
_Steps to reproduce the behavior:_
Build project in PlatformIO with env:tasmota32 selected

### EXPECTED BEHAVIOUR
_A clear and concise description of what you expected to happen._
Successful Build completion

### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._

### ADDITIONAL CONTEXT
_Add any other context about the problem here._

**(Please, remember to close the issue when the problem has been addressed)**

Jason2866, 2024-12-30:
CI does compile successfully https://github.com/arendst/Tasmota/actions/runs/12544524542
Check your compile environment. Something is broken there.

closed 2024-12-30 by Jason2866

Bunkerschild, 2025-02-13:
And what was the solution? Have the same Problem.

pmendiuk, 2025-02-13:
I never could get rid of that error without commenting out the line with esp_gpio_revoke() in Tasmota.ino prior to 14.4.1.  "Magically" it stopped occurring for me in 14.4.1.

barbudor, 2025-02-13:
Using today's up-to-date code base, everything compiles as expected on Windows 11

arendst, 2025-02-13:
It's closed for a reason. It failed on lower versions of IDF. Since Tasmota v14.3.0 we use IDF5.3 which supports `esp_gpio_revoke()`.