arendst/Tasmota #22742: IP config bug

on4ami, 2024-12-30:
After an IP conflict with static configured IP's (bad line copy) i tried to solve the issue with setting the IPaddress1 to 0.0.0.0 via MQTT.
This solved the issue on the first sonoff basic and received a DHCP address as expected. The second sonoff basic became unresponsive and now states he is using IP 0.0.0.0
 
Both run 14.4.1(release-tasmota)

[image]

I'm unable to reset to default as i would also reset other devices and have no physical access to the sonoff :-(

Is there a way to reset via MAC?

KR and HNY
Pat

sfromis, 2024-12-30:
No access via MAC, but if you have MQTT access, there likely is a working IP address. Anyway, after using the `IpAddress` command, you should also restart to apply changes.

More generally, what I'd suggest would be to avoid using static IP addresses in each device, to instead reserve fixed IP addresses in your DHCP server. No risk of accidentally assigning same address to two devices.

on4ami, 2024-12-30:
No, after sending the mqtt message IPaddress1 0.0.0.0 and restart i lost mqtt communication to.
I see it broadcasting with ESP_4f64b, after removing all his known SSID. When trying to connect no DHCP IP received and disconnected. Then tried to connect with a fix IP 192.168.4.2 and connection succesfull but unable to open http://192.168.4.1