arendst/Tasmota #22743: ShutterStop doesn't work correctly

DennisXK, 2024-12-30:
Hardware: ESP32 C3
Firmware: 14.4.1

I want to control the shutter with 3 buttons (Up, Stop, Down).
I do up with "ShutterButton1 1 up 1" and down with "ShutterButton1 2 down 1".
I want to stop the shutter using this script. Unfortunately, the "ShutterStop1" function doesn't work in the script. It works in the console. As soon as I configure the stop button (Switch_N 3), the shutter moves when the stop button is pressed without the script being activated.

`>D
sw3=0

>F
sw3=sw[2]
if chg[sw3]>0
then 
->ShutterStop1
endif

>W
SW3 STOP=%sw3%`

[image]

stefanbode, 2025-01-02:
I asume the main cause is in using switches instead of buttons in the configuration. Shutter only work with buttons and not witch switches. Please change in configuration and let us know if this already fixes the issue.

DennisXK, 2025-01-02:
>I asume the main cause is in using switches instead of buttons in the configuration. Shutter only work with buttons and not witch switches. Please change in configuration and let us know if this already fixes the issue.

I changed the switches to buttons. Unfortunately the shutter does not stop. But I don't know if the script is correct. In the console I get the following message "RESULT = {"Button3":{"Action":"SINGLE"}}" when I press it. But the variable "button3" is not set.

For testing, I added a web button and it works!

"
>D
button3=0
webbutton=0

>b
button3=bt[3]
if button3==1  
then  
=>ShutterStop1
endif 
 
>F
if webbutton==1  
then  
=>shutterStop1
webbutton=0
endif

>W
bu(webbutton "Stop" "Stop")
webbutton=%webbutton%
button3=%button3%
endif"