arendst/Tasmota #22756: ESP8266EX: No working WifiManager access point

probonopd, 2025-01-02:
### PROBLEM DESCRIPTION

WeMOS NodeMCU module does not open a working, visible access point.

[image]

Console says:

```
00:00:00.001 HDW: ESP8266EX
00:00:00.034 CFG: Loaded from flash at F8, Count 4
00:00:00.040 QPC: Count 1
00:00:00.053 Project tasmota - Tasmota Version 14.4.1.2(ed7b582-tasmota)-2_7_8(2025-01-02T19:53:27)
00:00:00.307 WIF: WifiManager active for 3 minutes
00:00:00.815 HTP: Web server active on tasmota-...-... with IP address 192.168.4.1
00:00:06.929 QPC: Reset
````

Access point does not show up. Also tried `wifi 4` and rebooting.

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [x] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [x] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [x] Searched the problem in [discussions](https://github.com/arendst/Tasmota/discussions)
- [x] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [ ] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [x] Device used (e.g., Sonoff Basic): WeMOS NodeMCU
- [x] Tasmota binary firmware version number used: Tasmota Version 14.4.1.2(ed7b582-tasmota)-2_7_8(2025-01-02T19:53:27)
  - [x] Pre-compiled
  - [ ] Self-compiled
- [x] Flashing tools used: https://tasmota.github.io/install/
- [x] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```lua
  Configuration output here:

00:02:39.333 CMD: Backlog Template; Module; GPIO 255
00:02:39.383 RSL: RESULT = {"NAME":"Generic","GPIO":[1,1,1,1,1,1,1,1,1,1,1,1,1,1],"FLAG":0,"BASE":18}
00:02:39.632 RSL: RESULT = {"Module":{"1":"Sonoff Basic"}}
00:02:39.888 RSL: RESULT = {"GPIO0":{"32":"Button1"},"GPIO1":{"0":"None"},"GPIO2":{"0":"None"},"GPIO3":{"0":"None"},"GPIO4":{"0":"None"},"GPIO5":{"0":"None"},"GPIO9":{"0":"None"},"GPIO10":{"0":"None"},"GPIO12":{"224":"Relay1"},"GPIO13":{"320":"Led_i1"},"GPIO14":{"0":"None"},"GPIO15":{"0":"None"},"GPIO16":{"0":"None"},"GPIO17":{"0":"None"}}
```
- [x] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```lua
  Rules output here:

n/a
```
- [x] Provide the output of this command: `Status 0`:
```lua
  STATUS 0 output here:

00:03:17.263 CMD: Status 0
00:03:17.269 RSL: STATUS = {"Status":{"Module":1,"DeviceName":"Tasmota","FriendlyName":["Tasmota"],"Topic":"tasmota_...","ButtonTopic":"0","Power":"0","PowerLock":"0","PowerOnState":3,"LedState":1,"LedMask":"FFFF","SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0,"InfoRetain":0,"StateRetain":0,"StatusRetain":0}}
00:03:17.302 RSL: STATUS1 = {"StatusPRM":{"Baudrate":115200,"SerialConfig":"8N1","GroupTopic":"tasmotas","OtaUrl":"http://ota.tasmota.com/tasmota/release/tasmota.bin.gz","RestartReason":"External System","Uptime":"0T00:03:18","StartupUTC":"","Sleep":50,"CfgHolder":4617,"BootCount":2,"BCResetTime":"1970-01-01T00:00:00","SaveCount":5,"SaveAddress":"F7000"}}
00:03:17.333 RSL: STATUS2 = {"StatusFWR":{"Version":"14.4.1.2(ed7b582-tasmota)","BuildDateTime":"2025-01-02T19:53:27","Boot":31,"Core":"2_7_8","SDK":"2.2.2-dev(38a443e)","CpuFrequency":80,"Hardware":"ESP8266EX","CR":"349/699"}}
00:03:17.353 RSL: STATUS3 = {"StatusLOG":{"SerialLog":2,"WebLog":2,"MqttLog":0,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["",""],"TelePeriod":300,"Resolution":"558180C0","SetOption":["00008009","2805C80001000600003C5A0A192800000000","00000080","00006000","00004000","00000000"]}}
00:03:17.384 RSL: STATUS4 = {"StatusMEM":{"ProgramSize":652,"Free":348,"Heap":27,"ProgramFlashSize":1024,"FlashSize":4096,"FlashChipId":"1640C8","FlashFrequency":40,"FlashMode":"DOUT","Features":["0809","8F9AC787","04368001","000000CF","010013C0","C000F981","00004004","00001000","54000020","00000080","00000000"],"Drivers":"1,2,!3,!4,!5,!6,7,!8,9,10,12,!16,!18,!19,!20,!21,!22,!24,26,!27,29,!30,!35,!37,!45,62,!68","Sensors":"1,2,3,4,5,6","I2CDriver":"7"}}
00:03:17.418 RSL: STATUS5 = {"StatusNET":{"Hostname":"tasmota-B55C74-7284","IPAddress":"0.0.0.0","Gateway":"192.168.1.1","Subnetmask":"255.255.255.0","DNSServer1":"192.168.1.1","DNSServer2":"0.0.0.0","Mac":"5C:CF:7F:B5:5C:74","Webserver":2,"HTTP_API":1,"WifiConfig":4,"WifiPower":17.0}}
00:03:17.442 RSL: STATUS6 = {"StatusMQT":{"MqttHost":"","MqttPort":1883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_B55C74","MqttUser":"DVES_USER","MqttCount":0,"MAX_PACKET_SIZE":1200,"KEEPALIVE":30,"SOCKET_TIMEOUT":4}}
00:03:17.464 RSL: STATUS7 = {"StatusTIM":{"UTC":"1970-01-01T00:03:17Z","Local":"1970-01-01T00:03:17","StartDST":"1970-01-01T00:00:00","EndDST":"1970-01-01T00:00:00","Timezone":"+00:00","Sunrise":"20:13","Sunset":"05:47"}}
00:03:17.482 RSL: STATUS10 = {"StatusSNS":{"Time":"1970-01-01T00:03:17"}}
00:03:17.488 RSL: STATUS11 = {"StatusSTS":{"Time":"1970-01-01T00:03:17","Uptime":"0T00:03:18","UptimeSec":198,"Heap":27,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":0,"POWER":"OFF"}}
```
- [x] Set `weblog` to 4 and then, when you experience your issue, provide the output of the Console log:
```lua
  Console output here:

n/a
```

### TO REPRODUCE

* Flash firmware onto device erasing settings

### EXPECTED BEHAVIOUR

* A working WifiManager access point shows up on the mobile phone

### SCREENSHOTS

n/a

### ADDITIONAL CONTEXT

A known good version that works reliably on this module is Tasmota 9.2.0

probonopd, 2025-01-02:
Additional information - not sure if in any way related to the above, but when trying to find a version which works with the same hardware I found the following.

Flashing Tasmota 9.5.0 directly (erasing the settings) results in a working WifiManager, but no connection to real WLAN is possible:

```
00:00:00.001 HDW: ESP8266EX
00:00:00.030 CFG: Loaded from flash at F9, Count 3
00:00:00.036 QPC: Count 1
00:00:00.043 Project tasmota Tasmota Version 9.5.0(display)-2_7_4_9(2021-06-17T08:26:42)
00:00:00.194 WIF: WifiManager active for 3 minutes
00:00:00.701 HTP: Web server active on tasmota_ ...-... with IP address 192.168.4.1
00:00:06.964 QPC: Reset
00:00:31.771 WIF: Connecting to AP ... as tasmota_ ...-... ...
00:00:39.897 WIF: Connect failed as AP cannot be reached
```

When one flashes 9.2.0 (erasing the settings) first leads to a working connection to real WLAN:

```
00:00:00 CFG: Loaded from flash at F9, Count 3
00:00:00 QPC: Count 1
00:00:00 Project tasmota Tasmota Version 9.2.0(display)-2_7_4_9(2020-12-21T15:05:26)
00:00:00 WIF: WifiManager active for 3 minutes
00:00:00 HTP: Web server active on tasmota_ ...-... with IP address 192.168.4.1
00:00:06 QPC: Reset
00:00:43 WIF: Hostname %s-%04d, SSId1 ..., SSId2 , CORS 
00:00:45 APP: Restarting

 ets Jan  8 2013,rst cause:2, boot mode:(3,6)

load 0x4010f000, len 3584, room 16 
tail 0
chksum 0xb0
csum 0xb0
v3969889e
~ld

00:00:00 CFG: Loaded from flash at F7, Count 5
00:00:00 QPC: Count 1
00:00:00 Project tasmota Tasmota Version 9.2.0(display)-2_7_4_9(2020-12-21T15:05:26)
00:00:00 WIF: Connecting to AP1 ... in mode 11n as tasmota_ ...-......
00:00:05 WIF: Connected
00:00:05 HTP: Web server active on tasmota_ ...-... with IP address 192.168.0.115
22:36:41 RSL: tele/tasmota_.../INFO1 = {"Module":"Sonoff Basic","Version":"9.2.0(display)","FallbackTopic":"cmnd/DVES_..._fb/","GroupTopic":"cmnd/tasmotas/"}
22:36:41 RSL: tele/tasmota_.../INFO2 = {"WebServerMode":"Admin","Hostname":"tasmota_ ...-...","IPAddress":"192.168.0.115"}
22:36:41 RSL: tele/tasmota_.../INFO3 = {"RestartReason":"Software/System restart"}
22:36:41 RSL: stat/tasmota_.../RESULT = {"POWER":"OFF"}
22:36:41 RSL: stat/tasmota_.../POWER = OFF
22:36:41 QPC: Reset
22:36:45 RSL: tele/tasmota_.../STATE = {"Time":"2025-01-02T22:36:45","Uptime":"0T00:00:12","UptimeSec":12,"Heap":31,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":0,"POWER":"OFF","Wifi":{"AP":1,"SSId":"...","BSSId":"...","Channel":11,"RSSI":56,"Signal":-72,"LinkCount":1,"Downtime":"0T00:00:06"}}
```

When one then flashes 9.5.0 WITHOUT ERASING, then it STILL works:

```
00:00:00.002 HDW: ESP8266EX
00:00:00.046 CFG: Loaded from flash at FA, Count 9
00:00:00.052 QPC: Count 1
00:00:00.059 Project tasmota Tasmota Version 9.5.0(display)-2_7_4_9(2021-06-17T08:26:42)
00:00:00.546 WIF: Connecting to AP1 ... Channel 11 BSSId ... in mode 11n as tasmota_ ...-......
00:00:01.753 WIF: Connected
00:00:02.004 HTP: Web server active on tasmota_ ...-... with IP address 192.168.0.115
22:39:03.006 RSL: INFO1 = {"Info1":{"Module":"Sonoff Basic","Version":"9.5.0(display)","FallbackTopic":"cmnd/DVES_..._fb/","GroupTopic":"cmnd/tasmotas/"}}
22:39:03.011 RSL: INFO2 = {"Info2":{"WebServerMode":"Admin","Hostname":"tasmota_ ...-...","IPAddress":"192.168.0.115"}}
22:39:03.022 RSL: INFO3 = {"Info3":{"RestartReason":"External System"}}
22:39:03.028 RSL: RESULT = {"POWER":"OFF"}
22:39:03.032 RSL: POWER = OFF
22:39:06.443 QPC: Reset
22:39:07.407 RSL: STATE = {"Time":"2025-01-02T22:39:07","Uptime":"0T00:00:09","UptimeSec":9,"Heap":31,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":0,"POWER":"OFF","Wifi":{"AP":1,"SSId":"...","BSSId":"...","Channel":11,"Mode":"11n","RSSI":56,"Signal":-72,"LinkCount":1,"Downtime":"0T00:00:03"}}
```

arendst, 2025-01-03:
No problem at all:
```
12:10:33.672 CMD: wificonfig
12:10:33.674 SRC: Serial
12:10:33.678 CMD: Grp 0, Cmd 'WIFICONFIG', Idx 1, Len 0, Pld -99, Data ''
12:10:33.687 MQT: stat/tasmota/RESULT = {"WifiConfig":{"4":"Retry"}}
12:10:39.403 CMD: wificonfig 1
12:10:39.405 SRC: Serial
12:10:39.408 CMD: Grp 0, Cmd 'WIFICONFIG', Idx 1, Len 1, Pld 1, Data '1'
12:10:39.417 MQT: stat/tasmota/RESULT = {"WifiConfig":{"2":"WifiManager"}}
12:10:40.241 CFG: Saved to flash at FD, Count 10, Bytes 4096
12:10:40.398 WIF: WifiManager active for 3 minutes
12:10:40.505 WIF: Wifimanager set AccessPoint and keep Station
12:10:41.011 HTP: Web server active on tasmota-1460 with IP address 192.168.4.1
12:11:20.337 HTP: Redirected to captive portal
12:11:21.741 HTP: WiFi
12:11:21.769 WIF: Scanning...
12:11:23.974 WIF: Scan done
12:11:24.153 HTP: Redirected to captive portal
12:11:24.306 HTP: Redirected to captive portal
12:11:59.113 HTP: WiFi
12:11:59.115 CFG: CR 377/699, Busy 0
12:11:59.117 CFG: CR 377/699, Busy 0
12:11:59.119 WIF: Connecting to AP indebuurt_IoT as tasmota-1460 ...
12:11:59.219 WIF: Trying to connect device to network indebuurt_IoT
12:11:59.289 HTP: Redirected to captive portal
12:12:00.219 WIF: Trying to connect device to network indebuurt_IoT
12:12:00.515 HTP: Redirected to captive portal
12:12:01.220 WIF: Trying to connect device to network indebuurt_IoT
12:12:02.003 WIF: Trying to connect device to network indebuurt_IoT
12:12:02.613 WIF: Trying to connect device to network indebuurt_IoT
12:12:02.615 WIF: SSId indebuurt_IoT: Connected - IP Address 192.168.2.136
12:12:08.624 HTP: WiFi
12:12:08.654 HTP: Restart
12:12:08.656 SRC: WebGui from 192.168.4.2
12:12:08.693 WIF: Scanning...
12:12:10.303 WIF: Scan done
12:12:10.319 HTP: Redirected to captive portal
12:12:10.520 HTP: Redirected to captive portal
12:12:11.564 APP: Restarting

 ets Jan  8 2013,rst cause:2, boot mode:(3,6)

load 0x4010f000, len 3584, room 16 
tail 0
chksum 0xb0
csum 0xb0
v3969889e
~ld

00:00:00.001 HDW: ESP8266EX
00:00:00.010 UFS: FlashFS mounted with 1976 kB free
00:00:00.060 CFG: Loaded from File, Count 10
00:00:00.065 QPC: Count 1
00:00:00.067 CFG: CR 377/699, Busy 0
00:00:00.074 ROT: Mode 1
00:00:00.076 SRC: Restart
00:00:00.078 Project tasmota - Tasmota Version 14.4.1.1(theo)-2_7_8(2024-12-15T15:44:38)
00:00:00.181 WIF: Attempting connection...
00:00:01.001 WIF: Connecting to AP1 indebuurt_IoT Channel 11 BSSId 18:E8:29:CA:17:C1 in mode 11n as tasmota-1460...
00:00:02.512 WIF: Connected
00:00:02.765 HTP: Web server active on tasmota-1460 with IP address 192.168.2.136
00:00:04.288 NTP: No reply from 127.0.1.1
00:00:04.315 RTC: UTC 2025-01-03T11:12:18Z, DST 2025-03-30T02:00:00, STD 2025-10-26T03:00:00
12:12:18.001 RTC: Synced by NTP
12:12:18.226 MQT: Attempting connection...
12:12:18.239 MQT: Connected
12:12:18.242 MQT: tele/tasmota/LWT = Online (retained)
12:12:18.244 MQT: cmnd/tasmota/POWER = 
12:12:18.246 MQT: Subscribe to cmnd/tasmota/#
12:12:18.248 MQT: Subscribe to cmnd/tasmotas/#
12:12:18.249 MQT: Subscribe to cmnd/DVES_4765B4_fb/#
12:12:18.256 MQT: tele/tasmota/INFO1 = {"Info1":{"Module":"Sonoff Basic","Version":"14.4.1.1(theo)","FallbackTopic":"cmnd/DVES_4765B4_fb/","GroupTopic":"cmnd/tasmotas/"}}
12:12:18.270 MQT: tele/tasmota/INFO2 = {"Info2":{"WebServerMode":"Admin","Hostname":"tasmota-1460","IPAddress":"192.168.2.136"}}
12:12:18.282 MQT: tele/tasmota/INFO3 = {"Info3":{"RestartReason":"Software/System restart","BootCount":4}}
12:12:18.292 MQT: stat/tasmota/RESULT = {"POWER":"OFF"}
12:12:18.297 MQT: stat/tasmota/POWER = OFF
12:12:19.267 QPC: Reset
12:12:21.224 APP: Boot Count 4
12:12:22.266 CFG: Saved to flash at 3FB, Count 11, Bytes 4096
12:12:22.275 MQT: tele/tasmota/STATE = {"Time":"2025-01-03T12:12:22","Uptime":"0T00:00:10","UptimeSec":10,"Heap":24,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":1,"POWER":"OFF","Wifi":{"AP":1,"SSId":"indebuurt_IoT","BSSId":"18:E8:29:CA:17:C1","Channel":11,"Mode":"11n","RSSI":100,"Signal":-30,"LinkCount":1,"Downtime":"0T00:00:04"}}
```

Frogmore42, 2025-01-03:
What @arendst is saying is this is very basic functionality of the project that has worked for MANY people and worked in the test he did.  This means it is likely something in the way you are doing it.  Use the chat function to get help (that is one of the steps in the template that you skipped).