arendst/Tasmota #22765: Unable to flash new (SonoffZBPro_coord_20240710.hex) zigbee coordinator firmware on Sonoff Zigbee Bridge Pro

spamik, 2025-01-04:
### PROBLEM DESCRIPTION
Flashing zigbee coordinator firmware [SonoffZBPro_coord_20240710.hex](https://github.com/arendst/Tasmota/blob/development/tools/fw_SonoffZigbeeBridgePro_cc2652/SonoffZBPro_coord_20240710.hex) from this repository fails with error invalid size for line.

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [ x] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [ x] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [ x] Searched the problem in [discussions](https://github.com/arendst/Tasmota/discussions)
- [ x] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [ x] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [ x] Device used (e.g., Sonoff Basic): Sonoff Zigbee Bridge Pro
- [ x] Tasmota binary firmware version number used: 14.4.1
  - [ x] Pre-compiled
  - [  ] Self-compiled
- [ x] Flashing tools used: esptool
- [x ] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```lua
  Configuration output here:
15:28:46.112 MQT: stat/tasmota_bridge1/RESULT = {"NAME":"Sonoff Zigbee Pro","GPIO":[0,0,576,0,480,0,0,0,0,1,1,5792,0,0,0,3552,0,320,5793,3584,0,640,608,32,0,0,0,0,0,1,0,0,0,0,0,0],"FLAG":0,"BASE":1}
15:28:46.334 MQT: tele/tasmota_bridge1/RESULT = {"ZbState":{"Status":99,"Message":"Abort"}}
15:28:46.340 MQT: stat/tasmota_bridge1/RESULT = {"Module":{"0":"Sonoff Zigbee Pro"}}
15:28:46.594 MQT: stat/tasmota_bridge1/RESULT = {"GPIO0":{"0":"None"},"GPIO1":{"0":"None"},"GPIO2":{"576":"LedLink_i"},"GPIO3":{"0":"None"},"GPIO4":{"480":"Buzzer"},"GPIO5":{"0":"None"},"GPIO6":{"0":"None"},"GPIO7":{"0":"None"},"GPIO8":{"0":"None"},"GPIO9":{"0":"None"},"GPIO10":{"0":"None"},"GPIO11":{"0":"None"},"GPIO12":{"0":"None"},"GPIO13":{"0":"None"},"GPIO14":{"0":"None"},"GPIO15":{"5792":"Zigbee Rst1"},"GPIO16":{"0":"None"},"GPIO17":{"0":"None"},"GPIO18":{"0":"None"},"GPIO19":{"3552":"Zigbee Tx"},"GPIO20":{"0":"None"},"GPIO21":{"320":"Led_i1"},"GPIO22":{"5793":"Zigbee Rst2"},"GPIO23":{"3584":"Zigbee Rx"},"GPIO24":{"0":"None"},"GPIO25":{"640":"I2C SDA1"},"GPIO26":{"608":"I2C SCL1"},"GPIO27":{"32":"Button1"},"GPIO32":{"0":"None"},"GPIO33":{"0":"None"},"GPIO34":{"0":"None"},"GPIO35":{"0":"None"},"GPIO36":{"0":"None"},"GPIO37":{"0":"None"},"GPIO38":{"0":"None"},"GPIO39":{"0":"None"}}
```

### TO REPRODUCE
Upload  [SonoffZBPro_coord_20240710.hex](https://github.com/arendst/Tasmota/blob/development/tools/fw_SonoffZigbeeBridgePro_cc2652/SonoffZBPro_coord_20240710.hex) to ESP filesystem and run in berry console:

```
import sonoff_zb_pro_flasher as cc
cc.load("SonoffZBPro_coord_20240710.hex")
cc.check()
```

### EXPECTED BEHAVIOUR
VerificationOK, not error

### ADDITIONAL CONTEXT
From my research, main deference between old file (SonoffZBPro_coord_20220219.hex) which works ok is, that SonoffZBPro_coord_20220219.hex uses \n (LF) for new lines, and SonoffZBPro_coord_20240710.hex uses \n\r (LFCR) for new line. And in intelhex.be file this section will be the issue:

```
      var line = readline(self_f)   # self.f.readline()
      if line[-1] == '\n'   line = line[0..-2]  end
      if line[-1] == '\r'   line = line[0..-2]  end
```

As in newer firmware lines are longer about 1 byte (CR). So with \n line endings this will remove from line line ending and checksum byte. But for LFCR checksum byte stays in place and it's calculated with data for length which cause mismatch.

But probably this is not the only issue. Even after converting hex file to LF line endings it fails with error unsupported record_type: 2. Not sure why it's happening, in newer firmware is really record type 2 present near end of file but it should pass this check...

debsahu, 2025-01-05:
Did you update the berry scripts as well? https://github.com/arendst/Tasmota/tree/development/tasmota/berry/zigbee

spamik, 2025-01-05:
Well, I did, but I probably didn't restart module after new files upload which seems that this was the issue... with new scripts and restarts it seems working.

closed 2025-01-05 by spamik

ThorPrez, 2025-04-02:
I´ve to reopen this case.
What was the fault ?
Because I´m trying to flash the current files and it´s not working.
Sometimes no response, sometimes the message "exeption raised protocol error".
I´m using these commands:
import sonoff_zb_pro_flasher as cc
cc.load("SonoffZBPro_coord_20240710.hex")
cc.check()
cc.flash()
After confirming "cc.check()" nothing happens.

s-hadinger, 2025-04-03:
Please send complete logs, not a fragment on the error.

ThorPrez, 2025-04-03:
Sorry, the berry console isn´t giving any feedback.
After confirming "cc.check()" the page is stucking and after a few seconds the Web GUI isn´t reachable.
I´ve to replug the device.
On a second window with console open I´m getting this:

```
12:25:36.878 FLH: Starting verification of HEX file
12:25:56.695 FLH: Verification of HEX file OK
12:27:45.881 FLH: cc2652_flasher rx=23 tx=19 rst=15 bsl=22
12:27:45.894 FLH: Flashing started (takes 5-8 minutes during which Tasmota is unresponsive)
12:35:55.163 FLH: Flashing completed: OK
12:35:55.270 FLH: Flash crc32 0x000000 - 0x2FFFF = bytes('D0D7B0AA')
12:40:28.324 APP: Serial logging disabled
```

Tasmota version 14.5.0.3
ZBBridge-P version 1.1, 2021.11.12

ThorPrez, 2025-04-03:
I´ve tried it with two open windows for the first time and haven´t seen before that it was working.
Because Berry console still shows nothing !

spamik, 2025-04-04:
In my case there were probably issue that after uploading new berry scripts it was needed to restart the device. Having opened web browser window more ten 1 time is also probably not a good idea - I've seen some strange behaviour when I have opened it from two compusters. And yep - console shows nothing until it's finished.

ThorPrez, 2025-04-04:
When flashing two other devices months ago I had no problems and Berry console was showing status when flashing.
But it was Tasmota version 14.0.0.
That´s why I was confused.