arendst/Tasmota #22773: GPS Module, NEO-6M not disolaying latitude in Web UI or Tele messages

lengross, 2025-01-06:
### PROBLEM DESCRIPTION
I have attached this GPS Module   [](https://www.amazon.com/dp/B0B31NRSD2?ref=ppx_yo2ov_dt_b_fed_asin_title)

 to a NodeMCU ESP8266    With a debug version I can see it is correctly displaying the lat /long data in the console, but the lat value does not show up in the Web UI or the Tele message

The module is using Ublox7 protocol and data correctly shows on the u-center app.

Tele message:

18:44:26.806 MQT: tele/tasmota_679696/SENSOR = {"Time":"2025-01-05T18:44:26","GPS":{"Lat":,"Lon":-117.2653867,"Alt":22.478,"hAcc":15.530,"vAcc":13.272,"Fix":"3D"}}
Notice blank latitude in Web UI

[image]

But correct lat in the console debug window:

18:32:28.525 UBX: lat/lon: 328183580 / -1172654815
18:32:28.550 UBX: hAcc: 18378
18:32:28.644 UBX: got NAV_TIME_UTC
18:32:28.647 UBX: UTC-Time: 2025-1-6 2:32:31
18:32:28.668 UBX: UTC-Time is valid
18:32:29.458 UBX: got NAV_STATUS
18:32:29.489 UBX: gpsFix: 3, valid: 1
18:32:29.518 UBX: got NAV_POSLLH
18:32:29.520 UBX: iTOW: 95570000
18:32:29.523 UBX: lat/lon: 328183560 / -1172654829
18:32:29.532 UBX: hAcc: 18365
18:32:29.625 UBX: got NAV_TIME_UTC
18:32:29.628 UBX: UTC-Time: 2025-1-6 2:32:32
18:32:29.649 UBX: UTC-Time is valid
18:32:30.512 UBX: got NAV_STATUS
18:32:30.514 UBX: gpsFix: 3, valid: 1

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [x ] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [ x] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [ x] Searched the problem in [discussions](https://github.com/arendst/Tasmota/discussions)
- [ x] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [ ] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [ x] Device used (e.g., Sonoff Basic): _____ NodeMcu V3 Esp8266
- [ ] Tasmota binary firmware version number used: 
[image]

compiled with tasmoCompiler  using 14.1.1

  - [ ] Pre-compiled
  - [ x] Self-compiled
- [ ] Flashing tools used: Tasmotizer
- [ ] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```lua
  Configuration output here:
18:21:15.863 CMD: Backlog Template; Module; GPIO 255
18:21:15.937 MQT: stat/tasmota_679696/RESULT = {"NAME":"GPS Clock","GPIO":[1,1,1,1,4384,4352,1,1,6624,6656,1,1,1,1],"FLAG":0,"BASE":18}
18:21:16.181 MQT: stat/tasmota_679696/RESULT = {"Module":{"0":"GPS Clock"}}
18:21:16.507 MQT: stat/tasmota_679696/RESULT = {"GPIO0":{"0":"None"},"GPIO1":{"0":"None"},"GPIO2":{"0":"None"},"GPIO3":{"0":"None"},"GPIO4":{"4384":"GPS TX"},"GPIO5":{"4352":"GPS RX"},"GPIO9":{"0":"None"},"GPIO10":{"0":"None"},"GPIO12":{"6624":"TM1637 CLK"},"GPIO13":{"6656":"TM1637 DIO"},"GPIO14":{"0":"None"},"GPIO15":{"0":"None"},"GPIO16":{"0":"None"},"GPIO17":{"0":"None"}}

```
- [ ] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```lua
  Rules output here:
No rules required

```
- [ ] Provide the output of this command: `Status 0`:
```lua
  STATUS 0 output here:

18:53:20.456 CMD: status 0
18:53:20.469 MQT: stat/tasmota_679696/STATUS = {"Status":{"Module":0,"DeviceName":"Tasmota","FriendlyName":["GPS clcok"],"Topic":"tasmota_679696","ButtonTopic":"0","Power":"1","PowerLock":"0","PowerOnState":3,"LedState":1,"LedMask":"FFFF","SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0,"InfoRetain":0,"StateRetain":0,"StatusRetain":0}}
18:53:20.871 MQT: stat/tasmota_679696/STATUS1 = {"StatusPRM":{"Baudrate":9600,"SerialConfig":"8N1","GroupTopic":"tasmotas","OtaUrl":"http://ota.tasmota.com/tasmota/release/tasmota-lite.bin.gz","RestartReason":"Software Watchdog","Uptime":"0T00:00:31","StartupUTC":"2025-01-06T02:52:49","Sleep":50,"CfgHolder":4617,"BootCount":56,"BCResetTime":"2025-01-05T00:12:40","SaveCount":94,"SaveAddress":"F9000"}}
18:53:21.007 MQT: stat/tasmota_679696/STATUS2 = {"StatusFWR":{"Version":"14.4.1(TasmoCompiler-esp8266generic)","BuildDateTime":"2025-01-04T18:48:38","Boot":31,"Core":"2_7_8","SDK":"2.2.2-dev(38a443e)","CpuFrequency":80,"Hardware":"ESP8266EX","CR":"359/699"}}
18:53:21.278 MQT: stat/tasmota_679696/STATUS3 = {"StatusLOG":{"SerialLog":2,"WebLog":2,"MqttLog":0,"SysLog":4,"LogHost":"","LogPort":514,"SSId":["102",""],"TelePeriod":300,"Resolution":"558180C0","SetOption":["00008009","2805C80001000600003C5A0A192800000000","00000080","00006000","00004000","00000000"]}}
18:53:21.603 MQT: stat/tasmota_679696/STATUS4 = {"StatusMEM":{"ProgramSize":542,"Free":460,"Heap":28,"ProgramFlashSize":4096,"FlashSize":4096,"FlashChipId":"164054","FlashFrequency":40,"FlashMode":"DOUT","Features":["0809","0F804582","04260001","00000003","00000000","C040E881","00000004","00001000","54000000","00000080","00000000"],"Drivers":"1,2,!4,!6,!8,9,10,12,!18,!19,!22,!24,26,29,!30,!35,!37,!45,62,!68","Sensors":"1,2,60"}}
18:53:22.010 MQT: stat/tasmota_679696/STATUS5 = {"StatusNET":{"Hostname":"tasmota-679696-5782","IPAddress":"192.168.1.152","Gateway":"192.168.1.1","Subnetmask":"255.255.255.0","DNSServer1":"209.18.47.63","DNSServer2":"209.18.47.62","Mac":"08:F9:E0:67:96:96","Webserver":2,"HTTP_API":1,"WifiConfig":4,"WifiPower":17.0}}
18:53:22.371 MQT: stat/tasmota_679696/STATUS6 = {"StatusMQT":{"MqttHost":"192.168.1.109","MqttPort":1883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_679696","MqttUser":"DVES_USER","MqttCount":1,"MAX_PACKET_SIZE":1200,"KEEPALIVE":30,"SOCKET_TIMEOUT":4}}
18:53:22.608 MQT: stat/tasmota_679696/STATUS7 = {"StatusTIM":{"UTC":"2025-01-06T02:53:22Z","Local":"2025-01-05T18:53:22","StartDST":"2025-03-09T02:00:00","EndDST":"2025-11-02T02:00:00","Timezone":99,"Sunrise":"06:58","Sunset":"16:56"}}
18:53:23.007 MQT: stat/tasmota_679696/STATUS10 = {"StatusSNS":{"Time":"2025-01-05T18:53:23","GPS":{"Lat":,"Lon":-117.2655403,"Alt":44.252,"hAcc":21.355,"vAcc":17.023,"Fix":"3D"}}}
18:53:23.195 MQT: stat/tasmota_679696/STATUS11 = {"StatusSTS":{"Time":"2025-01-05T18:53:23","Uptime":"0T00:00:34","UptimeSec":34,"Heap":24,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":1,"Wifi":{"AP":1,"SSId":"102","BSSId":"C0:C1:C0:CF:7C:2A","Channel":11,"Mode":"11n","RSSI":82,"Signal":-59,"LinkCount":1,"Downtime":"0T00:00:03"}}}
18:53:23.555 MQT: stat/tasmota_679696/STATUS12 = {"StatusSTK":{"Exception":4,"Reason":"Software Watchdog","EPC":["4023ecc1","00000000","00000000"],"EXCVADDR":"00000000","DEPC":"00000000"}}

```
- [ ] Set `weblog` to 4 and then, when you experience your issue, provide the output of the Console log:
```lua
Note the  Lat/long entry with non-blank lat.  (Value are correct BTW)
  Console output here:
18:32:28.525 UBX: lat/lon: 328183580 / -1172654815
18:32:28.550 UBX: hAcc: 18378
18:32:28.644 UBX: got NAV_TIME_UTC
18:32:28.647 UBX: UTC-Time: 2025-1-6 2:32:31
18:32:28.668 UBX: UTC-Time is valid
18:32:29.458 UBX: got NAV_STATUS
18:32:29.489 UBX: gpsFix: 3, valid: 1
18:32:29.518 UBX: got NAV_POSLLH
18:32:29.520 UBX: iTOW: 95570000
18:32:29.523 UBX: lat/lon: 328183560 / -1172654829
18:32:29.532 UBX: hAcc: 18365
18:32:29.625 UBX: got NAV_TIME_UTC
18:32:29.628 UBX: UTC-Time: 2025-1-6 2:32:32
18:32:29.649 UBX: UTC-Time is valid
18:32:30.512 UBX: got NAV_STATUS
18:32:30.514 UBX: gpsFix: 3, valid: 1

```

### TO REPRODUCE
_Steps to reproduce the behavior:_
Once the GPS module LED starts to flash, the module has a lat/long fix and the values should show up on the WebUI or the Tele messages, but they are not showing up there.

### EXPECTED BEHAVIOUR
_A clear and concise description of what you expected to happen._
The Lat should be filled in in the WebUI and the Tele Messages.

### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._

### ADDITIONAL CONTEXT
_Add any other context about the problem here._

Since the lat shows up correctly in the console when debug is enabled, but does not show up in the WebUI or the Tele message, it appears like it is something like a "parsing" problem.  It is possible that this is something specific to Ublox 7 protocol, but that seems unlikley since everything else seems to be right.
I understand you may not have this exact hardware, but the problem may be "obvious" from eyeballing the relevant code.   I am not skilled enough to do that, but if there is someway I can help (e.g. if you create a special debug version) I am certainly willing to do that.

Thanks for killer system!

**(Please, remember to close the issue when the problem has been addressed)**

lengross, 2025-01-06:
Looks like it lost the link to the GPS module I am using;
https://www.amazon.com/dp/B0B31NRSD2?ref=ppx_yo2ov_dt_b_fed_asin_title

lengross, 2025-01-10:
Solution is to change precision value in calls to dtostrfd from 7 to 6.
Will try to generate a Pull request,

sfromis, 2025-01-10:
If reducing number of decimals *appears* to solve the issue, it would more likely be relevant to change the length of the result strings from 12 to 13, due to null-terminated strings needing one extra byte. OTOH, the value of having the 7th decimal may already be "questionable".

lengross, 2025-01-11:
I just verified that changing char[12] to char[13] in 4 places right before the dtostrfd  calls does work correctly, I thought I had tried that  :(  
It's better to let the end user determine how much accuracy is valid, so that is a superior soluton.
I tried to make a pull request with the precision change, but It doesn't seem to have propagated (It's my first try).
Can someone make the changes that you suggested?
Thanks.

sfromis, 2025-01-11:
My experience setting up a PR is quite limited, but I've found it to be reasonably straightforward without doing it the "proper" way.

Only using a browser, my steps has been to simply go into the [source file](https://github.com/arendst/Tasmota/blob/development/tasmota/tasmota_xsns_sensor/xsns_60_GPS.ino) in the development branch of the repository, and do the changes after hitting the "edit" button (pencil icon). It may suggest to create a fork, if you do not already have one (and without permission to update directly). With changes done, hit the green commit button, and explain the change. Next you can create a PR and submit it. After it is submitted, tick the appropriate check boxes on the first page of the thread for the PR (do not edit the raw form lines with the check boxes before submitting).

My words about "proper way" is due to this "entry-level" process for very small changes not directly including build and test steps, presumed to already have been done by other means.

arendst, 2025-01-11:
Best solution is changing 12 to FLOATSZ

FLOATSZ is defined by tasmota as 16 and used everytime a float is needed

lengross, 2025-01-11:
I verified using FLOATSZ solves the issue,  and it appears the PR was successfully entered. (Thanks to sfromis for  the "entry-level" approach for a  PR.)

PR #22804 (2025-01-12): Update xsns_60_GPS.ino
Using FLOATSZ instead of 12 or 13 in lat/lon calls to dtostrfd

## Description:

**Related issue (if applicable):** fixes #22773

## Checklist:
  - [x ] The pull request is done against the latest development branch
  - [x] Only relevant files were touched
  - [x] Only one feature/fix was added per PR and the code change compiles without warnings
  - [x ] The code change is tested and works with Tasmota core ESP8266 V.2.7.8
  - [ ] The code change is tested and works with Tasmota core ESP32 V.3.1.1.250109
  - [x] I accept the [CLA](https://github.com/arendst/Tasmota/blob [... cut]
closed 2025-01-12 by arendst