arendst/Tasmota #22821: Sometimes wrong result with float add in script (x+0.01=>x instead of correct value)

dideldum, 2025-01-15:
### PROBLEM DESCRIPTION
The ticks from a reed counter are added to a float value showing the current gas counter value. The ticks of the gas counter are in the second digit after the comma. Thus the number of ticks since the last call (usually 1 tick in the >F section) are divided by 100 and added to the last counter value. Sometimes the result is not (as expected) a value with the correct added 0.01, but the value stays the same. See the following log for counter value 26096.71, which at 07:39:05.854 stays the same though 0.01 should be added.
As the counter values have only 5 digits before the floating point and 2 digits after, I wouldn't expect strange rounding problems with the float.

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [X] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [X] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [X] Searched the problem in [discussions](https://github.com/arendst/Tasmota/discussions)
- [X] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [ ] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [X] Device used (e.g., Sonoff Basic): NodeMCU V2 WiFi Amica ESP3266
- [ ] Tasmota binary firmware version number used: ___
  - [ ] Pre-compiled
  - [X] Self-compiled (14.4.1)
- [X] Flashing tools used: PlatformIO
- [ ] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```lua
  Configuration output here:

```
- [ ] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```lua
  Rules output here:

```
- [X] Provide the output of this command: `Status 0`:
08:19:47.812 CMD: status 0
08:19:47.820 RSL: STATUS = {"Status":{"Module":0,"DeviceName":"Gaszaehler","FriendlyName":["Gaszaehler"],"Topic":"gaszaehler","ButtonTopic":"0","Power":"0","PowerLock":"0","PowerOnState":3,"LedState":1,"LedMask":"FFFF","SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0,"InfoRetain":0,"StateRetain":0,"StatusRetain":0}}
08:19:47.825 RSL: STATUS1 = {"StatusPRM":{"Baudrate":115200,"SerialConfig":"8N1","GroupTopic":"tasmotas","OtaUrl":"http://pi.dedio.de/tasmota/tasmota-ulf-reed-smartreader-with-script.bin.gz","RestartReason":"Software/System restart","Uptime":"0T00:54:24","StartupUTC":"2025-01-15T06:25:23","Sleep":50,"CfgHolder":4617,"BootCount":9,"BCResetTime":"2025-01-14T07:35:26","SaveCount":47,"SaveAddress":"FD000"}}
08:19:47.829 RSL: STATUS2 = {"StatusFWR":{"Version":"14.4.1(tasmota)","BuildDateTime":"2025.01.13 13:07:14","Boot":31,"Core":"2_7_8","SDK":"2.2.2-dev(38a443e)","CpuFrequency":80,"Hardware":"ESP8266EX","CR":"428/699"}}
08:19:47.833 RSL: STATUS3 = {"StatusLOG":{"SerialLog":0,"WebLog":2,"MqttLog":0,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["DDOBFB1","isecklei"],"TelePeriod":0,"Resolution":"558180C0","SetOption":["00008001","2805C80001000600003C5A0A192800000000","00000080","00006000","00004000","00000000"]}}
08:19:47.847 RSL: STATUS4 = {"StatusMEM":{"ProgramSize":753,"Free":1292,"Heap":13,"ProgramFlashSize":4096,"FlashSize":4096,"FlashChipId":"164054","FlashFrequency":40,"FlashMode":"DOUT","Features":["0407","878AC78F","043E8001","000000CF","010013C0","C000F981","00004084","00201000","54000020","00000080","00000000"],"Drivers":"1,!2,!3,!4,!5,!6,!8,9,10,!12,!16,!18,!19,!20,!21,!22,!24,26,!27,29,!30,!35,!37,38,!45,50,62,!68","Sensors":"1,2,3,4,5,6","I2CDriver":"7"}}
08:19:47.852 RSL: STATUS5 = {"StatusNET":{"Hostname":"Gaszaehler","IPAddress":"192.168.178.42","Gateway":"192.168.178.1","Subnetmask":"255.255.255.0","DNSServer1":"192.168.178.9","DNSServer2":"0.0.0.0","Mac":"C8:C9:A3:0E:53:02","Webserver":2,"HTTP_API":1,"WifiConfig":4,"WifiPower":17.0}}
08:19:47.858 RSL: STATUS7 = {"StatusTIM":{"UTC":"2025-01-15T07:19:47Z","Local":"2025-01-15T08:19:47","StartDST":"2025-03-30T02:00:00","EndDST":"2025-10-26T03:00:00","Timezone":99,"Sunrise":"08:24","Sunset":"16:34"}}
08:19:47.862 RSL: STATUS10 = {"StatusSNS":{"Time":"2025-01-15T08:19:47","COUNTER":{"C1":58}}}
08:19:47.868 RSL: STATUS11 = {"StatusSTS":{"Time":"2025-01-15T08:19:47","Uptime":"0T00:54:24","UptimeSec":3264,"Heap":13,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":0,"Wifi":{"AP":1,"SSId":"DDOBFB1","BSSId":"5C:49:79:9B:AD:6A","Channel":11,"Mode":"11n","RSSI":66,"Signal":-67,"LinkCount":1,"Downtime":"0T00:00:05"}}}

```
- [X ] Set `weblog` to 4 and then, when you experience your issue, provide the output of the Console log:
07:35:35.860 GPIO14: 17.00, Ticks: 17.00, Counted: 16.00, Add: 0.01, Alter Zähler: 26096.66, Neuer Zähler: 26096.67
07:36:17.767 GPIO14: 18.00, Ticks: 18.00, Counted: 17.00, Add: 0.01, Alter Zähler: 26096.67, Neuer Zähler: 26096.68
07:36:59.686 GPIO14: 19.00, Ticks: 19.00, Counted: 18.00, Add: 0.01, Alter Zähler: 26096.68, Neuer Zähler: 26096.69
07:37:41.682 GPIO14: 20.00, Ticks: 20.00, Counted: 19.00, Add: 0.01, Alter Zähler: 26096.69, Neuer Zähler: 26096.70
07:38:23.755 GPIO14: 21.00, Ticks: 21.00, Counted: 20.00, Add: 0.01, Alter Zähler: 26096.70, Neuer Zähler: 26096.71
07:39:05.854 GPIO14: 22.00, Ticks: 22.00, Counted: 21.00, Add: 0.01, Alter Zähler: 26096.71, Neuer Zähler: 26096.71
07:39:47.946 GPIO14: 23.00, Ticks: 23.00, Counted: 22.00, Add: 0.01, Alter Zähler: 26096.71, Neuer Zähler: 26096.72
07:40:30.168 GPIO14: 24.00, Ticks: 24.00, Counted: 23.00, Add: 0.01, Alter Zähler: 26096.72, Neuer Zähler: 26096.73
07:41:12.622 GPIO14: 25.00, Ticks: 25.00, Counted: 24.00, Add: 0.01, Alter Zähler: 26096.73, Neuer Zähler: 26096.74
07:41:55.150 GPIO14: 26.00, Ticks: 26.00, Counted: 25.00, Add: 0.01, Alter Zähler: 26096.74, Neuer Zähler: 26096.75
07:42:37.468 GPIO14: 27.00, Ticks: 27.00, Counted: 26.00, Add: 0.01, Alter Zähler: 26096.75, Neuer Zähler: 26096.76
07:43:19.775 GPIO14: 28.00, Ticks: 28.00, Counted: 27.00, Add: 0.01, Alter Zähler: 26096.76, Neuer Zähler: 26096.77

```

### TO REPRODUCE
I can't tell exactly, as the problem happens irregular with different numbers.

### EXPECTED BEHAVIOUR
A correct add execution with 2 digit precision.

### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._

### ADDITIONAL CONTEXT
Here is the script used to calculate the current value from reed contact ticks (the current counter value is initially set with the nm-field in the Web UI:

>D
p:zaehlerwert=0
p:cntd_ticks=0
p:ticks=0
add_ticks=0
zneu=0

>BS
=> counterdebounce 500
=> counterdebouncelow 500
=> counterdebouncehigh 500
=#setzero

>W
nm(0 99999.99 0.01 zaehlerwert "{s}Zählerstand{m}" (100) (2)){e}

>F
ticks=pc[1]
if (chg[ticks]>0)
then
if (ticks-cntd_ticks>0)
then
add_ticks=(ticks-cntd_ticks)/100
zneu=zaehlerwert+add_ticks
print GPIO14: %pc[1]%, Ticks: %ticks%, Counted: %cntd_ticks%, Add: %add_ticks%, Alter Zähler: %zaehlerwert%, Neuer Zähler: %zneu%
cntd_ticks=ticks
zaehlerwert=zneu
endif
endif

#setzero
=> counter1 0
ticks=0
cntd_ticks=0
add_ticks=0
zneu=0

**(Please, remember to close the issue when the problem has been addressed)**

sfromis, 2025-01-15:
It is pushing the limits of accuracy when wanting to use single precision floating point for 7 digits including decimals. The binary format cannot be accurate for decimals which are not a sum of negative powers of two, and the number of possible decimal digits is rather low with 7 digits in total, where last digit can only have a reasonable expectation of accuracy with integer values.

You may get better results by doing it all in integers, and only divide by 100 after having done all the calculations.

In short, I don't think that there is "something wrong", only you having to realize the limitations of the data formats used.

dideldum, 2025-01-15:
@sfromis Thank you for your response.
Indeed following Wikipedia a signed float (32bit) should have a precision of 2^(-7)=0,007 for values between 2^14 to 2^15. So I understand, that an add of +0.01 in reality with the floats is a +0.007, which can lead to false results just due to that precision and the following rounding to 2 precision digits.
I will follow your advice and do the calculation with ints and divide by 100 after the calculations.

closed 2025-01-15 by dideldum