arendst/Tasmota #22826: MQT: Connect failed to server <SERVER>:1882, rc -2

schweini, 2025-01-16:
### PROBLEM DESCRIPTION
I am going mad, I fear.
No matter what I do, some Sonoff devices I just flashed with Tasmota-sensors 14.4.1.3 (ddcdc85-sensors).
They just refuse to connect to my (mosquitto) MQTT server with the error "rc -2".
These devices are quite remote. I have exactly the same device (Sonoff POW R2) here, running exactly the same firmware, and it works fine here!

I have tried a lot of things, but nothing works:
- tried different versions of mosquitto (1.x and 2.x).
- changed to a simple password in the UI (it used to have some special characters).
- I can succesfully connect to my server using mosquitto_sub from a windows laptop that is on exactly the same WIFI network as the Tasmota device.
- I used three different servers of mine.
- running mosquitto on a different port, in case 1883 is blocked somewhere.
- I shutdown mosquitto server and ran netcat on port 1883. "netcat" shows that there DOES seem to be some traffic from the remote location. But mosquitto's logs (in mode 'all') don't show anything. So the initial connection seems to be interrupted somehow? But it's definitely not some firewall along the route blocking it.

So: 
- it doesn't seem to be a ISP issue (since it works fine using the same port from the laptop on the same WIFI)
- it doesn't seem to be a server issue (since my local dev device works perfectly with 100% the same settings)

Is there any way to debug this in a more detailed way than jut 'RC -2'? I am aware that RC -2 indicated network issues, so I have turned on logging in Tasmota, and right before the failed MQTT connection I just get a successful DNS request line.

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [ x ] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [ x ] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [ x ] Searched the problem in [discussions](https://github.com/arendst/Tasmota/discussions)
- [ x ] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [ x ] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [ x ] Device used (e.g., Sonoff Basic): Sonoff POW R2 and Sonoff TH16
- [ x ] Tasmota binary firmware version number used: Tasmota-sensors 14.4.1.3 (ddcdc85-sensors)
  - [ x ] Pre-compiled
  - [ ] Self-compiled
- [ x ] Flashing tools used: web flasher and python flasher
- [ x ] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```lua
  Configuration output here:
can't do that, because remote copy and paste doesnt work. also doesnt seem to help with this.

```
- [ ] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```lua
  Rules output here:

```
- [ ] Provide the output of this command: `Status 0`:
```lua
  STATUS 0 output here:

```
- [ ] Set `weblog` to 4 and then, when you experience your issue, provide the output of the Console log:
```lua
  Console output here:
indicated above

```

### TO REPRODUCE
_Steps to reproduce the behavior:_

### EXPECTED BEHAVIOUR
_A clear and concise description of what you expected to happen._

### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._

### ADDITIONAL CONTEXT
_Add any other context about the problem here._

**(Please, remember to close the issue when the problem has been addressed)**

sfromis, 2025-01-16:
As return code -2 indeed [means](https://tasmota.github.io/docs/MQTT/#return-codes-rc) "network connection failed", whatever can cause that would be outside of Tasmota, including but not limited to if you did not specify the right host/IP for your broker. Hence there would not be useful info available for Tasmota to log.

It being "quite remote" sounds like a use case for encrypted MQTT-TLS, which requires a special build and configuration, usually not compatible with using port 1883. In case of the bad practice of trying to use unencrypted MQTT via the open Internet, there is also the risk of an ISP messing with certain types of traffic (using a different port number does not check this), for protection purposes.

If the issue was about credentials, you'd get a different return code than -2.

schweini, 2025-01-17:
This must definitely be a regession type bug in Tasmota. I downgraded to version 8.5.1, and - without changing any settings at all - it immediatly connected to the same MQTT server and is working now.

barbudor, 2025-01-17:
If you jumped from 8.5.1 to 14.x, and using TLS, indeed there were quite some changes

I'm not sure exactly what have been done but I remember a recent discussion about preventing to use TLS over 1883 (or was that preventin,g using non-TLS over 8883) ?
However I'm not sure if anything was changed in that topic

Making big jump in version number is definitively taking a risk of breaking changes that would have been smoothed regularly following the releases

s-hadinger, 2025-01-17:
Can you provide logs please?
Are you using TLS?
Are you using mDNS?

Jason2866, 2025-01-17:
If not using TLS, this error message often appears when the power supply is border line.
Not unusual with old Sonoff devices. Older firmware often works since less features and needs a little less current.