arendst/Tasmota #22839: 14.3.0 - enabling `SetOption130` prevents working script from running

mi-hol, 2025-01-18:
> Using `SetOption130` you can get Tasmota to include heap size in every console log message, which may work better as long as there is activity.

_Originally posted by @sfromis in https://github.com/arendst/Tasmota/discussions/20430#discussioncomment-11874705_

on ESP32C6 Tasmota 14.3.0 it actually prevents working script from running!

Program Version	14.3.0 (ottelo.jimdo.de)
Build Date & Time	2025-01-16T12:02:13
Core/SDK Version	3_1_0/5.3.1.240924

Console:
13.339 CMD: & b
13:53:13.501 CMD: Y
13:53:13.503 RSL: RESULT = {"Command":"Unknown","Input":"Y"}
13:53:14.339 CMD: & b
13:53:14.501 CMD: Y
13:53:14.504 RSL: RESULT = {"Command":"Unknown","Input":"Y"}
13:53:14.617 CMD: 
13:53:15.339 CMD: & b

sfromis, 2025-01-18:
That looks more like serial crosstalk, not a direct result of `SetOption130`

I saw no trouble activating it on an ESP32-C6 (not using scripting)
```
14:58:08.046-259/10 CMD: setoption130
14:58:08.047-259/10 SRC: WebConsole from 192.168.xx.xx
14:58:08.048-259/10 CMD: Grp 0, Cmd 'SETOPTION', Idx 130, Len 0, Pld -99, Data ''
14:58:08.050-256/09 MQT: xxxxxxxx/stat/SETOPTION = {"SetOption130":"ON"}
```
If you were already having crashes within a few minutes, that's more likely involved in the cause (if it is not actually crosstalk).

arendst, 2025-01-18:
Indeed, nothing wrong with SO130 1 on ESP32C6
```
16:12:41.460 CMD: so130 1
16:12:41.460 SRC: WebConsole from 192.168.2.1
16:12:41.462 CMD: Grp 0, Cmd 'SO', Idx 130, Len 1, Pld 1, Data '1'
16:12:41.466-275/08 MQT: stat/remeha2/RESULT = {"SetOption130":"ON"}
16:12:42.268-279/09 CFG: Saved, Count 188, Bytes 4096
16:12:54.785-282/10 HTP: Tools
16:12:57.454-281/09 HTP: Main Menu
16:12:58.803-282/11 HTP: Information
16:13:03.813-282/11 SRC: Berry
16:13:03.814-282/11 CMD: Grp 0, Cmd 'DZSEND', Idx 1, Len 8, Pld 526, Data '526,44.1'
16:13:03.818-279/10 MQT: domoticz/in = {"idx":526,"nvalue":0,"svalue":"44.1","Battery":100,"RSSI":6}
16:13:08.020-279/10 HTP: Main Menu
16:13:10.815-276/09 HTP: Tools
16:13:12.097-276/09 HTP: Console
16:13:25.798-278/09 SRC: Berry
16:13:25.799-278/09 CMD: Grp 0, Cmd 'DZSEND', Idx 1, Len 8, Pld 525, Data '525,42.7'
16:13:25.802-275/09 MQT: domoticz/in = {"idx":525,"nvalue":0,"svalue":"42.7","Battery":100,"RSSI":6}
16:14:54.733-280/09 CMD: status 0
16:14:54.734-280/09 SRC: WebConsole from 192.168.2.1
16:14:54.736-280/09 CMD: Grp 0, Cmd 'STATUS', Idx 1, Len 1, Pld 0, Data '0'
16:14:54.739-277/10 MQT: stat/remeha2/STATUS = {"Status":{"Module":0,"DeviceName":"Remeha2","FriendlyName":["Tasmota"],"Topic":"remeha2","ButtonTopic":"0","Power":"0","PowerLock":"0","PowerOnState":3,"LedState":1,"LedMask":"FFFF","SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0,"InfoRetain":0,"StateRetain":0,"StatusRetain":0}}
16:14:54.744-276/10 MQT: stat/remeha2/STATUS1 = {"StatusPRM":{"Baudrate":115200,"SerialConfig":"8N1","GroupTopic":"tasmotas","OtaUrl":"http://otaserver/ota/tasmota/tasmota32c6-theo.bin","RestartReason":"Software reset CPU","Uptime":"0T00:17:30","StartupUTC":"2025-01-18T14:57:24","Sleep":50,"CfgHolder":4617,"BootCount":66,"BCResetTime":"2024-12-29T13:47:26","SaveCount":188}}
16:14:54.746-276/09 MQT: stat/remeha2/STATUS2 = {"StatusFWR":{"Version":"14.4.1.3(theo)","BuildDateTime":"2025-01-16T15:19:55","Core":"3_1_1","SDK":"5.3.2.250106","CpuFrequency":160,"Hardware":"ESP32-C6FH4 v0.1","CR":"413/699"}}
16:14:54.749-276/09 MQT: stat/remeha2/STATUS3 = {"StatusLOG":{"SerialLog":0,"WebLog":3,"MqttLog":0,"FileLog":0,"SysLog":0,"LogHost":"loghost","LogPort":514,"SSId":["indebuurt_IoT","indebuurt"],"TelePeriod":300,"Resolution":"558180C0","SetOption":["00008009","2805C80001000600003C5A0A192800000000","00000088","00006100","00014000","00028000"]}}
16:14:54.757-274/10 MQT: stat/remeha2/STATUS4 = {"StatusMEM":{"ProgramSize":1518,"Free":1361,"Heap":276,"StackLowMark":4,"PsrMax":0,"PsrFree":0,"ProgramFlashSize":4096,"FlashSize":4096,"FlashChipId":"164020","FlashFrequency":80,"FlashMode":"QIO","Features":["0809","8F9A85CF","00040001","05021ECF","01000BC0","00010301","40004080","00200000","A49C002D","00020070","00000B44"],"Drivers":"1,2,!3,!4,!5,7,!8,9,10,12,!24,!27,!28,31,38,50,52,!56,!60,62,!67,!75,82,!86,!87,!88,!91,!121","Sensors":"1,2,3,5,6,8,9,10,14,21,28,34,85,97,127","I2CDriver":"2,7,9,10,11,15,18,58,65,68,77,88,89,90"}}
16:14:54.765-275/11 MQT: stat/remeha2/STATUS5 = {"StatusNET":{"Hostname":"remeha2","IPAddress":"192.168.2.128","Gateway":"192.168.2.254","Subnetmask":"255.255.255.0","DNSServer1":"192.168.2.27","DNSServer2":"192.168.2.24","Mac":"54:32:04:1F:D8:68","IP6Global":"fd84:7996:2cbf:1349:5632:4ff:fe1f:d868","IP6Local":"fe80::5632:4ff:fe1f:d868%st1","Ethernet":{"Hostname":"","IPAddress":"0.0.0.0","Gateway":"0.0.0.0","Subnetmask":"0.0.0.0","DNSServer1":"192.168.2.27","DNSServer2":"192.168.2.24","Mac":"00:00:00:00:00:00","IP6Global":"","IP6Local":""},"Webserver":2,"HTTP_API":1,"WifiConfig":4,"WifiPower":16.0}}
16:14:54.770-274/10 MQT: stat/remeha2/STATUS6 = {"StatusMQT":{"MqttHost":"mqttbroker","MqttPort":1883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_1FD868","MqttUser":"DVES_USER","MqttCount":1,"MAX_PACKET_SIZE":1200,"KEEPALIVE":30,"SOCKET_TIMEOUT":4}}
16:14:54.774-274/10 MQT: stat/remeha2/STATUS7 = {"StatusTIM":{"UTC":"2025-01-18T15:14:54Z","Local":"2025-01-18T16:14:54","StartDST":"2025-03-30T02:00:00","EndDST":"2025-10-26T03:00:00","Timezone":99,"Sunrise":"08:35","Sunset":"16:49"}}
16:14:54.779-274/10 MQT: stat/remeha2/STATUS10 = {"StatusSNS":{"Time":"2025-01-18T16:14:54","Calenta":{"AM012":3,"AM014":30,"Pressure":2.4,"Setpoint":42.70,"Flow":44.10},"PressureUnit":"hPa"}}
16:14:54.784-276/11 MQT: stat/remeha2/STATUS11 = {"StatusSTS":{"Time":"2025-01-18T16:14:54","Uptime":"0T00:17:30","UptimeSec":1050,"Heap":274,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":105,"MqttCount":1,"Berry":{"HeapUsed":8,"Objects":101},"Wifi":{"AP":1,"SSId":"indebuurt_IoT","BSSId":"18:E8:29:CA:17:C1","Channel":11,"Mode":"HT40","RSSI":66,"Signal":-67,"LinkCount":1,"Downtime":"0T00:00:03"}}}
16:15:05.808-279/09 SRC: Berry
16:15:05.809-279/09 CMD: Grp 0, Cmd 'DZSEND', Idx 1, Len 8, Pld 525, Data '525,42.8'
16:15:05.813-276/08 MQT: domoticz/in = {"idx":525,"nvalue":0,"svalue":"42.8","Battery":100,"RSSI":6}
```

closed 2025-01-18 by arendst

mi-hol, 2025-01-18:
> I saw no trouble activating it on an ESP32-C6 (not using scripting)

May I kindly ask for a test what happens with scripting enabled (as in my case)?

arendst, 2025-01-18:
As I never use/know scripting someone else might be willing to assist...

ottelo9, 2025-01-18:
My C3 is running with Option and Script enabled,  (apart from the reboot problem).

mi-hol, 2025-01-19:
I found a hint in Seeed's doc for my  C6 board
"Currently we recommend version 7.0.0 of the EspSoftwareSerial library. Other versions may have varying degrees of problems that prevent the soft serial port from working properly."

if above mentioned minimum version is included in ESP32/Arduino library Core version v3.1.0.240926. used in Tasmota 14,3,0 is unclear to me. Search Espressif's repo I was unable to find this infos :(

arendst, 2025-01-19:
We do not support software serial on esp32. Although the SML driver might (which is a parasite on tasmota). 

How many serial interfaces do you use? The esp32c6 only has two uarts of which one is used for serial console leaving one for SML.

mi-hol, 2025-01-19:
ok, with no software serial support the hint seems not relevant.

> Although the SML driver might (which is a parasite on tasmota).

Not sure to understand, but sounds concerning. Shall we ask the author of driver for details?

IR TTL sensor for SML is connected to HW serial (RX GPIO port 17 and TX GPIO port 16 ) see below

[image]

Currently I'm try to convert my script to Berry.
Guess it would be better to continue the conversation on Discord, right?

sfromis, 2025-01-19:
There is no concept of certain pins being hardware serial, such labeling on boards is at most a suggestion. You can configure any pin pair for hardware serial. That also goes for other pin types, except that analog usage of the pin does not have this flexibility.