arendst/Tasmota #22861: SwitchMode 13/14 wrong way round?

Altblechkutscher, 2025-01-21:
### PROBLEM DESCRIPTION
Recently when trying to make my PIR smart (external Steinel which has its output contact wired to the switch input of a Shelly Mini 1 Gen3) i've experienced diffuiculties when selecting SwitchMode. I've red the documentation:

[image]

If I interpret correctly SwitchMode 13 would be correct (PIR sets 230V output to HIGH, therefore the Output Relais is switched on). But when trying this out, the PIRs internal relais is switched on (input Shelly HIGH), but nothing happens. When the PIR shuts down, the output of the Shelly switches on.

After lots of experimenting, I've tried SwitchMode 14, which seems to work correctly. Am I reading the documentation wrong?

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [x] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [x] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [x] Searched the problem in [discussions](https://github.com/arendst/Tasmota/discussions)
- [x] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [ ] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [x] Device used (e.g., Sonoff Basic): Shelly Mini 1 Gen3
- [ ] Tasmota binary firmware version number used: Tasmota 14.4.1 (release-tasmota32)](https://bit.ly/tasmota)
  - [x] Pre-compiled
  - [ ] Self-compiled
- [ ] Flashing tools used: _____
- [ ] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```lua
  Configuration output here:

```
- [ ] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```lua
  Rules output here:

```
- [ ] Provide the output of this command: `Status 0`:
```lua
  STATUS 0 output here:

```
- [ ] Set `weblog` to 4 and then, when you experience your issue, provide the output of the Console log:
```lua
  Console output here:

```

### TO REPRODUCE
_Steps to reproduce the behavior:_

### EXPECTED BEHAVIOUR
_A clear and concise description of what you expected to happen._

### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._

### ADDITIONAL CONTEXT
_Add any other context about the problem here._

**(Please, remember to close the issue when the problem has been addressed)**

sfromis, 2025-01-21:
Just tested it, and SwitchMode 13 worked as documented, "push to on" with a "normally wired" physical button.

If your device outputs high for activated (like a PIR often does), it is **inverted**, and then SwitchMode 14 may indeed be what you want for that.

And the docs does very clearly mention that SwitchMode 14 "is useful with PIR sensors".
https://tasmota.github.io/docs/Buttons-and-Switches/#switchmode

Working as designed.

Altblechkutscher, 2025-01-21:
Thank you for your explanation, still I do have a knot in my understanding. I dont't really get why it should be inverted?

Push to on (line gets high) -> Output on. Same for a Switch and a PIR Output, where is the difference? If I put a switch there instead of the PIR Output then there is no difference?

sfromis, 2025-01-21:
It should be inverted because the PIR signaling is *already* inverted. 

Normal wiring for a button is that pushing it connects the gpio to ground, and releasing it lets the internal pull-up resistor bring it to high level. It is "active low". And that's why "active high" is inverted signals.

Altblechkutscher, 2025-01-22:
Now it makes sense. Thank you for the explanation!

closed 2025-01-22 by Altblechkutscher