arendst/Tasmota #22866: I am testing Tasmota 14 on the WT32_ETH01. When I use Backlog to change the Topic and then restart the device, it does not take the new value.

bilgeexuu1, 2025-01-23:
### PROBLEM DESCRIPTION
_A clear and concise description of what the problem is._

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [ ] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [ ] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [ ] Searched the problem in [discussions](https://github.com/arendst/Tasmota/discussions)
- [ ] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [ ] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [ ] Device used (e.g., Sonoff Basic): _____
- [ ] Tasmota binary firmware version number used: _____
  - [ ] Pre-compiled
  - [ ] Self-compiled
- [ ] Flashing tools used: _____
- [ ] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```lua
  Configuration output here:

```
- [ ] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```lua
  Rules output here:

```
- [ ] Provide the output of this command: `Status 0`:
```lua
  STATUS 0 output here:

```
- [ ] Set `weblog` to 4 and then, when you experience your issue, provide the output of the Console log:
```lua
  Console output here:

```

### TO REPRODUCE
_Steps to reproduce the behavior:_

### EXPECTED BEHAVIOUR
_A clear and concise description of what you expected to happen._

### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._

### ADDITIONAL CONTEXT
_Add any other context about the problem here._

**(Please, remember to close the issue when the problem has been addressed)**

bilgeexuu1, 2025-01-23:
I am testing Tasmota 14 on the WT32_ETH01. When I use Backlog to change the Topic and then restart the device, it does not take the new Topic.

sfromis, 2025-01-23:
You "forgot" to provide **any** of the requested details, including logs. No basis for even guessing what happens.

bilgeexuu1, 2025-01-24:
00:00:00.001 HDW: ESP32-D0WD v1.0 
00:00:00.013 UFS: FlashFS mounted with 300 kB free
00:00:00.028 CFG: Loaded from LKG File, Count 100
00:00:00.037 QPC: Count 1
00:00:00.063 BRY: Berry initialized, RAM used 3407 bytes
00:00:00.077 CFG: Energy loaded from file
00:00:00.077 NRG: Init driver 5
00:00:00.081 Project WT32-ETH01 - PZEM_004T_WT32-ETH01 Version 14.3.0(tasmota)-3_1_0(2025-01-24T21:42:16)
00:00:01.691 ETH: Connected at 100Mbps Full Duplex, Mac 8C:CE:4E:95:C0:13, Hostname WT32-1073499209-eth
00:00:02.295 mDN: Initialized 'WT32-1073499209.local'
00:00:02.301 HTP: Web server active on WT32-1073499209-eth.local with IP address 10.2.2.201
21:53:11.395 mDN: Query done. MQTT services found 0
21:53:11.396 MQT: Attempting connection...
21:53:11.712 MQT: Connected
21:53:11.714 MQT: /v1/ondo/power/default/8CCE4E95C010/tele/LWT = Online (retained)
21:53:11.716 MQT: /v1/ondo/power/default/8CCE4E95C010/cmnd/POWER = 
21:53:11.731 MQT: /v1/ondo/power/default/8CCE4E95C010/tele/INFO1 = {"Info1":{"Module":"WT32-PZEM-004T","Version":"14.3.0(tasmota)","FallbackTopic":"cmnd/8CCE4E95C010_fb/","GroupTopic":"/tasmotas/cmnd/"}}
21:53:11.744 MQT: /v1/ondo/power/default/8CCE4E95C010/tele/INFO2 = {"Info2":{"WebServerMode":"Admin","Ethernet":{"Hostname":"WT32-1073499209-eth","IPAddress":"10.2.2.201"}}}
21:53:11.756 MQT: /v1/ondo/power/default/8CCE4E95C010/tele/INFO3 = {"Info3":{"RestartReason":"Software reset CPU","BootCount":29}}
21:53:11.769 MQT: /v1/ondo/power/default/8CCE4E95C010/stat/POWER1 = {"POWER1":"OFF"}
21:53:11.781 MQT: /v1/ondo/power/default/8CCE4E95C010/stat/POWER1 = OFF (retained)
21:53:11.782 MQT: /v1/ondo/power/default/8CCE4E95C010/stat/POWER2 = {"POWER2":"OFF"}
21:53:11.793 MQT: /v1/ondo/power/default/8CCE4E95C010/stat/POWER2 = OFF (retained)

change Topic:

Topic /v1/ondo/power/app/8CCE4E95C010

Backlog:   

[image]

restart
Topic not changing
21:57:08.407 mDN: Query done. MQTT services found 0
21:57:08.407 MQT: Attempting connection...
21:57:08.713 MQT: Connected
21:57:08.714 MQT: /v1/ondo/power/**default**/8CCE4E95C010/tele/LWT = Online (retained)
21:57:08.716 MQT: /v1/ondo/power/**default**/8CCE4E95C010/cmnd/POWER = 
21:57:08.731 MQT: /v1/ondo/power/default/8CCE4E95C010/tele/INFO1 = {"Info1":{"Module":"WT32-PZEM-004T","Version":"14.3.0(tasmota)","FallbackTopic":"cmnd/8CCE4E95C010_fb/","GroupTopic":"/tasmotas/cmnd/"}}
21:57:08.744 MQT: /v1/ondo/power/default/8CCE4E95C010/tele/INFO2 = {"Info2":{"WebServerMode":"Admin","Ethernet":{"Hostname":"WT32-1073499209-eth","IPAddress":"10.2.2.201"}}}
21:57:08.757 MQT: /v1/ondo/power/default/8CCE4E95C010/tele/INFO3 = {"Info3":{"RestartReason":"Software reset CPU","BootCount":29}}
21:57:08.769 MQT: /v1/ondo/power/**default**/8CCE4E95C010/stat/POWER1 = {"POWER1":"OFF"}
21:57:08.781 MQT: /v1/ondo/power/**default**/8CCE4E95C010/stat/POWER1 = OFF (retained)
21:57:08.782 MQT: /v1/ondo/power/**default**/8CCE4E95C010/stat/POWER2 = {"POWER2":"OFF"}
21:57:08.794 MQT: /v1/ondo/power/**default/**8CCE4E95C010/stat/POWER2 = OFF (retained)