arendst/Tasmota #22896: All device sensors move to unavailable state in Home Assistant, after Home Assistant restart

simowilso, 2025-01-29:
### PROBLEM DESCRIPTION
As title, if I restart Home  Assistant, then all sensors from all Tasmota 14.4.1 devices change to unavailable. Restarting mosquitto/mqtt docker only does not cause this problem.

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [ X] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [ X] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [ ] Searched the problem in [discussions](https://github.com/arendst/Tasmota/discussions)
- [ X] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)

The FAQ says to "Make sure it's enabled in Tasmota it with SetOption19 1" if I do this the device disappears from HA completely. If I set it back 0, the device appears, but the sensors remain unavailable. This all used to work, I don't think I have changed anything other than upgrading tasmota.

- [ ] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [ ] Device used (e.g., Sonoff Basic): many different ones
- [ ] Tasmota binary firmware version number used: various 14.4.1
  - [ X] Pre-compiled
  - [ ] Self-compiled
- [X ] Flashing tools used: ota
- [ ] Provide the output of command: `Backlog Template; Module; GPIO 255`:

```lua
  Configuration output here:
14:04:31.046 MQT: stat/powermon17/RESULT = {"NAME":"Gosund UP111","GPIO":[0,320,0,32,2720,2656,0,0,2624,576,224,0,0,0],"FLAG":0,"BASE":18}
14:04:31.257 MQT: stat/powermon17/RESULT = {"Module":{"0":"Gosund UP111"}}
14:04:31.463 MQT: stat/powermon17/RESULT = {"GPIO0":{"0":"None"},"GPIO1":{"320":"Led_i1"},"GPIO2":{"0":"None"},"GPIO3":{"32":"Button1"},"GPIO4":{"2720":"BL0937 CF"},"GPIO5":{"2656":"HLWBL CF1"},"GPIO9":{"0":"None"},"GPIO10":{"0":"None"},"GPIO12":{"2624":"HLWBL SEL_i"},"GPIO13":{"576":"LedLink_i"},"GPIO14":{"224":"Relay1"},"GPIO15":{"0":"None"},"GPIO16":{"0":"None"},"GPIO17":{"0":"None"}}

```
- [ ] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```lua
  Rules output here:
14:03:28.443 MQT: stat/powermon17/RESULT = {"Rule1":{"State":"OFF","Once":"OFF","StopOnError":"OFF","Length":0,"Free":511,"Rules":""}}
14:03:28.656 MQT: stat/powermon17/RESULT = {"Rule2":{"State":"OFF","Once":"OFF","StopOnError":"OFF","Length":0,"Free":511,"Rules":""}}
14:03:28.859 MQT: stat/powermon17/RESULT = {"Rule3":{"State":"OFF","Once":"OFF","StopOnError":"OFF","Length":0,"Free":511,"Rules":""}}

```
- [ ] Provide the output of this command: `Status 0`:
```lua
  STATUS 0 output here:
14:01:45.041 MQT: stat/powermon17/STATUS = {"Status":{"Module":0,"DeviceName":"Humax","FriendlyName":["Humax"],"Topic":"powermon17","ButtonTopic":"0","Power":"1","PowerLock":"0","PowerOnState":3,"LedState":1,"LedMask":"FFFF","SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0,"InfoRetain":0,"StateRetain":0,"StatusRetain":0}}
14:01:45.048 MQT: stat/powermon17/STATUS1 = {"StatusPRM":{"Baudrate":115200,"SerialConfig":"8N1","GroupTopic":"tasmotas","OtaUrl":"http://ota.tasmota.com/tasmota/release/tasmota.bin.gz","RestartReason":"Software/System restart","Uptime":"0T23:37:45","StartupUTC":"2025-01-28T14:24:00","Sleep":50,"CfgHolder":4617,"BootCount":186,"BCResetTime":"2021-06-16T14:07:53","SaveCount":1826,"SaveAddress":"F9000"}}
14:01:45.055 MQT: stat/powermon17/STATUS2 = {"StatusFWR":{"Version":"14.4.1(release-tasmota)","BuildDateTime":"2024-12-15T13:29:54","Boot":31,"Core":"2_7_8","SDK":"2.2.2-dev(38a443e)","CpuFrequency":80,"Hardware":"ESP8285N08","CR":"369/699"}}
14:01:45.062 MQT: stat/powermon17/STATUS3 = {"StatusLOG":{"SerialLog":2,"WebLog":2,"MqttLog":0,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["heskem-iot",""],"TelePeriod":30,"Resolution":"558180C0","SetOption":["00008009","2805C80001000600003C5A0A002800000000","00000080","00006000","00004000","00000000"]}}
14:01:45.076 MQT: stat/powermon17/STATUS4 = {"StatusMEM":{"ProgramSize":652,"Free":348,"Heap":23,"ProgramFlashSize":1024,"FlashSize":1024,"FlashChipId":"144051","FlashFrequency":40,"FlashMode":"DOUT","Features":["0809","8F9AC787","04368001","000000CF","010013C0","C000F981","00004004","00001000","54000020","00000080","00000000"],"Drivers":"1,2,3,!4,!5,!6,7,!8,9,10,12,!16,!18,!19,!20,!21,!22,!24,26,!27,29,!30,!35,!37,!45,62,!68","Sensors":"1,2,3,4,5,6","I2CDriver":"7"}}
14:01:45.083 MQT: stat/powermon17/STATUS5 = {"StatusNET":{"Hostname":"humax","IPAddress":"172.16.20.113","Gateway":"172.16.20.1","Subnetmask":"255.255.255.0","DNSServer1":"8.8.8.8","DNSServer2":"0.0.0.0","Mac":"24:A1:60:09:44:92","Webserver":2,"HTTP_API":1,"WifiConfig":4,"WifiPower":17.0}}
14:01:45.089 MQT: stat/powermon17/STATUS6 = {"StatusMQT":{"MqttHost":"192.168.0.54","MqttPort":1883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_094492","MqttUser":"mqttuser","MqttCount":2,"MAX_PACKET_SIZE":1200,"KEEPALIVE":30,"SOCKET_TIMEOUT":4}}
14:01:45.097 MQT: stat/powermon17/STATUS7 = {"StatusTIM":{"UTC":"2025-01-29T14:01:45Z","Local":"2025-01-29T14:01:45","StartDST":"2025-03-30T01:00:00","EndDST":"2025-10-26T01:00:00","Timezone":99,"Sunrise":"07:46","Sunset":"16:48"}}
14:01:45.104 MQT: stat/powermon17/STATUS9 = {"StatusPTH":{"PowerDelta":0,"PowerLow":0,"PowerHigh":0,"VoltageLow":0,"VoltageHigh":0,"CurrentLow":0,"CurrentHigh":0,"MaxPower":0,"MaxPowerHold":10,"MaxPowerWindow":30,"MaxEnergy":0,"MaxEnergyStart":0}}
14:01:45.114 MQT: stat/powermon17/STATUS10 = {"StatusSNS":{"Time":"2025-01-29T14:01:45","ENERGY":{"TotalStartTime":"2021-06-16T14:07:53","Total":58.495,"Yesterday":0.050,"Today":0.029,"Power":2,"ApparentPower":7,"ReactivePower":7,"Factor":0.27,"Voltage":240,"Current":0.029}}}
14:01:45.123 MQT: stat/powermon17/STATUS11 = {"StatusSTS":{"Time":"2025-01-29T14:01:45","Uptime":"0T23:37:45","UptimeSec":85065,"Heap":23,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":2,"POWER":"ON","Wifi":{"AP":1,"SSId":"heskem-iot","BSSId":"6E:D7:9A:17:A0:FE","Channel":6,"Mode":"11n","RSSI":72,"Signal":-64,"LinkCount":1,"Downtime":"0T00:00:04"}}}

```
- [ ] Set `weblog` to 4 and then, when you experience your issue, provide the output of the Console log:
```lua
  Console output here:

```

### TO REPRODUCE
_Steps to reproduce the behavior:_restart Home Assistant

### EXPECTED BEHAVIOUR
_A clear and concise description of what you expected to happen._
All sensors should still be available

### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._

[image]

### ADDITIONAL CONTEXT
_Add any other context about the problem here._

**(Please, remember to close the issue when the problem has been addressed)**

sfromis, 2025-01-29:
I'd recommend to have a stable broker running outside of HA, to reduce the impact of "brutality" like restarting it all. I'm not using HA, but my *guess* would be that retained messages were nuked, and I'd try restarting Tasmota devices to see if it helps.

simowilso, 2025-01-29:
I thought I mentioned that do have a separate mqtt docker  but perhaps not. I can restart (mosquitto) and don't get any problems when I do that.

sfromis, 2025-01-29:
Well, it was not very explicit then. I'd still suggest to check if it helps restarting Tasmota devices.

NeosNDG, 2025-01-29:
Just updated 10 Refoss P11 to 14.4.1 and I can´t reproduce this behavior in HA. 
Connected via HA Mosquitto broker only. 

[image]

[image]

simowilso, 2025-01-29:
> Well, it was not very explicit then. I'd still suggest to check if it helps restarting Tasmota devices.

Yes restarting tasmota device works, but it's a pain to do 34 of them each time I need to restart HA

closed 2025-01-29 by simowilso