arendst/Tasmota #22919: 433MHz  0x or no 0x depending on module

bobybobybob, 2025-02-02:
### PROBLEM DESCRIPTION
Inconsistent output of received data at 433MHz      one have 0x and other not.

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [ ] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [ ] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [ ] Searched the problem in [discussions](https://github.com/arendst/Tasmota/discussions)
- [ ] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [ ] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [ ] Device used (e.g., Sonoff Basic): _____
- [ X] Tasmota binary firmware version number used: 14.4.1
  - [ ] Pre-compiled
  - [X ] Self-compiled
- [ ] Flashing tools used: _____
- [ ] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```lua
  Configuration output here:

### TO REPRODUCE
Compare Data in Hex     with Tasmota 433MHz Bridge-Box   and  ESP32-C + RXB6 433MHz Module

433MHz Bride-Box
19:03:02.388 CMD: SetOption28
19:03:02.394 MQT: stat/tasmota_rf433_C37FA9/RESULT = {"SetOption28":"OFF"}
19:04:18.462 MQT: tele/tasmota_rf433_C37FA9/RESULT = {"Time":"2025-01-31T19:04:18","RfReceived":{"Sync":11230,"Low":330,"High":1050,"Data":"FF5F78","RfKey":"None"}}
19:07:16.771 MQT: stat/tasmota_rf433_C37FA9/RESULT = {"SetOption28":"ON"}
19:07:19.255 MQT: tele/tasmota_rf433_C37FA9/RESULT = {"Time":"2025-01-31T19:07:19","RfReceived":{"Sync":11190,"Low":340,"High":1060,"Data":16736114,"RfKey":"None"}}

ESP32-C + RXB6 MHz
9:03:09.036 CMD: SetOption28
19:03:09.041 MQT: stat/tasmota_61FEE0/RESULT = {"SetOption28":"OFF"}
19:04:11.207 CMD: SetOption28 1
19:04:11.212 MQT: stat/tasmota_61FEE0/RESULT = {"SetOption28":"ON"}
19:04:17.851 MQT: tele/tasmota_61FEE0/RESULT = {"Time":"2025-01-31T19:04:17","RfReceived":{"Data":16736120,"Bits":24,"Protocol":1,"Pulse":362}}
19:05:31.162 CMD: SetOption28 0
19:05:31.167 MQT: stat/tasmota_61FEE0/RESULT = {"SetOption28":"OFF"}
19:05:34.372 MQT: tele/tasmota_61FEE0/RESULT = {"Time":"2025-01-31T19:05:34","RfReceived":{"Data":"0xFF5F72","Bits":24,"Protocol":1,"Pulse":360}}

 
The one show Data in Hex with beginning 0x  the other one not.   

### EXPECTED BEHAVIOUR
I think both should in hex look same. 

### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._

![Image](https://github.com/user-attachments/assets/2747d943-1678-4f4a-8f60-4b41065c0820)

### ADDITIONAL CONTEXT
In a Tasmota help form, a user named Noschvie pointed out that the following lines probably show the difference clearly. 

tasmota\tasmota_xdrv_driver\xdrv_17_rcswitch.ino / Line number 69

tasmota\tasmota_xdrv_driver\xdrv_06_snfbridge.ino / Line number 245

**(Please, remember to close the issue when the problem has been addressed)**

Noschvie, 2025-02-02:
Discussion on Discord:

https://discord.com/channels/479389167382691863/490244847568158751/1335280188002406494

Maybe SetOption28 can be extended with 2 additional options, „Hex with 0x as prefix“ and „Hex without the prefix“.

arendst, 2025-02-02:
It's all about squeezing the most functionality out of as small as possible code base.

JSON does not support hexadecimal as a number so it needs to be shown as a string hence the quotes. For  Tasmota that is enough to separate between decimal and hexadecimel (number or string).

Later implementations started to still use a string but prefixed it with "0x" which makes totally sense but increased the code base by 2 bytes for every hexadecimal value. 

Changing the legacy version from "FDF0A4" to "0xFDF0A4" while likely break current automations in the back-end.

So for now we keep it as it is and your automations can still function as they should as long as you process the data missing "0x".

NB. While at the topic of hexadecimal and decimal numbers, Tasmota's JSON input implementation does allow a hexadecimal number as number (and not a s string) for input. This is not according to the JSON definition but luckily there is a JSON5 definition (which I never heard of before today but seems to be much more relaxed).

bobybobybob, 2025-02-02:
Summary: No global solution, because a solution can cause unexpected problems with other existing systems if the 0X is suddenly new or no longer included. 

If necessary, change the corresponding lines of code self in a user-defined manner or prepare these characters externally in the desired format.

closed 2025-02-02 by bobybobybob

Noschvie, 2025-02-02:
Thanks Theo for your explanation.

Another possibility would be to use a #define (for "self compiled") in order to remain backwards compatible on the one hand and to achieve harmonization on the other.

arendst, 2025-02-02:
A define can be done easily. Let me chk ...

arendst, 2025-02-02:
Done.

Noschvie, 2025-02-02:
Thanks again for this great project!