arendst/Tasmota #22959: Unable to configure relays with opposite states

TheFax, 2025-02-06:
### PROBLEM DESCRIPTION
I'm encountering a problem configuring my Sonoff Dual R3 (ESP32 based) with Tasmota32 (version 14.4.1) to achieve inverted relay states. My goal is to control both relays with a single web button, ensuring that when one relay is ON, the other is OFF, and vice-versa.

### REQUESTED INFORMATION
- [x] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [x] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [x] Searched the problem in [discussions](https://github.com/arendst/Tasmota/discussions)
- [x] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [ ] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [x] Device used: Sonoff Dual R3
- [x] Tasmota binary firmware version number used: 14.4.1
  - [x] Pre-compiled
  - [ ] Self-compiled
- [x] Flashing tools used: RS232
- [x] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```lua
00:00:31.678 RSL: RESULT = {"NAME":"ESP32-DevKit","GPIO":[1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,0,1,1,1,0,1,1,1,0,0,0,0,1,1,1,1,1,0,0,1],"FLAG":0,"BASE":1}
00:00:31.901 RSL: RESULT = {"Module":{"1":"ESP32-DevKit"}}
00:00:32.110 RSL: RESULT = {"GPIO0":{"0":"None"},"GPIO1":{"0":"None"},"GPIO2":{"0":"None"},"GPIO3":{"0":"None"},"GPIO4":{"0":"None"},"GPIO5":{"0":"None"},"GPIO6":{"0":"None"},"GPIO7":{"0":"None"},"GPIO8":{"0":"None"},"GPIO9":{"0":"None"},"GPIO10":{"0":"None"},"GPIO11":{"0":"None"},"GPIO12":{"0":"None"},"GPIO13":{"0":"None"},"GPIO14":{"224":"Relay1"},"GPIO15":{"0":"None"},"GPIO16":{"0":"None"},"GPIO17":{"0":"None"},"GPIO18":{"0":"None"},"GPIO19":{"0":"None"},"GPIO20":{"0":"None"},"GPIO21":{"0":"None"},"GPIO22":{"0":"None"},"GPIO23":{"0":"None"},"GPIO24":{"0":"None"},"GPIO25":{"0":"None"},"GPIO26":{"0":"None"},"GPIO27":{"256":"Relay_i1"},"GPIO32":{"0":"None"},"GPIO33":{"0":"None"},"GPIO34":{"0":"None"},"GPIO35":{"0":"None"},"GPIO36":{"0":"None"},"GPIO37":{"0":"None"},"GPIO38":{"0":"None"},"GPIO39":{"0":"None"}}
```
- [ ] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
NOT using rules at all.
- [x] Provide the output of this command: `Status 0`:
```lua
00:02:58.178 RSL: STATUS = {"Status":{"Module":1,"DeviceName":"Tasmota","FriendlyName":["Tasmota"],"Topic":"tasmota_AFAE54","ButtonTopic":"0","Power":"1","PowerLock":"0","PowerOnState":3,"LedState":1,"LedMask":"FFFF","SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0,"InfoRetain":0,"StateRetain":0,"StatusRetain":0}}
00:02:58.215 RSL: STATUS1 = {"StatusPRM":{"Baudrate":115200,"SerialConfig":"8N1","GroupTopic":"tasmotas","OtaUrl":"http://ota.tasmota.com/tasmota32/release/tasmota32.bin","RestartReason":"Software reset CPU","Uptime":"0T00:02:59","StartupUTC":"","Sleep":50,"CfgHolder":4617,"BootCount":15,"BCResetTime":"1970-01-01T00:00:00","SaveCount":94}}
00:02:58.250 RSL: STATUS2 = {"StatusFWR":{"Version":"14.4.1(release-tasmota32)","BuildDateTime":"2024-12-15T13:33:17","Core":"3_1_0","SDK":"5.3.2","CpuFrequency":160,"Hardware":"ESP32-D0WD-V3 v3.1","CR":"407/699"}}
00:02:58.265 RSL: STATUS3 = {"StatusLOG":{"SerialLog":2,"WebLog":2,"MqttLog":0,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["OpenWrt",""],"TelePeriod":300,"Resolution":"558180C0","SetOption":["00008009","2805C80001000600003C5A0A192800000000","00000080","00006000","00004000","00000000"]}}
00:02:58.309 RSL: STATUS4 = {"StatusMEM":{"ProgramSize":2017,"Free":862,"Heap":138,"StackLowMark":4,"PsrMax":0,"PsrFree":0,"ProgramFlashSize":4096,"FlashSize":4096,"FlashChipId":"1640C8","FlashFrequency":40,"FlashMode":"DIO","Features":["0809","9F9AD7DF","0015A001","B7F7BFCF","05DA9BC4","E0360DC7","480840D2","20200000","D4BC482D","810A80F1","00000014"],"Drivers":"1,2,!3,!4,!5,7,!8,9,10,11,12,!14,!16,!17,!20,!21,!24,26,!27,29,!34,!35,38,50,52,!59,!60,62,!63,!66,!67,!68,!73,!75,82,!86,!87,!88,121","Sensors":"1,2,3,5,6,7,8,9,10,11,12,13,14,15,17,18,19,20,21,22,26,31,34,37,39,40,42,43,45,51,52,55,56,58,59,64,66,67,74,85,92,95,98,103,105,109,127","I2CDriver":"7,8,9,10,11,12,13,14,15,17,18,20,24,29,31,36,41,42,44,46,48,58,62,65,69,76,77,82,89"}}
00:02:58.377 RSL: STATUS5 = {"StatusNET":{"Hostname":"tasmota-AFAE54-3668","IPAddress":"192.168.7.132","Gateway":"192.168.7.1","Subnetmask":"255.255.255.0","DNSServer1":"192.168.7.1","DNSServer2":"0.0.0.0","Mac":"A8:42:E3:AF:AE:54","IP6Global":"","IP6Local":"fe80::aa42:e3ff:feaf:ae54%st1","Ethernet":{"Hostname":"","IPAddress":"0.0.0.0","Gateway":"0.0.0.0","Subnetmask":"0.0.0.0","DNSServer1":"192.168.7.1","DNSServer2":"0.0.0.0","Mac":"00:00:00:00:00:00","IP6Global":"","IP6Local":""},"Webserver":2,"HTTP_API":1,"WifiConfig":4,"WifiPower":16.0}}
00:02:58.423 RSL: STATUS6 = {"StatusMQT":{"MqttHost":"","MqttPort":1883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_AFAE54","MqttUser":"DVES_USER","MqttCount":0,"MAX_PACKET_SIZE":1200,"KEEPALIVE":30,"SOCKET_TIMEOUT":4}}
00:02:58.442 RSL: STATUS7 = {"StatusTIM":{"UTC":"1970-01-01T00:02:58Z","Local":"1970-01-01T00:02:58","StartDST":"1970-01-01T00:00:00","EndDST":"1970-01-01T00:00:00","Timezone":"+00:00","Sunrise":"20:13","Sunset":"05:47"}}
00:02:58.461 RSL: STATUS10 = {"StatusSNS":{"Time":"1970-01-01T00:02:58"}}
00:02:58.471 RSL: STATUS11 = {"StatusSTS":{"Time":"1970-01-01T00:02:58","Uptime":"0T00:02:59","UptimeSec":179,"Heap":141,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":0,"Berry":{"HeapUsed":4,"Objects":47},"POWER":"ON","Wifi":{"AP":1,"SSId":"OpenWrt","BSSId":"A0:F3:C1:99:1B:3D","Channel":11,"Mode":"HT40","RSSI":96,"Signal":-52,"LinkCount":1,"Downtime":"0T00:00:04"}}}
```
- [x] Set `weblog` to 4 and then, when you experience your issue, provide the output of the Console log:
```
00:01:54.830 CMD: weblog 4
00:01:54.832 RSL: RESULT = {"WebLog":4}
00:01:55.846 CFG: Saved, Count 101, Bytes 4096
00:01:57.656 SRC: WebGui from 192.168.7.109
00:01:57.660 RSL: RESULT = {"POWER":"OFF"}
00:01:57.661 RSL: POWER = OFF
00:01:57.816 CFG: Saved, Count 102, Bytes 4096
00:01:59.524 SRC: WebGui from 192.168.7.109
00:01:59.528 RSL: RESULT = {"POWER":"ON"}
00:01:59.529 RSL: POWER = ON
00:01:59.837 CFG: Saved, Count 103, Bytes 4096
00:02:07.944 WIF: Checking connection...
00:02:14.247 CMD: weblog 2
00:02:14.251 RSL: RESULT = {"WebLog":2}
```

### TO REPRODUCE
1. Flash Tasmota32 onto a Sonoff Dual R3 device. (Probably this issue is present also in every other similar configuration)
2. Configure GPIO14 as `Relay  1`  and GPIO27 as `Relay_i  1`.
3. Observe that the relays do not function as expected.

### EXPECTED BEHAVIOUR
I expect the relays operate in opposite states.
When GPIO14 is ON, GPIO27 should be inverted (OFF) and vice-versa.
Every time I press the "Toggle" button on web interface, I expect both relays change their states, always one opposite the other.

### ADDITIONAL CONTEXT
Configuring GPIO14 as `Relay  1`  and GPIO27 as `Led_i  1` brings the device to work as I want.
However, this is not the configuration I expected to use. Could someone explain me why the solution with Relay/Relay_i doesn't work? I haven't found any answers online.

[Issue 8857](https://github.com/arendst/Tasmota/issues/8857) seems very similar to mine, but has been marked as a "question" and closed probably because it lacks information.

[image]

sfromis, 2025-02-06:
You cannot configure both relay pins with the same index; you need a proper config instead. Using the `Interlock` command/feature sounds like what you're after.

TheFax, 2025-02-06:
Thank you @sfromis for your advise.
Unfortunately 'interlock' is different from what I need.

Could you please help me to find the documentation that prohibits assigning the same index to multiple relay pins?