arendst/Tasmota #22974: Dummy power meter does not work in tasmota32 | option A2

garyh2626, 2025-02-09:
### PROBLEM DESCRIPTION
_A clear and concise description of what the problem is._

Dummy power meter does not work in tasmota32 | option A2 

Program Version | 14.4.1 (release-tasmota32)
Precompiled binaries upgraded via OTA
Module configured as dummy energy monitor using option A 2 on GPIO5

See attached images below. none of the commands powerset, voltageset, currentset and frequencyset work as they do using older ESP8266
also commands powercal, voltagecal, currentcaland frequencycal do not work as they do using older ESP8266

19:43:19.549 CMD: powerset
19:43:19.552 RSL: RESULT = {"Command":"Error","Input":"POWERSET"}
19:43:23.641 CMD: currentset
19:43:23.644 RSL: RESULT = {"Command":"Error","Input":"CURRENTSET"}
19:43:27.728 CMD: voltageset
19:43:27.731 RSL: RESULT = {"Command":"Error","Input":"VOLTAGESET"}
19:43:35.522 CMD: frequencyset
19:43:35.525 RSL: RESULT = {"Command":"Error","Input":"FREQUENCYSET"}

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [ ] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [ ] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [ ] Searched the problem in [discussions](https://github.com/arendst/Tasmota/discussions)
- [ ] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [ ] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [ ] Device used (e.g., Sonoff Basic): _____
- [ ] Tasmota binary firmware version number used: _____
  - [ ] Pre-compiled
  - [ ] Self-compiled
- [ ] Flashing tools used: _____
- [ ] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```lua
  Configuration output here:

10:44:54.742 CMD: Backlog Template; Module; GPIO 255
10:44:54.763 RSL: RESULT = {"NAME":"ESP32-DevKit","GPIO":[1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,0,1,1,1,0,1,1,1,0,0,0,0,1,1,1,1,1,0,0,1],"FLAG":0,"BASE":1}
10:44:54.961 RSL: RESULT = {"Module":{"1":"ESP32-DevKit"}}
10:44:55.169 RSL: RESULT = {"GPIO0":{"0":"None"},"GPIO1":{"0":"None"},"GPIO2":{"0":"None"},"GPIO3":{"0":"None"},"GPIO4":{"0":"None"},"GPIO5":{"6209":"Option A2"},"GPIO6":{"0":"None"},"GPIO7":{"0":"None"},"GPIO8":{"0":"None"},"GPIO9":{"0":"None"},"GPIO10":{"0":"None"},"GPIO11":{"0":"None"},"GPIO12":{"0":"None"},"GPIO13":{"0":"None"},"GPIO14":{"0":"None"},"GPIO15":{"0":"None"},"GPIO16":{"0":"None"},"GPIO17":{"0":"None"},"GPIO18":{"0":"None"},"GPIO19":{"0":"None"},"GPIO20":{"0":"None"},"GPIO21":{"0":"None"},"GPIO22":{"0":"None"},"GPIO23":{"0":"None"},"GPIO24":{"0":"None"},"GPIO25":{"0":"None"},"GPIO26":{"0":"None"},"GPIO27":{"0":"None"},"GPIO32":{"0":"None"},"GPIO33":{"0":"None"},"GPIO34":{"0":"None"},"GPIO35":{"0":"None"},"GPIO36":{"0":"None"},"GPIO37":{"0":"None"},"GPIO38":{"0":"None"},"GPIO39":{"0":"None"}}

```
- [ ] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```lua
  Rules output here:

10:45:46.949 CMD: backlog rule1; rule2; rule3
10:45:46.967 RSL: RESULT = {"Rule1":{"State":"ON","Once":"OFF","StopOnError":"OFF","Length":106,"Free":405,"Rules":"ON SYSTEM#BOOT DO Backlog baudrate 57600; seriallog 2; voltageset 120; frequencyset 60; serialsend 1 ENDON"}}
10:45:47.169 RSL: RESULT = {"Rule2":{"State":"ON","Once":"OFF","StopOnError":"OFF","Length":119,"Free":392,"Rules":"ON SerialReceived#Current DO Backlog currentset %value% ENDON ON SerialReceived#Power DO Backlog powerset %value% ENDON"}}
10:45:47.371 RSL: RESULT = {"Rule3":{"State":"OFF","Once":"OFF","StopOnError":"OFF","Length":0,"Free":511,"Rules":""}}

```
- [ ] Provide the output of this command: `Status 0`:
```lua
  STATUS 0 output here:

10:46:16.031 CMD: status 0
10:46:16.035 RSL: STATUS = {"Status":{"Module":1,"DeviceName":"Tasmota","FriendlyName":["Tasmota"],"Topic":"tasmota_303070","ButtonTopic":"0","Power":"1","PowerLock":"0","PowerOnState":3,"LedState":0,"LedMask":"FFFF","SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0,"InfoRetain":0,"StateRetain":0,"StatusRetain":0}}
10:46:16.042 RSL: STATUS1 = {"StatusPRM":{"Baudrate":57600,"SerialConfig":"8N1","GroupTopic":"tasmotas","OtaUrl":"http://ota.tasmota.com/tasmota32/release-14.4.1/tasmota32.bin","RestartReason":"Software reset CPU","Uptime":"0T00:03:10","StartupUTC":"2025-02-09T15:43:06","Sleep":50,"CfgHolder":4617,"BootCount":93,"BCResetTime":"2025-02-07T20:35:55","SaveCount":377}}
10:46:16.048 RSL: STATUS2 = {"StatusFWR":{"Version":"14.4.1(release-tasmota32)","BuildDateTime":"2024-12-15T13:33:17","Core":"3_1_0","SDK":"5.3.2","CpuFrequency":160,"Hardware":"ESP32-D0WDQ6 v1.0","CR":"390/699"}}
10:46:16.054 RSL: STATUS3 = {"StatusLOG":{"SerialLog":0,"WebLog":2,"MqttLog":0,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["IOTNET",""],"TelePeriod":300,"Resolution":"558180C0","SetOption":["00408101","2805C80001000600003C5A0A192800000000","00000080","00006000","00004000","00000000"]}}
10:46:16.074 RSL: STATUS4 = {"StatusMEM":{"ProgramSize":2017,"Free":862,"Heap":143,"StackLowMark":3,"PsrMax":0,"PsrFree":0,"ProgramFlashSize":4096,"FlashSize":4096,"FlashChipId":"1640D8","FlashFrequency":40,"FlashMode":"DIO","Features":["0809","9F9AD7DF","0015A001","B7F7BFCF","05DA9BC4","E0360DC7","480840D2","20200000","D4BC482D","810A80F1","00000014"],"Drivers":"1,!2,!3,!4,!5,!7,!8,9,10,11,!12,!14,!16,!17,!20,!21,!24,26,!27,29,!34,!35,38,50,52,!59,!60,62,!63,!66,!67,!68,!73,!75,82,!86,!87,!88,!121","Sensors":"1,2,3,5,6,7,8,9,10,11,12,13,14,15,17,18,19,20,21,22,26,31,34,37,39,40,42,43,45,51,52,55,56,58,59,64,66,67,74,85,92,95,98,103,105,109,127","I2CDriver":"7,8,9,10,11,12,13,14,15,17,18,20,24,29,31,36,41,42,44,46,48,58,62,65,69,76,77,82,89"}}
10:46:16.085 RSL: STATUS5 = {"StatusNET":{"Hostname":"tasmota-303070-4208","IPAddress":"192.168.10.218","Gateway":"192.168.10.1","Subnetmask":"255.255.254.0","DNSServer1":"8.8.8.8","DNSServer2":"4.2.2.4","Mac":"3C:61:05:30:30:70","IP6Global":"","IP6Local":"fe80::3e61:5ff:fe30:3070%st1","Ethernet":{"Hostname":"","IPAddress":"0.0.0.0","Gateway":"0.0.0.0","Subnetmask":"0.0.0.0","DNSServer1":"8.8.8.8","DNSServer2":"4.2.2.4","Mac":"00:00:00:00:00:00","IP6Global":"","IP6Local":""},"Webserver":2,"HTTP_API":1,"WifiConfig":4,"WifiPower":17.0}}
10:46:16.097 RSL: STATUS7 = {"StatusTIM":{"UTC":"2025-02-09T15:46:16Z","Local":"2025-02-09T10:46:16","StartDST":"2025-03-09T02:00:00","EndDST":"2025-11-02T02:00:00","Timezone":99,"Sunrise":"02:07","Sunset":"12:00"}}
10:46:16.107 RSL: STATUS10 = {"StatusSNS":{"Time":"2025-02-09T10:46:16"}}
10:46:16.117 RSL: STATUS11 = {"StatusSTS":{"Time":"2025-02-09T10:46:16","Uptime":"0T00:03:10","UptimeSec":190,"Heap":145,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":0,"Berry":{"HeapUsed":4,"Objects":42},"Wifi":{"AP":1,"SSId":"IOTNET","BSSId":"F8:1A:67:CA:94:CD","Channel":1,"Mode":"11g","RSSI":70,"Signal":-65,"LinkCount":1,"Downtime":"0T00:00:04"}}}

```
[... 43 lines cut ...]
10:53:40.266 CMD: backlog powerset; currentset; voltageset; frequencyset
10:53:40.289 RSL: RESULT = {"Command":"Unknown","Input":"POWERSET"}
10:53:40.490 RSL: RESULT = {"Command":"Unknown","Input":"CURRENTSET"}
10:53:40.692 RSL: RESULT = {"Command":"Unknown","Input":"VOLTAGESET"}
10:53:40.893 RSL: RESULT = {"Command":"Unknown","Input":"FREQUENCYSET"}

### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._

ESP32 Option A2

[image]

[image]

ESP32 Option A9 does give dummy power monitor on home page but still power calibration commands not available

[image]

[image]

### ADDITIONAL CONTEXT
_Add any other context about the problem here._

**(Please, remember to close the issue when the problem has been addressed)**

barbudor, 2025-02-09:
```
10:44:55.169 RSL: RESULT = {"GPIO0":{"0":"None"},"GPIO1":{"0":"None"},"GPIO2":{"0":"None"},"GPIO3":{"0":"None"},"GPIO4":{"0":"None"},"GPIO5":{"6209":"Option A2"},"GPIO6":{"0":"None"},"GPIO7":{"0":"None"},"GPIO8":{"0":"None"},"GPIO9":{"0":"None"},"GPIO10":{"0":"None"},"GPIO11":{"0":"None"},"GPIO12":{"0":"None"},"GPIO13":{"0":"None"},"GPIO14":{"0":"None"},"GPIO15":{"0":"None"},"GPIO16":{"0":"None"},"GPIO17":{"0":"None"},"GPIO18":{"0":"None"},"GPIO19":{"0":"None"},"GPIO20":{"0":"None"},"GPIO21":{"0":"None"},"GPIO22":{"0":"None"},"GPIO23":{"0":"None"},"GPIO24":{"0":"None"},"GPIO25":{"0":"None"},"GPIO26":{"0":"None"},"GPIO27":{"0":"None"},"GPIO32":{"0":"None"},"GPIO33":{"0":"None"},"GPIO34":{"0":"None"},"GPIO35":{"0":"None"},"GPIO36":{"0":"None"},"GPIO37":{"0":"None"},"GPIO38":{"0":"None"},"GPIO39":{"0":"None"}}
```

Where is your relay ?
There no energy monitoring without a relay

garyh2626, 2025-02-09:
Thank you so much, that did it.

I did try adding a relay earlier without success. This it worked perfectly as it did on older ESP8266.
Only difference is the ESP8266 automatically adds the relay when config module option A2

ESP32 needs option A9 and manual adding the relay.

Thank you again for your quick response on this.

13:35:00.856 CMD: Backlog Template; Module; GPIO 255
13:35:00.881 RSL: RESULT = {"NAME":"ESP32-DevKit","GPIO":[1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,1,0,1,1,1,0,1,1,1,0,0,0,0,1,1,1,1,1,0,0,1],"FLAG":0,"BASE":1}
13:35:01.077 RSL: RESULT = {"Module":{"1":"ESP32-DevKit"}}
13:35:01.287 RSL: RESULT = {"GPIO0":{"0":"None"},"GPIO1":{"0":"None"},"GPIO2":{"0":"None"},"GPIO3":{"0":"None"},"GPIO4":{"0":"None"},"GPIO5":{"6209":"Option A2"},"GPIO6":{"0":"None"},"GPIO7":{"0":"None"},"GPIO8":{"0":"None"},"GPIO9":{"224":"Relay1"},"GPIO10":{"0":"None"},"GPIO11":{"0":"None"},"GPIO12":{"0":"None"},"GPIO13":{"0":"None"},"GPIO14":{"0":"None"},"GPIO15":{"0":"None"},"GPIO16":{"0":"None"},"GPIO17":{"0":"None"},"GPIO18":{"0":"None"},"GPIO19":{"0":"None"},"GPIO20":{"0":"None"},"GPIO21":{"0":"None"},"GPIO22":{"0":"None"},"GPIO23":{"0":"None"},"GPIO24":{"0":"None"},"GPIO25":{"0":"None"},"GPIO26":{"0":"None"},"GPIO27":{"0":"None"},"GPIO32":{"0":"None"},"GPIO33":{"0":"None"},"GPIO34":{"0":"None"},"GPIO35":{"0":"None"},"GPIO36":{"0":"None"},"GPIO37":{"0":"None"},"GPIO38":{"0":"None"},"GPIO39":{"0":"None"}}
13:36:15.917 RSL: STATE = {"Time":"2025-02-09T13:36:15","Uptime":"0T00:05:07","UptimeSec":307,"Heap":146,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":0,"Berry":{"HeapUsed":4,"Objects":46},"POWER":"ON","Wifi":{"AP":1,"SSId":"IOTNET","BSSId":"F8:1A:67:CA:94:CD","Channel":1,"Mode":"11g","RSSI":78,"Signal":-61,"LinkCount":1,"Downtime":"0T00:00:03"}}
13:36:15.928 RSL: SENSOR = {"Time":"2025-02-09T13:36:15","ENERGY":{"TotalStartTime":"2025-02-07T18:59:29","Total":0.011,"Yesterday":0.000,"Today":0.011,"Period":10,"Power":125,"ApparentPower":125,"ReactivePower":0,"Factor":1.00,"Frequency":60,"Voltage":120,"Current":0.035}}

13:34:24.355 CMD: backlog powerset; currentset; voltageset; frequencyset
13:34:24.374 RSL: RESULT = {"PowerSetCal":12530}
13:34:24.574 RSL: RESULT = {"CurrentSetCal":3500}
13:34:24.775 RSL: RESULT = {"VoltageSetCal":12000}
13:34:24.976 RSL: RESULT = {"FrequencySetCal":6000}

closed 2025-02-09 by garyh2626

barbudor, 2025-02-09:
And there is no good reason to use both at the same time. It probably would do weird things. 

A2 is the dummy driver (esp8266 or esp32)
A9 allows to write a energy driver in Berry (only esp32)