arendst/Tasmota #22981: Second GPIO with WS2812 LED on same index doesn't work after power cut

TomTom3112, 2025-02-09:
### PROBLEM DESCRIPTION
I use a Shelly Plus Plug S. Two WS2812 LED's are internaly connected to GPIO 25, two additional to GPIO 26.
Template is {"NAME":"Shelly Plus Plug S","GPIO":[0,0,0,0,224,0,33,2720,0,0,0,0,0,0,0,2624,0,0,2656,0,544,1377,1376,0,0,0,0,0,0,4736,0,0,0,0,0,0],"FLAG":0,"BASE":1}

When I change GPIO 26 do Index 1, I can change all LED's at the same time using the Webinterface as well as MQTT (HA).
That also works fine after a Tasmota restart.

When I disconnect the plug from power and reconnect it, only LED on GPIO 25 works, LED on GPIO 26 stays dark.
Switching on of the GPIO's to Index 2, alls LED are working fine again - until the next power cut.

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [x] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [x] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [x] Searched the problem in [discussions](https://github.com/arendst/Tasmota/discussions)
- [x] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [x] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [x] Device used (e.g., Sonoff Basic): Shelly Plus Plug S
- [x] Tasmota binary firmware version number used: 14.4.1
  - [x] Pre-compiled
  - [ ] Self-compiled
- [x] Flashing tools used: mgos32-to-tasmota32
- [ ] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```lua
  Configuration output here:
20:44:41.499 MQT: stat/Steckdose/Heizlüfter/RESULT = {"NAME":"Shelly Plus Plug S","GPIO":[0,0,0,0,224,0,33,2720,0,0,0,0,0,0,0,2624,0,0,2656,0,544,1377,1376,0,0,0,0,0,0,4736,0,0,0,0,0,0],"FLAG":0,"BASE":1}
20:44:41.743 MQT: stat/Steckdose/Heizlüfter/RESULT = {"Module":{"0":"Shelly Plus Plug S"}}
20:44:41.997 MQT: stat/Steckdose/Heizlüfter/RESULT = {"GPIO0":{"0":"None"},"GPIO1":{"0":"None"},"GPIO2":{"0":"None"},"GPIO3":{"0":"None"},"GPIO4":{"224":"Relay1"},"GPIO5":{"0":"None"},"GPIO6":{"0":"None"},"GPIO7":{"0":"None"},"GPIO8":{"0":"None"},"GPIO9":{"33":"Button2"},"GPIO10":{"2720":"BL0937 CF"},"GPIO11":{"0":"None"},"GPIO12":{"0":"None"},"GPIO13":{"0":"None"},"GPIO14":{"0":"None"},"GPIO15":{"0":"None"},"GPIO16":{"0":"None"},"GPIO17":{"0":"None"},"GPIO18":{"0":"None"},"GPIO19":{"2624":"HLWBL SEL_i"},"GPIO20":{"0":"None"},"GPIO21":{"0":"None"},"GPIO22":{"2656":"HLWBL CF1"},"GPIO23":{"0":"None"},"GPIO24":{"544":"LedLink"},"GPIO25":{"1377":"WS28122"},"GPIO26":{"1376":"WS28121"},"GPIO27":{"0":"None"},"GPIO32":{"0":"None"},"GPIO33":{"4736":"ADC Temp1"},"GPIO34":{"0":"None"},"GPIO35":{"0":"None"},"GPIO36":{"0":"None"},"GPIO37":{"0":"None"},"GPIO38":{"0":"None"},"GPIO39":{"0":"None"}}

```
- [ ] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```lua
  Rules output here:

```
- [ ] Provide the output of this command: `Status 0`:
```lua
  STATUS 0 output here:
20:45:41.364 MQT: stat/Steckdose/Heizlüfter/STATUS = {"Status":{"Module":0,"DeviceName":"Steckdose Heizlüfter","FriendlyName":["Steckdose Heizlüfter",""],"Topic":"Steckdose/Heizlüfter","ButtonTopic":"0","Power":"10","PowerLock":"00","PowerOnState":3,"LedState":1,"LedMask":"FFFF","SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0,"InfoRetain":0,"StateRetain":0,"StatusRetain":0}}
20:45:41.376 MQT: stat/Steckdose/Heizlüfter/STATUS1 = {"StatusPRM":{"Baudrate":115200,"SerialConfig":"8N1","GroupTopic":"tasmotas","OtaUrl":"http://ota.tasmota.com/tasmota32/release/tasmota32solo1.bin","RestartReason":"Vbat power on reset","Uptime":"0T00:52:25","StartupUTC":"2025-02-09T18:53:16","Sleep":50,"CfgHolder":4617,"BootCount":34,"BCResetTime":"2025-02-08T18:38:54","SaveCount":228}}
20:45:41.394 MQT: stat/Steckdose/Heizlüfter/STATUS2 = {"StatusFWR":{"Version":"14.4.1(release-solo1)single-core","BuildDateTime":"2024-12-15T13:33:16","Core":"3_1_0","SDK":"5.3.2","CpuFrequency":240,"Hardware":"ESP32-U4WDH-D v3.1","CR":"480/699"}}
20:45:41.419 MQT: stat/Steckdose/Heizlüfter/STATUS3 = {"StatusLOG":{"SerialLog":2,"WebLog":2,"MqttLog":0,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["#####",""],"TelePeriod":15,"Resolution":"558180C0","SetOption":["00008009","2805C80001000600003C5A0A192800000000","00000080","00006000","00004000","00000000"]}}
20:45:41.442 MQT: stat/Steckdose/Heizlüfter/STATUS4 = {"StatusMEM":{"ProgramSize":1979,"Free":772,"Heap":150,"StackLowMark":3,"PsrMax":0,"PsrFree":0,"ProgramFlashSize":4096,"FlashSize":4096,"FlashChipId":"164020","FlashFrequency":80,"FlashMode":"DIO","Features":["0809","9F9AD7DF","0015A001","B7F7BFCF","05DA9BC4","E0360DC7","480840D2","20200000","D4BC482D","810A80F1","00000014"],"Drivers":"1,2,3,4,!5,7,!8,9,10,11,12,!14,!16,!17,!20,!21,!24,26,!27,29,!34,!35,38,50,52,!59,!60,62,!63,!66,!67,!68,!73,!75,82,!86,!87,!88,121","Sensors":"1,2,3,5,6,7,8,9,10,11,12,13,14,15,17,18,19,20,21,22,26,31,34,37,39,40,42,43,45,51,52,55,56,58,59,64,66,67,74,85,92,95,98,103,105,109,127","I2CDriver":"7,8,9,10,11,12,13,14,15,17,18,20,24,29,31,36,41,42,44,46,48,58,62,65,69,76,77,82,89"}}
20:45:41.453 MQT: stat/Steckdose/Heizlüfter/STATUS5 = {"StatusNET":{"Hostname":"Heizl--fter","IPAddress":"192.168.##.###","Gateway":"192.168.##.#","Subnetmask":"255.255.255.0","DNSServer1":"192.168.##.#","DNSServer2":"2a05:fa40:19e:8000:cece:1eff:feb3:99c1","Mac":"E4:65:B8:5D:DE:D0","IP6Global":"2a05:fa40:19e:8000:e665:b8ff:fe5d:ded0","IP6Local":"fe80::e665:b8ff:fe5d:ded0%st1","Ethernet":{"Hostname":"","IPAddress":"0.0.0.0","Gateway":"0.0.0.0","Subnetmask":"0.0.0.0","DNSServer1":"192.168.##.#","DNSServer2":"2a05:fa40:19e:8000:cece:1eff:feb3:99c1","Mac":"00:00:00:00:00:00","IP6Global":"","IP6Local":""},"Webserver":2,"HTTP_API":1,"WifiConfig":4,"WifiPower":17.0}}
20:45:41.458 MQT: stat/Steckdose/Heizlüfter/STATUS6 = {"StatusMQT":{"MqttHost":"192.168.##.##","MqttPort":1883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_5DDED0","MqttUser":"DVES_USER","MqttCount":1,"MAX_PACKET_SIZE":1200,"KEEPALIVE":30,"SOCKET_TIMEOUT":4}}
20:45:41.464 MQT: stat/Steckdose/Heizlüfter/STATUS7 = {"StatusTIM":{"UTC":"2025-02-09T19:45:41Z","Local":"2025-02-09T20:45:41","StartDST":"2025-03-30T02:00:00","EndDST":"2025-10-26T03:00:00","Timezone":"+01:00","Sunrise":"08:07","Sunset":"18:00"}}
20:45:41.469 MQT: stat/Steckdose/Heizlüfter/STATUS9 = {"StatusPTH":{"PowerDelta":0,"PowerLow":0,"PowerHigh":0,"VoltageLow":0,"VoltageHigh":0,"CurrentLow":0,"CurrentHigh":0,"MaxPower":0,"MaxPowerHold":10,"MaxPowerWindow":30,"MaxEnergy":0,"MaxEnergyStart":0}}
20:45:41.492 MQT: stat/Steckdose/Heizlüfter/STATUS10 = {"StatusSNS":{"Time":"2025-02-09T20:45:41","ANALOG":{"Temperature1":3.6},"ENERGY":{"TotalStartTime":"2025-02-08T18:44:46","Total":0.179,"Yesterday":0.000,"Today":0.179,"Period":0,"Power":0,"ApparentPower":0,"ReactivePower":0,"Factor":0.00,"Voltage":0,"Current":0.000},"TempUnit":"C"}}
20:45:41.518 MQT: stat/Steckdose/Heizlüfter/STATUS11 = {"StatusSTS":{"Time":"2025-02-09T20:45:41","Uptime":"0T00:52:25","UptimeSec":3145,"Heap":147,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":22,"MqttCount":1,"Berry":{"HeapUsed":23,"Objects":306},"POWER1":"OFF","POWER2":"ON","Dimmer":100,"Color":"FF0000","HSBColor":"0,100,100","Channel":[100,0,0],"Scheme":0,"Width":1,"Fade":"OFF","Speed":1,"LedTable":"ON","Wifi":{"AP":1,"SSId":"######","BSSId":"##:##:##:##:##:##","Channel":1,"Mode":"HT40","RSSI":100,"Signal":-42,"LinkCount":1,"Downtime":"0T00:00:04"}}}

```
- [ ] Set `weblog` to 4 and then, when you experience your issue, provide the output of the Console log:
```lua
  Console output here:

```

### TO REPRODUCE
_Steps to reproduce the behavior:_

### EXPECTED BEHAVIOUR
All LED's connected to both GPIO's working simultanious also after power cut

### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._

### ADDITIONAL CONTEXT
_Add any other context about the problem here._

**(Please, remember to close the issue when the problem has been addressed)**

s-hadinger, 2025-02-09:
You can only drive one WS2812 led strip with native Tasmota commands. To drive the second one, you need Berry scripting.

Leds keep their status after a restart, so indeed if you change the channel number, it will maintant the old state and change only 1 led strip.

TomTom3112, 2025-02-09:
Yes, but when I set both GPIO's to the same Index for the first or back to diffent Index after a power cut I can change all LED's on both GPIO's at the same time. Color, Brightness, ... - every setting works multiple times.
I'm with you, that according to the documentation I can only drive one WS2812 strip. But in practice I can control two LED strips on two GPIOs simultaneously - also afer a restart. But not after a power cut.

s-hadinger, 2025-02-09:
Setting two GPIO with the same WS2812 channel is not supported. The fact it worked as you wanted is pure chance.

It may be something we support in the future, if we figure out how to configure the CPIO matrix in ESP32. But I haven't looked at it at all. And it's not a priority for now.

If anyone wants to take a look?

TomTom3112, 2025-02-09:
I know, it is not supported - but it works. I doublechecked it several times, thats why I'm confused.
The way, how it works:

GPIO25 set to Ch1, GPIO26 set to Ch2 => I can change only 2 of the four LED as expected.

Switch GPIO26 to Ch2 and GPIO25 to CH1 => I can change all 4 LED at the same time - not as expected, but works. Also after restart.
Restart, change color, change brightness, change color again, restart... everything works with all 4 LED's.

Disconnect Plug from outlet and reconnect => only 2 LED's are working and changeable again.
Switch again Channels of GPIO25 + 26 = I can use all four LED's as described (and not expected ;-) )...