arendst/Tasmota #22992: Tasmota Bluetooth firmware cannot be installed on Heltec Wifi Kit 32

fkroepfl, 2025-02-12:
### PROBLEM DESCRIPTION
_A clear and concise description of what the problem is._

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [x] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [x] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [x] Searched the problem in [discussions](https://github.com/arendst/Tasmota/discussions)
- [x] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [-] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [x] Problem is not scripter related, in this case open a discussion and tag @gemu2015
- [x] Device used (e.g., Sonoff Basic): Heltec WiFi Kit 32
- [x] Tasmota binary firmware version number used: 14.0.0 (release-safeboot)
  - [x] Pre-compiled
  - [-] Self-compiled
- [x] Flashing tools used: built-in
### REQUESTED INFORMATION

*Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!*

```lua
  Configuration output here:
  (Please fill in the actual output)
- SAFEBOOT
  WiFi Kit 32
  Tasmota
  Program Version	14.0.0 (release-safeboot)
  Build Date & Time	2024-05-14T13:51:29
  Core/SDK Version	3\_0\_0/5.1.3.240430
  Uptime	0T00:14:52
  Flash Write Count	192
  Boot Count	51
  Restart Reason	Software reset CPU
  Friendly Name 1	Tasmota
  AP1 Information	SSId Braunetzwerk
  RSSI 54%, -73 dBm
  Mode HT20
  Channel 11
  BSSId 04:95\:E6\:DB:65:41
  Hostname	tasmota-E5B0EC-4332
  IPv6 Global (WiFi)	
  IPv6 Local (WiFi)	
  MAC Address	48:27\:E2\:E5\:B0\:EC
  IP Address (WiFi)	
  Gateway	
  Subnet Mask	
  DNS Server1	
  DNS Server2	
  HTTP API	Enabled
  MQTT Host&#x9;
  MQTT Port	1883
  MQTT TLS	Disabled
  MQTT User	DVES\_USER
  MQTT Client	DVES\_E5B0EC
  MQTT Topic	tasmota\_%06X
  MQTT Group Topic 1	cmnd/tasmotas/
  MQTT Full Topic	cmnd/tasmota\_E5B0EC/
  MQTT Fallback Topic	cmnd/DVES\_E5B0EC\_fb/
  MQTT No Retain	Disabled
  ESP Chip Id	15053036 (ESP32-S3 v0.2)
  Flash Chip Id	0x1740C8 (QIO)
  Flash Size	8192 KB
  Program Flash Size	8192 KB
  Program Size	755 KB
  Free Program Space	76 KB
  Free Memory	270.9 KB (frag. 5%)
  Partition safeboot\*	832 KB (used 90%)
  Partition app0	2880 KB
  Partition fs	4416 KB
  Main Menu
  Tasmota 14.0.0 (release-safeboot) by Theo Arends
```

```lua
  Rules output here:
  (Please fill in the actual output)
n.a.
```

```lua
  STATUS 0 output here:
  (Please fill in the actual output)
  11:44:55.310 CMD: STATUS 0
  11:44:55.313 RSL: STATUS = {"Status":{"Module":0,"DeviceName":"Tasmota","FriendlyName":["Tasmota"],"Topic":"tasmota\_E5B0EC","ButtonTopic":"0","Power":"1","PowerLock":"0","PowerOnState":3,"LedState":1,"LedMask":"FFFF","SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0,"InfoRetain":0,"StateRetain":0,"StatusRetain":0}}
```

```lua
  Console output here:
  (Please fill in the actual output)
  11:44:55.314 RSL: STATUS1 = {"StatusPRM":{"Baudrate":115200,"SerialConfig":"8N1","GroupTopic":"tasmotas","OtaUrl":"[[http://ota.tasmota.com/tasmota32/release-14.4.1/tasmota32-bluetooth.bin","RestartReason":"Software](http://ota.tasmota.com/tasmota32/release-14.4.1/tasmota32-bluetooth.bin%22,%22RestartReason%22:%22Software)](http://ota.tasmota.com/tasmota32/release-14.4.1/tasmota32-bluetooth.bin","RestartReason":"Software) reset CPU","Uptime":"0T00:16:31","StartupUTC":"2025-02-12T10:28:24","Sleep":50,"CfgHolder":4617,"BootCount":51,"BCResetTime":"2024-05-30T13:02:00","SaveCount":192}}
  11:44:55.316 RSL: STATUS2 = {"StatusFWR":{"Version":"14.0.0(release-safeboot)","BuildDateTime":"2024-05-14T13:51:29","Core":"3\_0\_0","SDK":"5.1.3.240430","CpuFrequency":240,"Hardware":"ESP32-S3 v0.2","CR":"408/699"}}
  11:44:55.317 RSL: STATUS3 = {"StatusLOG":{"SerialLog":0,"WebLog":2,"MqttLog":0,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["Braunetzwerk",""],"TelePeriod":300,"Resolution":"558180C0","SetOption":["00408009","2805C80001000600003C5A0A192800000000","00000080","00006000","00004000","00000000"]}}
  11:44:55.320 RSL: STATUS4 = {"StatusMEM":{"ProgramSize":755,"Free":76,"Heap":270,"StackLowMark":4,"PsrMax":0,"PsrFree":0,"ProgramFlashSize":8192,"FlashSize":8192,"FlashChipId":"1740C8","FlashFrequency":80,"FlashMode":"QIO","Features":["0809","000005C0","00000003","00000002","00000000","00000000","00000000","00000000","00000000","00000000","00000000"],"Drivers":"1,2","Sensors":""}}
  11:44:55.322 RSL: STATUS5 = {"StatusNET":{"Hostname":"tasmota-E5B0EC-4332","IPAddress":"192.168.1.124","Gateway":"192.168.1.254","Subnetmask":"255.255.255.0","DNSServer1":"192.168.1.130","DNSServer2":"1.1.1.2","Mac":"48:27\:E2\:E5\:B0\:EC","IP6Global":"fd18\:ecff:7991:5ef3:4a27\:e2ff\:fee5\:b0ec","IP6Local":"fe80::4a27\:e2ff\:fee5\:b0ec%st1","Webserver":2,"HTTP\_API":1,"WifiConfig":4,"WifiPower":19.0}}
  11:44:55.324 RSL: STATUS6 = {"StatusMQT":{"MqttHost":"","MqttPort":1883,"MqttClientMask":"DVES\_%06X","MqttClient":"DVES\_E5B0EC","MqttUser":"DVES\_USER","MqttCount":0,"MAX\_PACKET\_SIZE":1200,"KEEPALIVE":30,"SOCKET\_TIMEOUT":4}}
  11:44:55.326 RSL: STATUS7 = {"StatusTIM":{"UTC":"2025-02-12T10:44:55Z","Local":"2025-02-12T11:44:55","StartDST":"2025-03-30T02:00:00","EndDST":"2025-10-26T03:00:00","Timezone":"+01:00"}}
  11:44:55.328 RSL: STATUS10 = {"StatusSNS":{"Time":"2025-02-12T11:44:55"}}
  11:44:55.332 RSL: STATUS11 = {"StatusSTS":{"Time":"2025-02-12T11:44:55","Uptime":"0T00:16:31","UptimeSec":991,"Heap":270,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":0,"POWER":"ON","Wifi":{"AP":1,"SSId":"Braunetzwerk","BSSId":"04:95\:E6\:DB:65:41","Channel":11,"Mode":"HT20","RSSI":58,"Signal":-71,"LinkCount":1,"Downtime":"0T00:00:04"}}}
```

### TO REPRODUCE

*Steps to reproduce the behavior:*

1. Start with **Tasmota 14.0.0** (or another existing version) installed on Heltec WiFi Kit 32.
2. Attempt an OTA upgrade using the following firmware: [[tasmota32-bluetooth.bin](http://ota.tasmota.com/tasmota32/release-14.4.1/tasmota32-bluetooth.bin)](http://ota.tasmota.com/tasmota32/release-14.4.1/tasmota32-bluetooth.bin)
3. The device switches to **SafeBoot** mode instead of running the updated firmware.
4. Retrying the upgrade from SafeBoot does not install the intended firmware but remains in SafeBoot mode.
5. Flashing [[tasmota32s3.bin](http://ota.tasmota.com/tasmota32/release-14.4.1/tasmota32s3.bin)](http://ota.tasmota.com/tasmota32/release-14.4.1/tasmota32s3.bin) works without issues.

### EXPECTED BEHAVIOR

The device should correctly update to **Tasmota 14.4.1 (Bluetooth variant)** without falling back to SafeBoot.

### SCREENSHOTS

*If applicable, add screenshots to help explain your problem.*

### ADDITIONAL CONTEXT

- The issue is specific to `tasmota32-bluetooth.bin`, while `tasmota32s3.bin` works correctly.
- SafeBoot mode suggests a potential incompatibility or issue with this particular firmware build on the **Heltec WiFi Kit 32**.
- Further debugging logs can be provided upon request.

Please investigate if there is an issue with the build or a hardware compatibility problem.

Thank you for your support!

sfromis, 2025-02-12:
When it is ESP32-S3, you can only use binaries built for that variant, meaning those where the name starts with `tasmota32s3-`; the ones for the original ESP32 (names starting with just `tasmota32-`) are not usable. No issue here.

fkroepfl, 2025-02-12:
> When it is ESP32-S3, you can only use binaries built for that variant, meaning those where the name starts with `tasmota32s3`; the ones for the original ESP32 (names starting with just `tasmota32` without s3/s2/c3/c6/c2) are not usable. No issue here.

Thank you very much for the clarification. I was not sufficiently aware of this dependency. 
In other words, the solution is self-compiling or using an ESP32 board with which the pre-compiled firmware works.

closed 2025-02-12 by fkroepfl

sfromis, 2025-02-12:
There are also "unofficial" development builds, including `tasmota32s3-bluetooth.bin`, in:
https://github.com/tasmota/install/tree/firmware/firmware/unofficial

fkroepfl, 2025-02-12:
> There are also "unofficial" development builds, including `tasmota32s3-bluetooth.bin`, in: https://github.com/tasmota/install/tree/firmware/firmware/unofficial

Seems to work, at least if you upload the file locally. Many thanks for the tip. 🙏

sfromis, 2025-02-12:
It should also work via OTA upgrade, as long as you make sure to "dive down" to the raw URL.

fkroepfl, 2025-02-12:
> It should also work via OTA upgrade, as long as you make sure to "dive down" to the raw URL.

That's clear, but there are currently no S3 variants to be found. However, I was able to find and download one in the afternoon.
I'm a bit confused now. 

Last commit date appr. 18:04
[image]

[image]

sfromis, 2025-02-12:
If you scroll down, you *should have* found S3 builds on the same page, like you already did when downloading. Don't use those C2 variants for S3.

**However**, right now there has been a mishap in the build system, not populating all the binaries, and therefore you are missing a lot of files. They'll probably be back by tomorrow.

fkroepfl, 2025-02-12:
All right. ‘Good things take time.’ 😉

sfromis, 2025-02-12:
Not a long time, the files are there now, from 17 minutes ago.

fkroepfl, 2025-02-12:
Sometimes it happens faster than one think. 😉