arendst/Tasmota #23001: Tasmota unable to flash zigbee coordinator firmware

darek-margas, 2025-02-13:
### PROBLEM DESCRIPTION
Tasmota for Sonoff zigbee bridge (for EFR chip) has ability to accept zigbee coordinator firmware uploaded via UI and to flash this firmware into coordinator. In recent Tasmota, if incompatible firmware exists in coordinator, Tasmota is unable to re-flash it. After boot, Tasmota does check zigbee and makes status 99 due to incompatibility. In that state, after uploading new coordinator firmware, Tasmota does confirm upload in console and then nothing happens. 
It is easy to repeat, just enough to flash one of protocol 12 versions (7.x).

In such bricked state, re-flashing Tasmota itself, vi serial, to old version 9.1, brings coordinator flashing ability back. I was able to recover coordinator by flashing compatible firmware version using Tasmota 9.1.
There must be some change between version 9 and 14, which makes flashing broken if status is 99

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [x] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [x] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [x] Searched the problem in [discussions](https://github.com/arendst/Tasmota/discussions)
- [x] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [x] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [x] Problem is not scripter related, in this case open a discussion and tag @gemu2015
- [x] Device used (e.g., Sonoff Basic): Sonoff ZB Bridge (ESP8266)
- [x] Tasmota binary firmware version number used: 14.4.1
  - [x] Pre-compiled
  - [ ] Self-compiled
- [x] Flashing tools used: ota and serial via tasmotizer
- [x] Provide the output of command: `Backlog Template; Module; GPIO 255`: 
```lua
  Configuration output here:
{"GPIO0":{"320":"Led_i1"},"GPIO1":{"3552":"Zigbee Tx"},"GPIO2":{"0":"None"},"GPIO3":{"3584":"Zigbee Rx"},"GPIO4":{"5312":"Zigbee Rst1"},"GPIO5":{"0":"None"},"GPIO9":{"0":"None"},"GPIO10":{"0":"None"},"GPIO12":{"640":"I2C SDA"},"GPIO13":{"576":"LedLink_i"},"GPIO14":{"608":"I2C SCL"},"GPIO15":{"0":"None"},"GPIO16":{"32":"Button1"},"GPIO17":{"0":"None"}}

```
- [ ] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```lua
  Rules output here:

```
- [ ] Provide the output of this command: `Status 0`:
```lua
  STATUS 0 output here:
08:09:25.372 MQT: stat/tasmota_9E30EB/STATUS = {"Status":{"Module":75,"DeviceName":"Sonoff Zigbee bridge","FriendlyName":["Tasmota"],"Topic":"tasmota_9E30EB","ButtonTopic":"0","Power":"1","PowerLock":"0","PowerOnState":3,"LedState":1,"LedMask":"FFFF","SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":1,"SensorRetain":1,"PowerRetain":1,"InfoRetain":1,"StateRetain":1,"StatusRetain":1}} (retained)
08:09:25.388 MQT: stat/tasmota_9E30EB/STATUS1 = {"StatusPRM":{"Baudrate":115200,"SerialConfig":"8N1","GroupTopic":"tasmotas","OtaUrl":"http://ota.tasmota.com/tasmota/release/tasmota-zbbridge.bin.gz","RestartReason":"Software/System restart","Uptime":"0T00:17:18","StartupUTC":"2025-02-13T20:52:07","Sleep":50,"CfgHolder":4617,"BootCount":9,"BCResetTime":"2021-10-28T07:11:00","SaveCount":27,"SaveAddress":"1FB000"}} (retained)
08:09:25.405 MQT: stat/tasmota_9E30EB/STATUS2 = {"StatusFWR":{"Version":"14.4.1(release-zbbridge)","BuildDateTime":"2024-12-15T13:29:52","Boot":31,"Core":"2_7_8","SDK":"2.2.2-dev(38a443e)","CpuFrequency":160,"Hardware":"ESP8266EX","CR":"407/699"}} (retained)
08:09:25.420 MQT: stat/tasmota_9E30EB/STATUS3 = {"StatusLOG":{"SerialLog":0,"WebLog":2,"MqttLog":0,"SysLog":2,"LogHost":"192.168.1.247","LogPort":514,"SSId":["IOT",""],"TelePeriod":300,"Resolution":"558180C0","SetOption":["000082A9","2805C80001000600003C5A00002800000000","000080C0","00206082","80044180","00000000"]}} (retained)
08:09:25.440 MQT: stat/tasmota_9E30EB/STATUS4 = {"StatusMEM":{"ProgramSize":747,"Free":1044,"Heap":16,"ProgramFlashSize":2048,"FlashSize":2048,"FlashChipId":"1540A1","FlashFrequency":40,"FlashMode":"DOUT","Features":["0809","0F1007CE","04400001","00000003","00000000","00000000","00020080","00200000","04000800","00000080","00000000"],"Drivers":"1,2,!4,7,9,10,12,!20,23,38,!41,50,!59,62,!68","Sensors":"1","I2CDriver":""}} (retained)
08:09:25.463 MQT: stat/tasmota_9E30EB/STATUS5 = {"StatusNET":{"Hostname":"zigbee-bridge","IPAddress":"192.168.16.120","Gateway":"192.168.16.1","Subnetmask":"255.255.255.0","DNSServer1":"192.231.203.3","DNSServer2":"192.231.203.132","Mac":"E8:DB:84:9E:30:EB","Webserver":2,"HTTP_API":1,"WifiConfig":4,"WifiPower":17.0}} (retained)
08:09:25.485 MQT: stat/tasmota_9E30EB/STATUS6 = {"StatusMQT":{"MqttHost":"192.168.16.210","MqttPort":8883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_9E30EB","MqttUser":"tasmota","MqttCount":1,"MAX_PACKET_SIZE":1200,"KEEPALIVE":30,"SOCKET_TIMEOUT":4}} (retained)
08:09:25.505 MQT: stat/tasmota_9E30EB/STATUS7 = {"StatusTIM":{"UTC":"2025-02-13T21:09:25Z","Local":"2025-02-14T08:09:25","StartDST":"2025-10-05T02:00:00","EndDST":"2025-04-06T02:00:00","Timezone":99,"Sunrise":"06:29","Sunset":"19:47"}} (retained)
08:09:25.522 MQT: stat/tasmota_9E30EB/STATUS10 = {"StatusSNS":{"Time":"2025-02-14T08:09:25"}} (retained)
08:09:25.539 MQT: stat/tasmota_9E30EB/STATUS11 = {"StatusSTS":{"Time":"2025-02-14T08:09:25","Uptime":"0T00:17:18","UptimeSec":1038,"Vcc":3.471,"Heap":16,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":41,"MqttCount":1,"Wifi":{"AP":1,"SSId":"IOT","BSSId":"F4:8C:EB:86:83:31","Channel":9,"Mode":"11n","RSSI":94,"Signal":-53,"LinkCount":1,"Downtime":"0T00:00:07"}}} (retained)

```
- [ ] Set `weblog` to 4 and then, when you experience your issue, provide the output of the Console log:
```lua
  Console output here:
14:09:08.111 ZIG: Resetting EZSP device

14:09:09.366 RSL: RESULT = {"ZbState":{"Status":1,"Message":"EFR32 EZSP booted","RestartReason":"Power-on","Code":2}}

14:09:09.421 RSL: RESULT = {"ZbState":{"Status":55,"Version":"7.3.1.0","Protocol":12,"Stack":2}}

14:09:09.422 RSL: RESULT = {"ZbState":{"Status":98,"Message":"Only EZSP protocol v8 is currently supported"}}

14:09:09.424 RSL: RESULT = {"ZbState":{"Status":99,"Message":"Abort"}}

14:09:09.424 ZIG: Abort

14:09:09.425 ZIG: Stopping (99)

14:10:12.020 UPL: File ncp-uart-nsw_6.7.9_115200.ota (192240 bytes)

14:10:14.453 UPP: Multicast (re)joined

14:10:14.454 UPL: Transfer 192240 bytes

14:10:14.454 UPL: Successful 192240 bytes

(with weblog 4)
00:33:19.856 HTP: Firmware Upgrade

00:33:23.740 WIF: Checking connection...

00:33:32.461 UPL: File ncp-uart-nsw_6.7.9_115200.ota (192240 bytes)

00:33:32.559 CFG: Saved to flash at 1FB, Count 22, Bytes 4096

00:33:32.576 UPL: File type 5

00:33:34.540 UPL: Progress 100 KB

00:33:36.189 UPP: Multicast (re)joined

00:33:36.190 UPL: Transfer 192240 bytes

00:33:36.191 UPL: Successful 192240 bytes

00:33:36.203 HTP: Upload transfer

```

### TO REPRODUCE
_Steps to reproduce the behavior:_
Flash coordinator with firmware of 7.x versions

### EXPECTED BEHAVIOUR
_A clear and concise description of what you expected to happen._
Has to flash firmware despite being in 99 status

### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._

### ADDITIONAL CONTEXT
_Add any other context about the problem here._

**(Please, remember to close the issue when the problem has been addressed)**

Jason2866, 2025-02-13:
Flashing every other firmwares as the provided in Tasmota repo will left the zigbee mcu in a not flash able state. Reason: the configuration of the reset gpio is not adopted to the zigbee bridge hardware config in firmwares found "somewhere"

closed 2025-02-13 by Jason2866

sfromis, 2025-02-13:
What's still interesting is that using an old Tasmota binary made it possible to reflash the EFR32MG21, without stalling like current firmware did.

darek-margas, 2025-02-13:
> Flashing every other firmwares as the provided in Tasmota repo will left the zigbee mcu in a not flash able state. Reason: the configuration of the reset gpio is not adopted to the zigbee bridge hardware config in firmwares found "somewhere"

It is in flashable state as Tasmota 9.1 DOES flash it. The coordinator works, this is not about bricked EFR chip, it is about situation where EFR chip works reporting protocol 12,causing Tasmota to go to status 99 and being unable to flash alternative firmware. If it was coordinator issue, Tasmota 9.1 would not help

darek-margas, 2025-02-13:
> What's still interesting is that using an old Tasmota binary made it possible to reflash the EFR32MG21, without stalling like current firmware did.

That's exactly why I believe it is a bug

Jason2866, 2025-02-13:
Anyways not supported Zigbee firmware which is causing the issue. Not supported use case. No  issue accepted.

darek-margas, 2025-02-13:
> Anyways not supported Zigbee firmware which is causing the issue. Not supported use case. No issue accepted.

The bug isn't caused by using any particular firmware, it is caused by an error the the code, apparently made after version 9.1. SYMPTOMS  are caused by loading unsupported firmware, bringing bug to the surface. It is impossible to tell span of influence without understanding why it is happening. It could be that such symptom mentioned is the only one caused by the bug, with known workaround, possibly not critical one. However, it might be that span of influence gets further and the same bug may cause, even unknown as yet, other symptoms and failures. I believe it is at least worth debugging to find how does it break before making a call if it is worth to be fixed. If not worth fixing, I believe it should be listed as known bug with known (now) workaround, for those who stumble.

s-hadinger, 2025-02-14:
Lots of good ideas. There is a bug, yes. But considering that this is to fix an unsupported firmware, there is a work-around, and I prefer to spend resources on TI/ZNP/CC2652P, I will not dedicate time to fix this bug.
You are welcome to send a pull request.

darek-margas, 2025-02-14:
> Lots of good ideas. There is a bug, yes. But considering that this is to fix an unsupported firmware, there is a work-around, and I prefer to spend resources on TI/ZNP/CC2652P, I will not dedicate time to fix this bug. You are welcome to send a pull request.

I see your point. I don't think I will do PR but I may find which commit broke this, should be easy to tell why once commit is known and make call if is worth to fix.

s-hadinger, 2025-02-14:
Definitely a good plan and most useful.

sfromis, 2026-01-14:
When testing some time ago, my experience was that [version 14.1.0](https://ota.tasmota.com/tasmota/release-14.1.0/tasmota-zbbridge.bin.gz) worked for downgrading EFR32MG21 firmware, while it failed with version 14.2.0

peter--s, 2026-01-14:
@sformis: I just upgraded to 12.2.0 again and downgrading did work, I must have made a mistake during my previous attempt. I will edit my comment.