arendst/Tasmota #23044: DeepSleep is not working with Rule1 5 setting.  Device does not stay ON but instead goes right back into DeepSleep after Wakeup.

LucidEye, 2025-02-21:
### PROBLEM DESCRIPTION
DeepSleep Wakeup rule is not working with Timers correctly when setting Rule1 5 (wakeup ONCE and then STAY ON).
My goal is to run an ESP32-CAM module using a solar panel/battery as a power source, so I need the ESP32-CAM to go to into DeepSleep at sunset to conserve battery power at night, and then come out of DeepSleep at Sunrise... I need this to happen every day.

When setting Rule1 Wakeup, then setting Rule1 5, then setting a Timer to wakeup the device from DeepSleep ONCE and NOT go back to sleep... this fails... instead of staying awake, the device goes back into DeepSleep Countdown immediately after waking up.

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [ X] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [ X] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [ X] Searched the problem in [discussions](https://github.com/arendst/Tasmota/discussions)
- [ X] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [ X] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [X] Problem is not scripter related, in this case open a discussion and tag @gemu2015
- [ X] Device used (e.g., Sonoff Basic): Generic ESP32-CAM module
- [ X] Tasmota binary firmware version number used: Tasmota32 WebCam 14.4.1
  - [X ] Pre-compiled
  - [ ] Self-compiled
- [ X] Flashing tools used: Tasmota Web Page Installer
- [ ] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```lua
  Configuration output here:

```
- [ ] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```lua
  Rules output here:
14:37:03.073 MQT: stat/tasmota-cam/RESULT = {"Rule1":{"State":"ON","Once":"ON","StopOnError":"OFF","Length":6,"Free":505,"Rules":"Wakeup"}}
14:37:03.302 MQT: stat/tasmota-cam/RESULT = {"Rule2":{"State":"OFF","Once":"OFF","StopOnError":"OFF","Length":0,"Free":511,"Rules":""}}
14:37:03.552 MQT: stat/tasmota-cam/RESULT = {"Rule3":{"State":"OFF","Once":"OFF","StopOnError":"OFF","Length":0,"Free":511,"Rules":""}}

```
- [ ] Provide the output of this command: `Status 0`:
```lua
  STATUS 0 output here:
14:37:33.363 MQT: stat/tasmota-cam/STATUS = {"Status":{"Module":0,"DeviceName":"Tasmota-CAM","FriendlyName":["Tasmota-CAM"],"Topic":"tasmota-cam","ButtonTopic":"0","Power":"0","PowerLock":"0","PowerOnState":3,"LedState":1,"LedMask":"FFFF","SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0,"InfoRetain":0,"StateRetain":0,"StatusRetain":0}}
14:37:33.396 MQT: stat/tasmota-cam/STATUS1 = {"StatusPRM":{"Baudrate":115200,"SerialConfig":"8N1","GroupTopic":"tasmotas","OtaUrl":"http://ota.tasmota.com/tasmota32/release/tasmota32-webcam.bin","RestartReason":"Vbat power on reset","Uptime":"0T00:01:44","StartupUTC":"2025-02-21T21:35:49","Sleep":50,"CfgHolder":4617,"BootCount":143,"BCResetTime":"2024-02-22T06:42:25","SaveCount":711}}
14:37:33.432 MQT: stat/tasmota-cam/STATUS2 = {"StatusFWR":{"Version":"14.4.1(release-webcam)","BuildDateTime":"2024-12-15T13:32:56","Core":"3_1_0","SDK":"5.3.2","CpuFrequency":240,"Hardware":"ESP32-D0WDQ6 v1.0","CR":"405/699"}}
14:37:33.456 MQT: stat/tasmota-cam/STATUS3 = {"StatusLOG":{"SerialLog":2,"WebLog":2,"MqttLog":0,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["LucidWerks",""],"TelePeriod":900,"Resolution":"558180C0","SetOption":["00008109","2805C80000000600003C5A00192800000000","000080A8","00406000","00004000","00000001"]}}
14:37:33.485 MQT: stat/tasmota-cam/STATUS4 = {"StatusMEM":{"ProgramSize":1297,"Free":1582,"Heap":117,"StackLowMark":4,"PsrMax":4096,"PsrFree":3309,"ProgramFlashSize":4096,"FlashSize":4096,"FlashChipId":"1640D8","FlashFrequency":80,"FlashMode":"DIO","Features":["0809","0F0005CE","00040001","00000001","00000000","00000880","80000080","00200000","54000008","00000080","00000004"],"Drivers":"1,2,4,9,10,12,29,38,50,52,62,81,!121","Sensors":"1,127","I2CDriver":""}}
14:37:33.522 MQT: stat/tasmota-cam/STATUS5 = {"StatusNET":{"Hostname":"Tasmota-CAM","IPAddress":"192.168.3.65","Gateway":"192.168.3.1","Subnetmask":"255.255.255.0","DNSServer1":"192.168.3.1","DNSServer2":"0.0.0.0","Mac":"40:22:D8:06:28:74","IP6Global":"","IP6Local":"fe80::4222:d8ff:fe06:2874%st1","Webserver":2,"HTTP_API":1,"WifiConfig":4,"WifiPower":7.0}}
14:37:33.558 MQT: stat/tasmota-cam/STATUS6 = {"StatusMQT":{"MqttHost":"192.168.3.39","MqttPort":1883,"MqttClientMask":"Tasmota-CAM","MqttClient":"Tasmota-CAM","MqttUser":"lucid","MqttCount":1,"MAX_PACKET_SIZE":1200,"KEEPALIVE":30,"SOCKET_TIMEOUT":4}}
14:37:33.585 MQT: stat/tasmota-cam/STATUS7 = {"StatusTIM":{"UTC":"2025-02-21T21:37:33Z","Local":"2025-02-21T14:37:33","StartDST":"2025-03-30T02:00:00","EndDST":"2025-10-26T03:00:00","Timezone":"-07:00","Sunrise":"06:51","Sunset":"18:13"}}
14:37:33.602 MQT: stat/tasmota-cam/STATUS10 = {"StatusSNS":{"Time":"2025-02-21T14:37:33","ESP32":{"Temperature":119.0},"TempUnit":"F"}}
14:37:33.622 MQT: stat/tasmota-cam/STATUS11 = {"StatusSTS":{"Time":"2025-02-21T14:37:33","Uptime":"0T00:01:44","UptimeSec":104,"Heap":117,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":1,"Berry":{"HeapUsed":3,"Objects":34},"POWER":"OFF","Dimmer":100,"Fade":"OFF","Speed":1,"LedTable":"ON","Wifi":{"AP":1,"SSId":"LucidWerks","BSSId":"E4:F4:C6:14:AE:49","Channel":3,"Mode":"HT40","RSSI":100,"Signal":-32,"LinkCount":1,"Downtime":"0T00:00:04"}}}

```
- [ ] Set `weblog` to 4 and then, when you experience your issue, provide the output of the Console log:
```lua
  Console output here:
There is nothing different that occurs in the weblog between a wakeup and a full reboot.

```

### TO REPRODUCE
_Steps to reproduce the behavior:_
I followed the Documentation exactly...
I set "Rule1 Wakeup"
Then I set "Rule1 5" which is supposed to wake up the device from deepsleep oon the next wakeup event and then STAY ON and NOT go back into DeepSleep mode unless specifically commanded to do so again.

### EXPECTED BEHAVIOUR
I expected this to behave like it is described in the Documentation...
When setting "Rule1 Wakeup" to mode ONCE by setting "Rule1 5" and then having a timer wake the device from DeepSleep the device is supposed to stay ON and NOT go back into DeepSleep immediately after waking up.

[image]

### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._
Here is my Timer setting that I used to wake the device from DeepSleep at a specific time... the device does wake up at the set time, but then immediately goes back into DeepSleep Countdown instead of staying awake like it is supposed to.

[image]

### ADDITIONAL CONTEXT
_Add any other context about the problem here._

**(Please, remember to close the issue when the problem has been addressed)**

LucidEye, 2025-03-20:
Hello?  So my help request is being totally ignored????  
This problem still exists.  
Are the devs not going to bother testing this and trying to fix it?
If not then the documentation needs to be fixed because the firmware is clearly NOT behaving as the documentation claims.

stefanbode, 2025-03-27:
Hi, sorry missed that post. Let me try to reproduce this tomorrow when I am back. I agree that this should work and was tested before. what teleperiod are you using 300? 10?. From the screenshot you provided you should also enable "REPEAT" so it wakes up every day. But this is not the root cause. Will come back soon

LucidEye, 2025-04-01:
Thank you for answering my post. What I really need to know is how do you make the wakeup rule stay on so that this cycle repeats every day automatically?  The documentation makes it sound like that's what's supposed to happen when you set Rule1 Wakeup and Rule1 5 and then set a repeating timer.... but that's not how it is working for me.
The whole purpose of my issue is that I need it to go to sleep at sunset, and wakeup at sunrise, every day, automatically. So what should the rule set for that look like?

stefanbode, 2025-04-01:
As in the documentation mentioned the following should do it: There is/WAS a mistake in the documentation that shows == instead of = before the sunset. I updated this just now. Let me know if it works.

Rule2
  on time#minute=%sunset% do rule1 5 endon

Rule2 ON

LucidEye, 2025-04-01:
So Rule 1 should be kept "Off" in order to function correctly????  That contradicts what the Rules Documentation says about pretty much every rule... Rules need to be activated (set to on) to function... but if I activate Rule1 (turn it on with "Rule1 on") and then enable the timer, the sleep countdown starts immeadiately.  That's VERY confusing.

LucidEye, 2025-04-01:
> As in the documentation mentioned the following should do it: There is/WAS a mistake in the documentation that shows == instead of = before the sunset. I updated this just now. Let me know if it works. The countdown should start immediatly. There should be no additional command required to send it to sleep.

I think you actually broke the documentation again... it is supposed to be ==... NOT =.

LucidEye, 2025-04-01:
This is definitely NOT working like it is supposed to.
I can get the device to go into deepsleep mode by using "restart 9", but it wakes up only a few seconds after going to sleep, wakes up before the timer I have set to wake it up, and I get some kind of error that looks like this...
11:30:58.090 CAM: InitErr 0x103 try 1
11:30:58.242 CAM: InitErr 0xffffffff try 2
11:30:58.389 CAM: InitErr 0xffffffff try 3
11:30:58.390 CAM: InitErr 0xffffffff

Again, this is running the Tasmota32 WebCam build, so maybe there is a bug in that particular build that isn't happening in the "regular" firmware you tested this with?

sfromis, 2025-04-01:
While using double `==` strictly speaking is indeed the canonical way of comparing numbers, it should very often also work ok using single `=` to compare as strings.

For numbers it can make a difference if the values are not normalized, like having extraneous signs, leading zeroes or decimal point. E.g. when calculations with default `CalcRes 3` will produce results like `4.000` even when only trying to work with integers.

stefanbode, 2025-04-02:
Good morning. Restart 9 is not deepsleep; therefore don't use it. The documentation also does not mention it. Just a wrong guess. Deepsleep is much more complicated than restart 9. 

On rule there are two options one in ON and one is ONCE. You want to keep your device ON after wake-up. 

So you have to use ONCE. If you just want to wake-up to measure and direct go to deepsleep you use ON. This is neither confusing and well documented. The 1800 to postpone this for 30min is fine. The normal deepsleep also wakes up the device one for a few seconds to make a measurement.

sfromis, 2025-04-02:
Strictly speaking, the docs for the [Restart](https://tasmota.github.io/docs/Commands/#restart) command does mention "go into deepsleep" with parameter 9, but this is indeed not meant to be used with the complexity of normal managed wakeup cycles.

stefanbode, 2025-04-02:
Reverted back the documentation to == and added a comment explaining that restart 9 does not take the deepsleep timing into consideration. But maybe we want to change this in the future. There might be some issues with backward combability if used with rules.

LucidEye, 2025-04-03:
I only seem to be running into the camera initilization errors when trying to wake the ESP up after putting it to sleep using "restart 9".  So there is definitely something strange with how that command works.  I have tried manually cycling the power dozens of times and also used "restart 1" dozens of times and I only get the "CAM: InitErr 0x103 try 1" errors after waking up from a "restart 9" forced sleep.

LucidEye, 2025-04-03:
So when I set Rule1 5... the "Once" mode changes to "On".  Is there a way to turn the "Once" mode "Off" from the console??  I can't seem to reset that mode to "Off" without having to go through the whole deepsleep and wakeup cycle.  I need to be able to disable the Once mode manually from the console for testing proposes to save time.

stefanbode, 2025-04-05:
Rule1 4 Set the once to off. Deepsleep should be the same on restart 9 and the deepsleep with timers. There is only one command. Anyhow deepsleep is something different than a restart 1. maybe you should check on you camera module if you need to reinitiate it after deepsleep

closed 2025-05-05 by github-actions[bot]