arendst/Tasmota #23050: Too much Zeros in 433MHz Data

bobybobybob, 2025-02-22:
### PROBLEM DESCRIPTION
With reference to https://github.com/arendst/Tasmota/issues/22919   a new problem has arisen with 14.5. There are now additional zeros in one of the RF433 MHz modules. 

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [ ] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [ ] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [ ] Searched the problem in [discussions](https://github.com/arendst/Tasmota/discussions)
- [ ] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [ ] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [ ] Problem is not scripter related, in this case open a discussion and tag @gemu2015
- [ ] Device used (e.g., Sonoff Basic): _____
- [ ] Tasmota binary firmware version number used: 14.5.0
  - [ ] Pre-compiled
  - [yes ] Self-compiled
- [ ] Flashing tools used: _____
- [ ] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```lua
  Configuration output here:

```
- [ ] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```lua
  Rules output here:

```
- [ ] Provide the output of this command: `Status 0`:
```lua
  STATUS 0 output here:

```
- [ ] Set `weblog` to 4 and then, when you experience your issue, provide the output of the Console log:
```lua
  Console output here:

```

### TO REPRODUCE
ESP32 C3  and a RXBB6 433MHz Module

### EXPECTED BEHAVIOUR

It sould have the same output

23:37:39.742 MQT: tele/tasmota_61FEE0/RESULT = {"Time":"2025-02-22T23:37:39","RfReceived":{"Data":"0x0000000000755C03","Bits":24,"Protocol":1,"Pulse":303}}

23:38:40.771 MQT: tele/tasmota_rf433_C37FA9/RESULT = {"Time":"2025-02-22T23:38:40","RfReceived":{"Sync":9330,"Low":260,"High":920,"Data":"0x755C03","RfKey":"None"}}

### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._

[image]

### ADDITIONAL CONTEXT
_Add any other context about the problem here._

**(Please, remember to close the issue when the problem has been addressed)**

helmar74, 2025-02-23:
Same issue here on a Sonoff 4CHR3.
I changed my rule from:
```
rule1
ON RfReceived#Data=0x57E846 DO power1 on ENDON
ON RfReceived#Data=0xE70244 DO power1 on ENDON
ON RfReceived#Data=0x7FAB14 DO power1 on ENDON
ON RfReceived#Data=0xC4114 DO power1 on ENDON

rule1 1

rule2 
ON RfReceived#Data=0x57E84A DO power2 on ENDON
ON RfReceived#Data=0xE70248 DO power2 on ENDON
ON RfReceived#Data=0x7FAB18 DO power2 on ENDON
ON RfReceived#Data=0xC4118 DO power2 on ENDON

rule2 1
```

to:

```
rule1
ON RfReceived#Data=0x000000000057E846 DO power1 on ENDON
ON RfReceived#Data=0x0000000000E70244 DO power1 on ENDON
ON RfReceived#Data=0x00000000007FAB14 DO power1 on ENDON
ON RfReceived#Data=0x00000000000C4114 DO power1 on ENDON

rule1 1

rule2 
ON RfReceived#Data=0x000000000057E84A DO power2 on ENDON
ON RfReceived#Data=0x0000000000E70248 DO power2 on ENDON
ON RfReceived#Data=0x00000000007FAB18 DO power2 on ENDON
ON RfReceived#Data=0x00000000000C4118 DO power2 on ENDON

rule2 1
```

Now it's working again

helmar74, 2025-02-23:
btw. did you set:
```
#define FIX_JSON_HEXADECIMAL
```
as mentioned in 
[https://github.com/arendst/Tasmota/issues/22919#issuecomment-2629428963](https://github.com/arendst/Tasmota/issues/22919#issuecomment-2629428963)

?

bobybobybob, 2025-02-23:
> btw. did you set:
> 
> ```
> #define FIX_JSON_HEXADECIMAL
> ```
> 
> as mentioned in [#22919 (comment)](https://github.com/arendst/Tasmota/issues/22919#issuecomment-2629428963)
> 
> ?

Yes, this is corrected when compiling the other receiver (completely different design) and supplements the 0x that was previously missing there.  But that with the zeros is completely new.

helmar74, 2025-02-23:
Then you maybe can change the rules on the devices or in your connected system?

bobybobybob, 2025-02-23:
The IOBroker evaluates the MQTT signals from the two devices. For the one with too many zeros, I will go back one version for the time being so that it is consistent again. Rules are not used.

bobybobybob, 2025-02-23:
Thank you for fast fix

closed 2025-02-23 by bobybobybob

helmar74, 2025-02-23:
> Thank you for fast fix

I also can confirm the fix:
Version 14.5.0
```
15:25:53.119 MQT: tele/sonoff4CH_1/RESULT = {"Time":"2025-02-23T15:25:53","RfReceived":{"Data":"0x00000000000C4111","Bits":24,"Protocol":1,"Pulse":294}}
```

Version 14.5.1:
```
15:33:39.209 MQT: tele/sonoff4CH_1/RESULT = {"Time":"2025-02-23T15:33:39","RfReceived":{"Data":"0xC4111","Bits":24,"Protocol":1,"Pulse":294}}
```