arendst/Tasmota #23072: Berry Leds matrix code wrong

grmpl, 2025-02-25:
### PROBLEM DESCRIPTION
The berry-commands for adressing a matrix with adressable Leds is not consistent. I'm not sure if I'm missing a fact, but for me it is totally wrong and the commands don't work for my matrix (32x8 Ulanzi TC001):
see lib/libesp32/berry_tasmota/src/embedded/leds.be
- `create_matrix(w, h, offset)` is defined with "width, height" 
- but `set_matrix_pixel_color(x, y, col, bri)` will set pixel at (x*self.w + y), which does not work:
  - if x is horizontal and y vertical (as assumed by create_matrix) and led-strip is laid out horizontally, it should set pixel at (y*self.w + x)
  - if x is vertical and y horizontal and led-strip is laid out horizontally, it should set pixel at (x*self.h+y)
  - if x is horizontal and y vertical and led-strip is laid out vertically, it should set pixel at (x*self.h+y)
  - if x is vertical and y horizontal and led-strip is laid out vertically, it should set pixel at (y*self.w+x)
  - I don't find a combination of strip layout and x/y-combination where (x*self.w+y) would fit
- furthermore `set_alternate` will check for `y & 0x1` (every second row or every second column), then it would reverse also the y-values: `(x * self.w + self.h - y - 1)`. Correct would be to reverse the x-values, which are the leds in the y-line.
- and last, the `set_bytes(row, buf, offset, len)` does not work out: It says, one can set **one row**, but it checks for **height**-length instead of width-length: ` if (len > h_bytes) `. Seems like somebody confused "height" with "horizontal"

Maybe there is a LED-strip layout which would work with this code, but I can't think of one. The Leds in the Ulanzi TC001 are laid out in horizontal lines with zigzag-layout and the code definitely isn't working there. 

I would have written a pull request, but I'm not sure if my horizontal zigzag-layout is supposed to be the correct one.

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [x] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [x] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [x] Searched the problem in [discussions](https://github.com/arendst/Tasmota/discussions)
- [x] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [x] Problem is not scripter related, in this case open a discussion and tag  gemu2015
- [x] Device used (e.g., Sonoff Basic): Ulanzi TC001
- [x] Tasmota binary firmware version number used:14.4.1
  - [x] Pre-compiled
  - [x] Self-compiled
- [x] Flashing tools used: TasmoCompiler
- [x] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```lua
  Configuration output here:
21:18:52.190 MQT: stat/Ulanzi1/RESULT = {"NAME":"Ulanzi TC001","GPIO":[0,0,0,0,0,0,0,0,0,0,34,480,0,0,0,0,0,640,608,0,0,0,32,33,0,0,0,0,1377,0,4705,4768,0,0,0,0],"FLAG":0,"BASE":1}
21:18:52.479 MQT: stat/Ulanzi1/RESULT = {"Module":{"0":"Ulanzi TC001"}}
21:18:52.839 MQT: stat/Ulanzi1/RESULT = {"GPIO0":{"0":"None"},"GPIO1":{"0":"None"},"GPIO2":{"0":"None"},"GPIO3":{"0":"None"},"GPIO4":{"0":"None"},"GPIO5":{"0":"None"},"GPIO6":{"0":"None"},"GPIO7":{"0":"None"},"GPIO8":{"0":"None"},"GPIO9":{"0":"None"},"GPIO10":{"0":"None"},"GPIO11":{"0":"None"},"GPIO12":{"0":"None"},"GPIO13":{"0":"None"},"GPIO14":{"34":"Button3"},"GPIO15":{"480":"Summer"},"GPIO16":{"0":"None"},"GPIO17":{"0":"None"},"GPIO18":{"0":"None"},"GPIO19":{"0":"None"},"GPIO20":{"0":"None"},"GPIO21":{"640":"I2C SDA1"},"GPIO22":{"608":"I2C SCL1"},"GPIO23":{"0":"None"},"GPIO24":{"0":"None"},"GPIO25":{"0":"None"},"GPIO26":{"32":"Button1"},"GPIO27":{"33":"Button2"},"GPIO32":{"1377":"WS28122"},"GPIO33":{"0":"None"},"GPIO34":{"4705":"ADC Eingang2"},"GPIO35":{"4768":"ADC Licht1"},"GPIO36":{"0":"None"},"GPIO37":{"0":"None"},"GPIO38":{"0":"None"},"GPIO39":{"0":"None"}}

```
- [ ] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```lua
  Rules output here:
no rules used

```
- [x] Provide the output of this command: `Status 0`:
```lua
  STATUS 0 output here:
21:20:05.238 MQT: stat/Ulanzi1/STATUS0 = {"Status":{"Module":0,"DeviceName":"Ulanzi1","FriendlyName":["Ulanzi1"],"Topic":"Ulanzi1","ButtonTopic":"0","Power":"1","PowerLock":"0","PowerOnState":3,"LedState":1,"LedMask":"FFFF","SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0,"InfoRetain":0,"StateRetain":0,"StatusRetain":0},"StatusPRM":{"Baudrate":115200,"SerialConfig":"8N1","GroupTopic":"tasmotas","OtaUrl":"http://ota.tasmota.com/tasmota32/release/tasmota32.bin","RestartReason":"Software reset CPU","Uptime":"0T01:03:53","StartupUTC":"2025-02-25T19:16:12","Sleep":100,"CfgHolder":4617,"BootCount":184,"BCResetTime":"2025-01-04T20:23:46","SaveCount":694},"StatusFWR":{"Version":"14.4.1(TasmoCompiler-esp32generic)","BuildDateTime":"2025.01.15 17:25:33","Core":"3_1_0","SDK":"5.3.2","CpuFrequency":160,"Hardware":"ESP32-D0WD v1.0","CR":"649/699"},"StatusLOG":{"SerialLog":3,"WebLog":2,"MqttLog":2,"SysLog":3,"LogHost":"192.168.0.22","LogPort":514,"SSId":["Meiner2",""],"TelePeriod":30,"Resolution":"558180C0","SetOption":["0000A00B","0F05C80001000600003C5A0A192800000000","00800081","00006000","00004000","00000000"]},"StatusMEM":{"ProgramSize":1359,"Free":1520,"Heap":149,"StackLowMark":3,"PsrMax":0,"PsrFree":0,"ProgramFlashSize":16384,"FlashSize":16384,"FlashChipId":"184020","FlashFrequency":40,"FlashMode":"DIO","Features":["0407","0F00C5CE","00040001","680A3FC3","182C0400","D03A0881","0009B8C8","00200000","54000048","00000080","00000004"],"Drivers":"1,2,!4,9,10,12,24,26,29,!35,38,40,50,52,56,62,!68,!121","Sensors":"1,2,5,6,7,8,9,10,11,14,16,24,26,27,38,39,41,46,47,55,57,58,59,63,65,67,70,71,72,73,74,88,127","I2CDriver":"8,9,10,11,12,15,16,19,20,21,26,28,32,36,40,41,42,43,45,48,49,50,51,52,60"},"StatusNET":{"Hostname":"Ulanzi1-2656","IPAddress":"192.168.0.136","Gateway":"192.168.0.66","Subnetmask":"255.255.255.0","DNSServer1":"192.168.0.66","DNSServer2":"2003:c7:8f41:cb00:5e49:79ff:fe66:6711","Mac":"EC:62:60:51:8A:60","IP6Global":"2003:c7:8f41:cb00:ee62:60ff:fe51:8a60","IP6Local":"fe80::ee62:60ff:fe51:8a60%st1","Webserver":2,"HTTP_API":1,"WifiConfig":2,"WifiPower":16.0},"StatusMQT":{"MqttHost":"192.168.0.22","MqttPort":1883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_518A60","MqttUser":"DVES_USER","MqttCount":1,"MAX_PACKET_SIZE":1200,"KEEPALIVE":30,"SOCKET_TIMEOUT":4},"StatusTIM":{"UTC":"2025-02-25T20:20:05Z","Local":"2025-02-25T21:20:05","StartDST":"2025-03-30T02:00:00","EndDST":"2025-10-26T03:00:00","Timezone":99,"Sunrise":"07:03","Sunset":"17:52"},"StatusSNS":{"Time":"2025-02-25T21:20:05","ANALOG":{"A2":2819,"Illuminance1":28},"SHT3X":{"Temperature":31.2,"Humidity":29.1,"DewPoint":11.1},"TempUnit":"C"},"StatusSTS":{"Time":"2025-02-25T21:20:05","Uptime":"0T01:03:53","UptimeSec":3833,"Heap":148,"SleepMode":"Dynamic","Sleep":100,"LoadAvg":25,"MqttCount":1,"Berry":{"HeapUsed":3,"Objects":36},"Wifi":{"AP":1,"SSId":"Meiner2","BSSId":"E8:DF:70:B2:80:4F","Channel":11,"Mode":"HT40","RSSI":100,"Signal":-29,"LinkCount":1,"Downtime":"0T00:00:03"}}}

```
- [ ] Set `weblog` to 4 and then, when you experience your issue, provide the output of the Console log:
```lua
  Console output here:
no output there

```

### TO REPRODUCE
_Steps to reproduce the behavior:_
In Berry:
leds = Leds(
            32 * 8,
            gpio.pin(gpio.WS2812, 1), # Look up the correct GPIO pin for WS2812 with ID 2 (1 in Berry)
            Leds.WS2812_GRB,
            3 
        )
m=leds.create_matrix(32,8)
m.set_matrix_pixel_color(1,0,0x00ff00)
m.show()
m.set_alternate(true)
m.set_matrix_pixel_color(1,0,0x00ff00)
m.show()
m.set_bytes(0,bytes("00ff0000ff0000ff0000ff0000ff0000ff0000ff0000ff0000ff0000ff0000ff0000ff00"),0,12)
m.show()

### EXPECTED BEHAVIOUR
_A clear and concise description of what you expected to happen._
- a green pixel in second row at the left
- a green pixel in second row at the right
- 12 red pixels in first row

### Result
- only a green pixel in second row at the left
- no green pixe in second row at the right
- only 8 red leds in first row

**(Please, remember to close the issue when the problem has been addressed)**

s-hadinger, 2025-02-27:
Hmmm. This looks wrong indeed. This is very old code, and apparently not much used since.

@Staars you worked on the Ulanzi code, is this something you spotted as well? (inversion of x and y for `set_matrix_pixel_color`)

Staars, 2025-02-28:
`set_matrix_pixel_color` is indeed wrong, but I do not use it in my Ulanzi code anyway.

Apparently I stumbled upon it a few months ago, because I have corrected it in the Tasmota online emulator, where it works as it should with this implementation:
```
      def set_matrix_pixel_color(x, y, col, bri)
        # if y is odd, reverse the order
        if y & 1 == 1
          x = self.w - x - 1
        end

        self.strip.set_pixel_color(y * self.w + x, col, bri)     
      end
```
But then I forgot about it because there was no response to this topic and I didn't want to waste my developer time on things, that no one is interested in.

I don't remember ever trying `set_bytes(row, buf, offset, len)`.

@grmpl 
```
m.set_alternate(true)
m.set_matrix_pixel_color(1,0,0x00ff00)
m.show()
```
Why do you expect a green pixel in the second row at the right?

grmpl, 2025-02-28:
The green pixel on the right is expected from first set_matrix_pixel_color, where set_alternate is still false.

Sorry, for mentioning the issue. Nobody seems to use the code. I don't need the code, too. It's just a nuisance for beginners.

Staars, 2025-02-28:
The first line and all even lines are not affected by set_alternate, only the odd lines are reversed. Thus painting into the first line w/o set_alternate should lead to the same result.

No problem with the issue here, I was referring to potential example code.

grmpl, 2025-02-28:
It depends, what you expect from x. :-) 
I took x being height in the code as a bad design decision, not as an error, so I expect 1,0 to be second line, first column.

As nobody seems to care, I would suggest correcting the "bad design decision", too.
Should I create a pull request?

grmpl, 2025-03-02:
Sorry, I tried to do a correction and found  two additional issues, which I fear are not the last ones:
1) Matrices on sale have multiple layouts, starting LED and layout of strip can be individually defined in Neopixel library. Should code be fixed to one layout?
2) Scroll function has 3 bytes/pixel hardcoded, this should be pixel_size 

Maybe it would be better to eliminate matrix code instead of correcting it?

grmpl, 2025-03-04:
I had a deeper look into the code. The code about matrix-leds has some nice ideas, like the scrolling, but it does not work.
A simple rework is not possible, because parameters are used inconsistently:
- As already shown h, w, x, y are used inconsistently
- Offset is used in init, and set_matrix_pixel_color, but is it not used in scroll and wrongly used in set_bytes
- set_alternate is only used in set_matrix_pixel_color, but not in scroll and set_bytes
- pix_size is used in set_matrix_pixel_color, but not in scroll (not necessary in set_bytes)

To get it working, you would have to add a lot of additional code which would handle these options correctly, blowing up the code, increasing image sizes and slowing down execution. And it will still be limited to some display types, as it lacks options for full flexibility. If you want to give it the full flexibility this would be very much more code to write, most probably with very bad performance if written in berry only. 

As nobody seems to be very interested in the matrix-functions and as it is not very complicated to write your own functions, I don't think this would be worth the effort. 

So my strong suggestion would be: Throw out completely the matrix code from LED module and adapt the documentation accordingly. 

If anybody wants to keep the code, one has to make a decision, which options/layout to support and which to throw out. For keeping a tolerable performance and code size, I would suggest supporting only the Ulanzi layout:
- Layout is horizontally aligned strips with starting led top left
- x is horizontally, y is vertically, starting on top left
- Overall Offset: Throw out and set to 0 - Who would need an offset in a matrix?
- pix_size: Only allow pix_size=3, scroll will be getting complicated with variable pix_size
- set_alternate: Always set it to true - is there a display with not zigzag-layout?

I could do the code corrections, if these decisions are made by somebody and if there is anybody showing interest in the code. 

Just to make it clear: As my own code works already without this matrix functions, I have no interest to correct it other than avoiding others to get frustrated by documented, but buggy functions.

s-hadinger, 2025-03-06:
Thanks, I agree. There are so many possibilities that it's hard to abstract it. I think I will remove the matrix code for now.

@Staars are you ok with removing Led_matrix from the solidified code?

s-hadinger, 2025-03-06:
> * set_alternate: Always set it to true - is there a display with not zigzag-layout?

For this question, yes the M5Stack Atom Matrix is not zigzag-layout

s-hadinger, 2025-03-06:
The PR is ready, waiting for @Staars approval to remove matrix code

Jason2866, 2025-03-08:
Closing since the referred code is removed

closed 2025-03-08 by Jason2866