arendst/Tasmota #23081: Energy Counter Total Resolution and Behavior at Begin of Day

ptlrk, 2025-02-28:
### PROBLEM DESCRIPTION
_A clear and concise description of what the problem is._
#### Step down of the Total Energy Counter at begin of a new day
When EnergyRes is set to "5" the total enery counter discards the last two of its 5 digits after the decimal by begin of a new day. Effectively, this reduces the counter value at the begin of a new day, which leads to a (small) loss of counted energy, maximum of 1 Wh (average 0.5 Wh) per day. More important is, that other systems, as the homematic energy counter, are negatively affected, when using the total energy counter, e.g. by means of an MQTT-broker.

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [x] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [x] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [x] Searched the problem in [discussions](https://github.com/arendst/Tasmota/discussions)
- [x] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [x] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [x] Problem is not scripter related, in this case open a discussion and tag  gemu2015
- [x] Device used (e.g., Sonoff Basic): Nous A1T
- [x] Tasmota binary firmware version number used: 14.5.0 (release-tasmota)
  - [x] Pre-compiled
  - [ ] Self-compiled
- [x] Flashing tools used: WebGUI
- [x] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```lua
  Configuration output here:
11:50:04.179 MQT: tasmota/tasmota_F0B081/stat/RESULT = {"NAME":" NOUS A1T","GPIO":[32,0,0,0,2720,2656,0,0,2624,320,224,0,0,0],"FLAG":0,"BASE":49}
11:50:04.387 MQT: tasmota/tasmota_F0B081/stat/RESULT = {"Module":{"0":" NOUS A1T"}}
11:50:04.643 MQT: tasmota/tasmota_F0B081/stat/RESULT = {"GPIO0":{"32":"Button1"},"GPIO1":{"0":"None"},"GPIO2":{"0":"None"},"GPIO3":{"0":"None"},"GPIO4":{"2720":"BL0937 CF"},"GPIO5":{"2656":"HLWBL CF1"},"GPIO9":{"0":"None"},"GPIO10":{"0":"None"},"GPIO12":{"2624":"HLWBL SEL_i"},"GPIO13":{"320":"Led_i1"},"GPIO14":{"224":"Relay1"},"GPIO15":{"0":"None"},"GPIO16":{"0":"None"},"GPIO17":{"0":"None"}}

```
- [ ] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```lua
  Rules output here:

```
- [x] Provide the output of this command: `Status 0`:
```lua
  STATUS 0 output here:
11:51:50.017 MQT: tasmota/tasmota_F0B081/stat/STATUS = {"Status":{"Module":0,"DeviceName":"EG Kü Ladegeräte","FriendlyName":["Tasmota"],"Topic":"tasmota_F0B081","ButtonTopic":"0","Power":"1","PowerLock":"0","PowerOnState":3,"LedState":1,"LedMask":"FFFF","SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":1,"SensorRetain":1,"PowerRetain":1,"InfoRetain":1,"StateRetain":1,"StatusRetain":1}} (retained)
11:51:50.023 MQT: tasmota/tasmota_F0B081/stat/STATUS1 = {"StatusPRM":{"Baudrate":115200,"SerialConfig":"8N1","GroupTopic":"tasmotas","OtaUrl":"http://ota.tasmota.com/tasmota/release/tasmota.bin.gz","RestartReason":"Software/System restart","Uptime":"0T00:47:35","StartupUTC":"2025-02-28T10:04:15","Sleep":50,"CfgHolder":4617,"BootCount":120,"BCResetTime":"2023-11-27T12:34:55","SaveCount":2474,"SaveAddress":"FA000"}} (retained)
11:51:50.028 MQT: tasmota/tasmota_F0B081/stat/STATUS2 = {"StatusFWR":{"Version":"14.5.0(release-tasmota)","BuildDateTime":"2025-02-18T14:34:10","Boot":31,"Core":"2_7_8","SDK":"2.2.2-dev(38a443e)","CpuFrequency":80,"Hardware":"ESP8266EX","CR":"372/699"}} (retained)
11:51:50.033 MQT: tasmota/tasmota_F0B081/stat/STATUS3 = {"StatusLOG":{"SerialLog":2,"WebLog":2,"MqttLog":0,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["Luerkens557134",""],"TelePeriod":30,"Resolution":"569580C0","SetOption":["000082A9","2805C80001000600003C5A0A192800000000","00000081","00006000","80004180","00000000"]}} (retained)
11:51:50.046 MQT: tasmota/tasmota_F0B081/stat/STATUS4 = {"StatusMEM":{"ProgramSize":652,"Free":348,"Heap":21,"ProgramFlashSize":1024,"FlashSize":1024,"FlashChipId":"146085","FlashFrequency":40,"FlashMode":"DOUT","Features":["0809","8F9AC787","04368001","000000CF","010013C0","C000F981","00004004","00001000","54000020","00000080","00000000"],"Drivers":"1,2,3,!4,!5,!6,7,!8,9,10,12,!16,!18,!19,!20,!21,!22,!24,26,!27,29,!30,!35,!37,!45,62,!68","Sensors":"1,2,3,4,5,6","I2CDriver":"7"}} (retained)
11:51:50.054 MQT: tasmota/tasmota_F0B081/stat/STATUS5 = {"StatusNET":{"Hostname":"tasmota-F0B081-4225","IPAddress":"192.168.2.240","Gateway":"192.168.2.1","Subnetmask":"255.255.255.0","DNSServer1":"192.168.2.1","DNSServer2":"0.0.0.0","Mac":"08:3A:8D:F0:B0:81","Webserver":2,"HTTP_API":1,"WifiConfig":4,"WifiPower":17.0}} (retained)
11:51:50.059 MQT: tasmota/tasmota_F0B081/stat/STATUS6 = {"StatusMQT":{"MqttHost":"homematic","MqttPort":1883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_F0B081","MqttUser":"DVES_USER","MqttCount":1,"MqttTLS":0,"MAX_PACKET_SIZE":1200,"KEEPALIVE":30,"SOCKET_TIMEOUT":4}} (retained)
11:51:50.065 MQT: tasmota/tasmota_F0B081/stat/STATUS7 = {"StatusTIM":{"UTC":"2025-02-28T10:51:50Z","Local":"2025-02-28T11:51:50","StartDST":"2025-03-30T01:00:00","EndDST":"2025-10-26T01:00:00","Timezone":99,"Sunrise":"07:20","Sunset":"18:14"}} (retained)
11:51:50.069 MQT: tasmota/tasmota_F0B081/stat/STATUS9 = {"StatusPTH":{"PowerDelta":0,"PowerLow":0,"PowerHigh":0,"VoltageLow":0,"VoltageHigh":0,"CurrentLow":0,"CurrentHigh":0,"MaxPower":0,"MaxPowerHold":10,"MaxPowerWindow":30,"MaxEnergy":0,"MaxEnergyStart":0}} (retained)
11:51:50.077 MQT: tasmota/tasmota_F0B081/stat/STATUS10 = {"StatusSNS":{"Time":"2025-02-28T11:51:50","ENERGY":{"TotalStartTime":"2025-02-28T09:50:16","Total":17.45803,"Yesterday":0.01413,"Today":0.00103,"Power":0.00,"ApparentPower":0.00,"ReactivePower":0.00,"Factor":0.00,"Voltage":230.10,"Current":0.000}}} (retained)
11:51:50.083 MQT: tasmota/tasmota_F0B081/stat/STATUS11 = {"StatusSTS":{"Time":"2025-02-28T11:51:50","Uptime":"0T00:47:35","UptimeSec":2855,"Heap":21,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":1,"POWER":"ON","Wifi":{"AP":1,"SSId":"Luerkens557134","BSSId":"C8:0E:14:DF:0E:8B","Channel":6,"Mode":"11n","RSSI":60,"Signal":-70,"LinkCount":1,"Downtime":"0T00:00:04"}}} (retained)

```
- [ ] Set `weblog` to 4 and then, when you experience your issue, provide the output of the Console log:
```lua
  Console output here:

```

### TO REPRODUCE
_Steps to reproduce the behavior:_
- Goto Console- Set EnergyRes 5
  - Reset Energy Counters to 0
  - Command "EnergyToday 0"
  - Command "EnergyTotal 0"
- Turn on some load until the energy counter has build up a non-zero-value eg (0.00156) in its last two digits.
- Switch-off load.
- Total Energy shows now a 5-digit after decimal value (e.g. 0.001**56**)
- Wait until midnight.
- After midnight the two last digits are reset to zero (e.g. 0.001**00**)

A faster way to reproduce:
- Goto Console
  - Set EnergyRes 5
  - Reset Energy Counters to 0
[... 2 lines cut ...]
  - type command "Time 1740783240" (this puts the system time 5 minutes before midnight on February 28, 2025)
  - type Energytoday 1 ( Wh fractions are not possible, here)
  - Turn on some load until the energy counter has increased to some non-zero-value eg (0.00156) in its last two digits (in less than 5 Minutes)
  - Switch-off load.
  - Total Energy shows now a 5-digit after decimal value (e.g. 0.001**56**)
  - Type command "Time 1740783240" (advances time of day to 23:59:30)
- Wait 30 seconds
- Energy-Today switches to 0.00000, Energy-Total to 0.001**00**
  

### EXPECTED BEHAVIOUR
_A clear and concise description of what you expected to happen._
At the begin of the new day the total energy counter should maintain the same state as just before, considerung the defined precision.

### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._

### ADDITIONAL CONTEXT
_Add any other context about the problem here._
It appears the the energy-total value is produced from an internal stored value, which is updated once at the begin of the day by the sum of it's current value and the energy-today value. The energy total value appears an integer Wh-quantity, while the energy-today has a higher resolution (at least 0.01 Wh). Thus, during update, fractional parts of the today-counter get lost. 
[... cut ...] ternal storage total variable by the amount of the current total energy counter value. It would probably not happen, if at midnight the today-counter value indentally has a value with the last two digits being 0, or if the precision had been set to 3.
Btw, while the energy fields today and total show fractional quantities of a Wh it's not possible with the commands EnergyToday or EnergyTotal to set a fractional value.

**(Please, remember to close the issue when the problem has been addressed)**

ptlrk, 2025-03-02:
I hope my contribution was appropriate, I'm not yet familiar very well with the habits on github. If not, please tell me, I'm preapred to learn.
Yes, the behavior is as designed, obviously. The downside is the (little) step down of the total counter in the moment when the today-counter es reset. I have added two screenshots (one from the data logger (Wh), the other one from MQTT-explorer (kWh)), to make ist a bit clearer, at least I hope so.

[image]
[image]

Jason2866, 2025-03-08:
Closing, since behaviour is as designed and not planned to change.

ptlrk, 2025-03-08:
Indeed, as designed but not "desirable". Anyway, I'll look for a workaround on my backend.
Thanks for looking into it.

Jason2866, 2025-03-08:
Closed since not an issue. You can open in discussions an idea (feature request).
Since the project owner gave the answer, i doubt it will be changed/considered