arendst/Tasmota #23085: Yeelight Crystal Pendant Light (YLDL01YL) not working as expected

DcFamas, 2025-03-01:
### PROBLEM DESCRIPTION
Hello guys,
I own a Yeelight Crytal Pendant Light since a few years. Years ago I flashed v.10 to the device, everything works like a charme. Now I wanted to use matter, so I migrated to the latest solo1.bin 14.4.1.4. Since then the device is not working as expected. I think there is a problem with the template:

{"NAME":"Yeelight Meteorite","GPIO":[0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,420,0,419,0,0,0,0,417,418,0,0,0,0,0,416,0,0,0,0,0,0],"FLAG":0,"BASE":1,"CMND":"SO37 128"}

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [X] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [X] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [X] Searched the problem in [discussions](https://github.com/arendst/Tasmota/discussions)
- [X] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [X] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [X] Problem is not scripter related, in this case open a discussion and tag  gemu2015
- [X] Device used (e.g., Sonoff Basic): YLDL01YL
- [X] Tasmota binary firmware version number used: tasmota32solo1.bin v. 14.4.1.4
  - [X] Pre-compiled
  - [ ] Self-compiled
- [X] Flashing tools used: OTA
- [X] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```lua
  Configuration output here:
10:20:38.635 CMD: Backlog Template; Module; GPIO 255
10:20:38.661 MQT: stat/tasmota_0266AB/RESULT = {"NAME":"Yeelight Meteorite","GPIO":[0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,420,0,419,224,0,0,0,417,418,0,0,0,0,0,416,0,0,0,0,0,0],"FLAG":0,"BASE":1}
10:20:38.901 MQT: stat/tasmota_0266AB/RESULT = {"Module":{"0":"Yeelight Meteorite"}}
10:20:39.153 MQT: stat/tasmota_0266AB/RESULT = {"GPIO0":{"0":"None"},"GPIO1":{"0":"None"},"GPIO2":{"0":"None"},"GPIO3":{"0":"None"},"GPIO4":{"0":"None"},"GPIO5":{"0":"None"},"GPIO6":{"0":"None"},"GPIO7":{"0":"None"},"GPIO8":{"0":"None"},"GPIO9":{"0":"None"},"GPIO10":{"0":"None"},"GPIO11":{"0":"None"},"GPIO12":{"0":"None"},"GPIO13":{"0":"None"},"GPIO14":{"0":"None"},"GPIO15":{"0":"None"},"GPIO16":{"0":"None"},"GPIO17":{"0":"None"},"GPIO18":{"0":"None"},"GPIO19":{"420":"PWM5"},"GPIO20":{"0":"None"},"GPIO21":{"419":"PWM4"},"GPIO22":{"224":"Relay1"},"GPIO23":{"0":"None"},"GPIO24":{"0":"None"},"GPIO25":{"0":"None"},"GPIO26":{"417":"PWM2"},"GPIO27":{"418":"PWM3"},"GPIO32":{"0":"None"},"GPIO33":{"416":"PWM1"},"GPIO34":{"0":"None"},"GPIO35":{"0":"None"},"GPIO36":{"0":"None"},"GPIO37":{"0":"None"},"GPIO38":{"0":"None"},"GPIO39":{"0":"None"}}

```
- [ ] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```lua
  Rules output here:

```
- [ ] Provide the output of this command: `Status 0`:
```lua
  STATUS 0 output here:
10:25:26.559 CMD: Status 0
10:25:26.567 MQT: stat/tasmota_0266AB/STATUS = {"Status":{"Module":0,"DeviceName":"Tasmota-Arbeitszimmer-Meteroite","FriendlyName":["Tasmota-Arbeitszimmer-Meteroite","",""],"Topic":"tasmota_0266AB","ButtonTopic":"0","Power":"011","PowerLock":"000","PowerOnState":1,"LedState":1,"LedMask":"FFFF","SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0,"InfoRetain":0,"StateRetain":0,"StatusRetain":0}}
10:25:26.636 MQT: stat/tasmota_0266AB/STATUS1 = {"StatusPRM":{"Baudrate":115200,"SerialConfig":"8N1","GroupTopic":"tasmotas","OtaUrl":"http://ota.tasmota.com/tasmota32/release/tasmota32.bin","RestartReason":"Software reset CPU","Uptime":"0T00:02:25","StartupUTC":"2025-03-01T09:23:01","Sleep":50,"CfgHolder":4617,"BootCount":395,"BCResetTime":"2021-10-31T10:37:37","SaveCount":17969}}
10:25:26.682 MQT: stat/tasmota_0266AB/STATUS2 = {"StatusFWR":{"Version":"14.4.1.4(28b22de-solo1)single-core","BuildDateTime":"2025-02-04T19:16:12","Core":"3_1_1","SDK":"5.3.2.250120","CpuFrequency":240,"Hardware":"ESP32-S0WD-OEM v1.0","CR":"483/699"}}
10:25:26.708 MQT: stat/tasmota_0266AB/STATUS3 = {"StatusLOG":{"SerialLog":2,"WebLog":2,"MqttLog":0,"FileLog":0,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["WLAN",""],"TelePeriod":300,"Resolution":"558180C0","SetOption":["00008009","2805C80001800600003C5A0A002800000000","00000080","00006400","00004000","00000000"]}}
10:25:26.761 MQT: stat/tasmota_0266AB/STATUS4 = {"StatusMEM":{"ProgramSize":1998,"Free":881,"Heap":162,"StackLowMark":3,"PsrMax":0,"PsrFree":0,"ProgramFlashSize":4096,"FlashSize":4096,"FlashChipId":"16701C","FlashFrequency":40,"FlashMode":"DIO","Features":["0809","9F9AD7DF","0015A001","B7F7BFCF","05DA9BC4","E0360DC7","480840D2","20200000","D4BC482D","810A80F1","00000814"],"Drivers":"1,2,!3,4,!5,7,!8,9,10,11,12,!14,!16,!17,!20,!21,!24,26,!27,29,!34,!35,38,50,52,!59,!60,62,!63,!66,!67,!68,!73,!75,82,!86,!87,!88,!91,!121","Sensors":"1,2,3,5,6,7,8,9,10,11,12,13,14,15,17,18,19,20,21,22,26,31,34,37,39,40,42,43,45,51,52,55,56,58,59,64,66,67,74,85,92,95,98,103,105,109,127","I2CDriver":"7,8,9,10,11,12,13,14,15,17,18,20,24,29,31,36,41,42,44,46,48,58,62,65,69,76,77,82,89"}}
10:25:26.849 MQT: stat/tasmota_0266AB/STATUS5 = {"StatusNET":{"Hostname":"tasmota-0266AB-1707","IPAddress":"172.16.0.57","Gateway":"172.16.0.1","Subnetmask":"255.240.0.0","DNSServer1":"172.16.0.1","DNSServer2":"2002:6470:fe3a:0:2e3a:fdff:febc:b51c","Mac":"58:B6:23:02:66:AB","IP6Global":"2002:6470:fe3a:0:5ab6:23ff:fe02:66ab","IP6Local":"fe80::5ab6:23ff:fe02:66ab%st1","Ethernet":{"Hostname":"","IPAddress":"0.0.0.0","Gateway":"0.0.0.0","Subnetmask":"0.0.0.0","DNSServer1":"172.16.0.1","DNSServer2":"2002:6470:fe3a:0:2e3a:fdff:febc:b51c","Mac":"00:00:00:00:00:00","IP6Global":"","IP6Local":""},"Webserver":2,"HTTP_API":1,"WifiConfig":4,"WifiPower":17.0}}
10:25:26.921 MQT: stat/tasmota_0266AB/STATUS6 = {"StatusMQT":{"MqttHost":"172.16.0.3","MqttPort":1883,"MqttClientMask":"Tasmota-Arbeitszimmer-Meteroite","MqttClient":"Tasmota-Arbeitszimmer-Meteroite","MqttUser":"mqtt","MqttCount":1,"MAX_PACKET_SIZE":1200,"KEEPALIVE":30,"SOCKET_TIMEOUT":4}}
10:25:26.960 MQT: stat/tasmota_0266AB/STATUS7 = {"StatusTIM":{"UTC":"2025-03-01T09:25:26Z","Local":"2025-03-01T10:25:26","StartDST":"2025-03-30T02:00:00","EndDST":"2025-10-26T03:00:00","Timezone":"+01:00","Sunrise":"07:31","Sunset":"18:33"}}
10:25:26.985 MQT: stat/tasmota_0266AB/STATUS10 = {"StatusSNS":{"Time":"2025-03-01T10:25:26"}}
10:25:27.005 MQT: stat/tasmota_0266AB/STATUS11 = {"StatusSTS":{"Time":"2025-03-01T10:25:26","Uptime":"0T00:02:25","UptimeSec":145,"Heap":163,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":21,"MqttCount":1,"Berry":{"HeapUsed":6,"Objects":87},"POWER1":"ON","POWER2":"ON","Dimmer1":100,"POWER3":"OFF","Dimmer2":100,"Color":"001EFF00FF","HSBColor":"233,100,100","White":100,"CT":500,"Channel":[0,12,100,0,100],"Scheme":0,"Fade":"OFF","Speed":1,"LedTable":"ON","Wifi":{"AP":1,"SSId":"WLAN","BSSId":"2C:3A:FD:BC:B5:1F","Channel":6,"Mode":"HT40","RSSI":94,"Signal":-53,"LinkCount":1,"Downtime":"0T00:00:03"}}}

```
- [ ] Set `weblog` to 4 and then, when you experience your issue, provide the output of the Console log:
```lua
  Console output here:
10:25:51.143 CMD: weblog 4
10:25:51.151 MQT: stat/tasmota_0266AB/RESULT = {"WebLog":4}
10:25:51.650 CFG: Saved, Count 17970, Bytes 4096
10:26:05.712 WIF: Checking connection...
10:26:08.987 SRC: WebGui from 172.16.0.20
10:26:08.000 MQT: stat/tasmota_0266AB/RESULT = {"POWER2":"OFF"}
10:26:09.003 MQT: stat/tasmota_0266AB/POWER2 = OFF
10:26:09.046 SRC: Berry
10:26:09.054 MQT: stat/tasmota_0266AB/RESULT = {"POWER1":"OFF"}
10:26:09.058 MQT: stat/tasmota_0266AB/POWER1 = OFF
10:26:09.060 Setting lamp main power: false
10:26:09.674 CFG: Saved, Count 17971, Bytes 4096
10:26:25.263 SRC: WebGui from 172.16.0.20
10:26:25.277 MQT: stat/tasmota_0266AB/RESULT = {"POWER2":"ON"}
10:26:25.279 MQT: stat/tasmota_0266AB/POWER2 = ON
10:26:25.319 SRC: Berry
10:26:25.327 MQT: stat/tasmota_0266AB/RESULT = {"POWER1":"ON"}
10:26:25.335 MQT: stat/tasmota_0266AB/POWER1 = ON
10:26:25.347 Setting lamp main power: true
10:26:25.674 CFG: Saved, Count 17972, Bytes 4096
10:26:25.689 WIF: Checking connection...

```

### TO REPRODUCE
_Steps to reproduce the behavior:_

### EXPECTED BEHAVIOUR
When turning on T2, both, the rgb and mainlight is turned on. But T2 should only turn on the rgb ceiling lights.

### SCREENSHOTS

[image]
[image]

### ADDITIONAL CONTEXT
_Add any other context about the problem here._

**(Please, remember to close the issue when the problem has been addressed)**

s-hadinger, 2025-03-01:
I will have a deeper look next week

DcFamas, 2025-03-01:
Thank you and no worries, take your time. :)

sfromis, 2025-03-02:
I see that you've activated `SetOption92` / `PWMCT` which is a special feature for a few types of lamp, not including this one. Try disabling this, and see how it works then.

DcFamas, 2025-03-03:
I changed the SetOption92:

`09:28:39.438 CMD: SetOption92 0
09:28:39.439 SRC: WebConsole from 172.16.0.51
09:28:39.441 CMD: Grp 0, Cmd 'SETOPTION', Idx 92, Len 1, Pld 0, Data '0'
09:28:39.447 MQT: stat/tasmota_0266AB/RESULT = {"SetOption92":"OFF"}
09:28:40.249 CFG: Saved, Count 18057, Bytes 4096`

And now everything is fine. :) Just to be said; I haven't set the option on purpose. Is this maybe in the template or the solo1.bin? I flashed and reset the device several times and this is the first time I heard about SetOption92. Thank you, now we can sit on our couch again without the floodlight. :D

sfromis, 2025-03-03:
Great that you now have the light you want. And no, the option is neither in the template nor a default in the binary.

DcFamas, 2025-03-03:
Okay I don't know where this comes from. :/ But thank you for your support and have a great day!

closed 2025-03-03 by DcFamas