arendst/Tasmota #23098: Matter + Hubitat Elevation hub

apc300, 2025-03-03:
### PROBLEM DESCRIPTION
_A clear and concise description of what the problem is._
when adding a matter device to hubitat by pasting in the pairing code, it fails to connect

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [Y] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [Y] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [Y] Searched the problem in [discussions](https://github.com/arendst/Tasmota/discussions)
- [Y] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [Y] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [Y] Problem is not scripter related, in this case open a discussion and tag  gemu2015
- [Y] Device used (e.g., Sonoff Basic): esp32-wroom-v3 D1 mini
- [Y] Tasmota binary firmware version number used: 14.5.0
  - [Y] Pre-compiled
  - [ ] Self-compiled
- [Y] Flashing tools used: web, chromebook
- [Y] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```lua
  Configuration output here:

```
- [ ] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```lua
  Rules output here:

```
- [ ] Provide the output of this command: `Status 0`:
```lua
  STATUS 0 output here:

```
- [ ] Set `weblog` to 4 and then, when you experience your issue, provide the output of the Console log:
```lua
  Console output here:

```

### TO REPRODUCE
_Steps to reproduce the behavior:_
enable matter on tasmota32, click add new device on hubitat, click matter, enter code

[seriallog3.txt](https://github.com/user-attachments/files/19055550/seriallog3.txt)

### EXPECTED BEHAVIOUR
_A clear and concise description of what you expected to happen._

### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._

### ADDITIONAL CONTEXT
_Add any other context about the problem here._

**(Please, remember to close the issue when the problem has been addressed)**

s-hadinger, 2025-03-03:
I have moved the logs here:

```
ets Jul 29 2019 12:21:46

rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0030,len:268
ho 0 tail 12 room 4
load:0x40078000,len:13224
ho 0 tail 12 room 4
load:0x40080400,len:4
load:0x40080404,len:2928
entry 0x4008055c

00:00:00.001 HDW: ESP32-D0WD-V3 v3.1
00:00:00.024 UFS: FlashFS mounted with 304 kB free
00:00:00.035 CFG: Loaded from File, Count 17
00:00:00.045 QPC: Count 1
00:00:00.046 CFG: CR 414/699, Busy 0
00:00:00.050 TYA: Active=0
00:00:00.056 ROT: Mode 1
00:00:00.083 CFG: No '*.autoconf' file found
00:00:00.110 MTR: Load_config = {}
00:00:00.178 MTR: Manual pairing code: 32968512601
00:00:00.186 RSL: RESULT = {"Matter":{"Commissioning":1,"PairingCode":"32968512601","QRCode":"MT:Y.K90C0R15RG4A3MG00"}}
00:00:00.512 BRY: Berry initialized, RAM used 10586 bytes
00:00:00.519 BRY: No 'preinit.be'
00:00:00.521 SRC: Restart
00:00:00.527 Project tasmota - Tasmota Version 14.5.0(release-tasmota32)-3_1_1(2025-02-18T14:37:27)
00:00:00.554 BRY: No 'autoexec.be'
00:00:00.836 WIF: Attempting connection...
00:00:02.000 WIF: Connecting to AP1 WildWireless Channel 11 BSSId 00:1E:42:2E:71:E7 in mode HT40 as tasmota-E75DE8-7656...
00:00:03.025 WIF: IPv4 192.168.1.210, mask 255.255.255.0, gateway 192.168.1.1
00:00:03.281 MTR: +Add_Event (CRIT      4001) [00]0028/00 (StartUp) - 14.5.0.0
00:00:03.286 MTR: +Add_Event (CRIT      4002) [00]0033/03 (BootReason) - 1U
00:00:03.287 MTR: Configuring endpoints
00:00:03.288 MTR:   endpoint =     0 type:root
00:00:03.300 MTR:   endpoint =     1 type:aggregator
00:00:03.302 RSL: RESULT = {"Matter":{"Initialized":1}}
00:00:03.306 MTR: Starting UDP server on port: 5540
00:00:03.319 MTR: start mDNS on wifi host '345F45E75DE8.local'
00:00:03.993 WIF: IPv6 Local fe80::365f:45ff:fee7:5de8%st1
00:00:04.095 WIF: Connected
00:00:04.154 MTR: starting mDNS on wifi '35B3D97169AF7F3A' ptr to `345F45E75DE8.local`
00:00:04.154 MTR: adding subtype: _L3421
00:00:04.156 MTR: adding subtype: _S13
00:00:04.167 MTR: adding subtype: _V65521
00:00:04.168 MTR: adding subtype: _CM1
00:00:04.365 HTP: Web server active on tasmota-E75DE8-7656 with IP address 192.168.1.210
00:00:05.224 RTC: UTC 2025-03-03T15:15:25Z, DST 2025-03-30T02:00:00, STD 2025-10-26T03:00:00
16:15:25.000 RTC: Synced by NTP
16:15:25.113 RSL: INFO1 = {"Info1":{"Module":"ESP32-DevKit","Version":"14.5.0(release-tasmota32)","FallbackTopic":"cmnd/DVES_E75DE8_fb/","GroupTopic":"cmnd/tasmotas/"}}
16:15:25.129 RSL: INFO2 = {"Info2":{"WebServerMode":"Admin","Hostname":"tasmota-E75DE8-7656","IPAddress":"192.168.1.210","IP6Global":"","IP6Local":"fe80::365f:45ff:fee7:5de8%st1"}}
16:15:25.143 RSL: INFO3 = {"Info3":{"RestartReason":"Vbat power on reset","BootCount":9}}
16:15:25.147 RSL: RESULT = {"POWER":"OFF"}
16:15:25.149 RSL: POWER = OFF
16:15:25.871 HTP: Logging
16:15:26.956 QPC: Reset
16:15:27.305 HTP: Configuration
16:15:28.948 APP: Boot Count 9
16:15:28.955 RSL: STATE = {"Time":"2025-03-03T16:15:28","Uptime":"0T00:00:08","UptimeSec":8,"Heap":131,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":0,"Berry":{"HeapUsed":11,"Objects":150},"POWER":"OFF","Wifi":{"AP":1,"SSId":"WildWireless","BSSId":"00:1E:42:2E:71:E7","Channel":11,"Mode":"HT40","RSSI":70,"Signal":-65,"LinkCount":1,"Downtime":"0T00:00:04"}}
16:15:29.768 HTP: Main Menu
16:15:29.001 CFG: Saved, Count 18, Bytes 4096
16:15:39.599 MTR: >Received  (     0) PBKDFParamRequest from [fe80::36e1:d1ff:fe81:1b9e%st1]:38884
16:15:39.609 MTR: +Session   ( 57417) from '[fe80::36e1:d1ff:fe81:1b9e%st1]:38884'
16:15:39.625 MTR: <Replied   (     0) PBKDFParamResponse
16:15:39.668 MTR: >Received  (     0) PASE_Pake1 from [fe80::36e1:d1ff:fe81:1b9e%st1]:38884
16:15:40.294 MTR: <Replied   (     0) PASE_Pake2
16:15:40.298 MTR: New Commissioning (PASE id=57417) from [fe80::36e1:d1ff:fe81:1b9e%st1]:38884
16:15:40.490 MTR: >Received  (     0) PASE_Pake3 from [fe80::36e1:d1ff:fe81:1b9e%st1]:38884
16:15:40.500 MTR: <Replied   (     0) StatusReport
16:17:09.981 MTR: -Session   (     0) (removed)
16:17:12.577 MTR: >Received  (     0) PBKDFParamRequest from [fe80::36e1:d1ff:fe81:1b9e%st1]:38884
16:17:12.587 MTR: +Session   ( 34298) from '[fe80::36e1:d1ff:fe81:1b9e%st1]:38884'
16:17:12.602 MTR: <Replied   (     0) PBKDFParamResponse
16:17:12.639 MTR: >Received  (     0) PASE_Pake1 from [fe80::36e1:d1ff:fe81:1b9e%st1]:38884
16:17:13.269 MTR: <Replied   (     0) PASE_Pake2
16:17:13.273 MTR: New Commissioning (PASE id=34298) from [fe80::36e1:d1ff:fe81:1b9e%st1]:38884
16:17:13.298 MTR: >Received  (     0) PASE_Pake3 from [fe80::36e1:d1ff:fe81:1b9e%st1]:38884
16:17:13.308 MTR: <Replied   (     0) StatusReport
```

s-hadinger, 2025-03-03:
You haven't declared any endpoint, so the bridge has no behavior:

```
00:00:03.288 MTR:   endpoint =     0 type:root
00:00:03.300 MTR:   endpoint =     1 type:aggregator**
```

Also the PASE session seems to fail. Have you used the QRCODE?

apc300, 2025-03-03:
what's an end-point? and i've seen bridge mode but haven't set up anything yet, i'd rather get one device paired and tested before i try bridge

[image]
this is all i've got on the web interface for my hub, so no i have not used QR code

apc300, 2025-03-03:
i did a bit more digging and there's a suggestion that 'an appropriate driver is required'
i'm not sure if this means it will pull built in setups for 'lamp' or 'button'
or instead if a complex driver is needed to handle everything tasmota could throw at it,
i can only guess since i am unfamiliar with it all

https://docs2.hubitat.com/en/how-to/use-matter
https://docs2.hubitat.com/en/developer/driver/building-a-matter-driver

s-hadinger, 2025-03-04:
Please start by reading the Tasmota Matter documentation. I have no clue what you did since you didn't use the QRcode 

https://tasmota.github.io/docs/Matter/

apc300, 2025-03-04:
Steps to reproduce the behavior:
enable matter on tasmota32, click add new device on hubitat, click matter, enter code

apc300, 2025-03-04:
From the logs 
16:15:40.500 MTR: <Replied   (     0) StatusReport
Is not enough for hubitat to connect, as there is no response

s-hadinger, 2025-03-04:
> Steps to reproduce the behavior: enable matter on tasmota32, click add new device on hubitat, click matter, enter code

What code did you enter? The logs looks like the code is not accepted by your controller.

> From the logs 16:15:40.500 MTR: <Replied ( 0) StatusReport Is not enough for hubitat to connect, as there is no response

Tasmota did respond a `StatusReport` to the controller but then nothing happens. You should definitely look at logs on the controller side, everything looks ok on Tasmota side. And it is already confirmed to work with: Alexa, Apple Home, Google Home and chip-controller (reference implementation)

apc300, 2025-03-04:
Copy paste the code from the tasmota devices webpage into the hubitat web page

I have looked and there are no logs from the controller about matter function, unless there is already a device set up, then perhaps there will be a logs page

s-hadinger, 2025-03-04:
Asking again the same question: what code did you enter?

`32968512601`
`MT:Y.K90C0R15RG4A3MG00`
Or something else?

apc300, 2025-03-04:
https://youtu.be/_tXIhljDVII
I took a video of the process

apc300, 2025-03-04:
well, it would be great to have info from the hubitat side, i imagine that would have to be raised to their support site, and suggested as a feature, i'm really not sure how long that would take
there are actually clues here, hubitat is 'trying to connect'
and tasmota is waiting for response from 'StatusReport'
the clues here being that hubitat will not respond to status report?
this does not time out on the tasmota side, and whatever packet it would send next, to possible make hubitat see a connection as successful, does not get transmitted

I have no control over the hubitat system but it might be useful to be able to skip the status report packet?
that's all i can guess from the clues

apc300, 2025-03-04:
i've scanned the documentation pages on hubitat, but nothing jumps out to me as relevant to the pairing functions, it all seems relevant to device functions

apc300, 2025-03-04:
https://xdaforums.com/t/hubitat-elevation-c7-rooting-guide.4178837/
it seems there's a possible way to get SSH access, to the device, this might be a faster way to get access to logs than waiting 2 years for a feature request to happen? looks like a pain but possible

s-hadinger, 2025-03-04:
> and tasmota is waiting for response from 'StatusReport'

Not exactly. The provisioning process is made of many requests/responses from the controller. The `StatusReport` is just Tasmota saying that the first establishment of the encrypted session is complete. Tasmota now waits for further instructions and requests, but nothing comes.

The first phase shows that there is no network problem, since the multiple steps went ok. So either the StatusResponse is considered incorrect by the controller (why?) or their implementation is faulty.

apc300, 2025-03-06:
Which would be the first direction to explore this to get a step closer?

s-hadinger, 2025-03-06:
There's not much to explore unfortunately. One thing you could try is setting `seriallog 4` to have more logs. I doubts it tells us more but who knows?

apc300, 2025-03-07:
Getting ssh access to the hub not a good direction?

Jason2866, 2025-03-08:
Moving to discussions, since it is most probably not an issue.