arendst/Tasmota #23138: Possible small bug in BerryInit

niteria, 2025-03-11:
I was reading the code and run into these two lines of code:

https://github.com/arendst/Tasmota/blob/master/tasmota/tasmota_xdrv_driver/xdrv_52_9_berry.ino#L384-L385

I think `ret_code2` is supposed to be used in the `if` statement.

s-hadinger, 2025-03-12:
Yes you are right. Well spotted

sfromis, 2025-03-12:
For fun, I tried asking Google Gemini if it could spot an error within that function, and it correctly identified it.

> The second `if` condition checks `ret_code1` again, but it should be checking `ret_code2`. `ret_code1` stores the result of `be_loadstring`, while `ret_code2` stores the result of `be_pcall`. We need to check if the execution of the Berry code (pcall) was successful.

s-hadinger, 2025-03-12:
Did you run it through the entire Tasmota code base?

sfromis, 2025-03-12:
Nah, it was just a test shot with that one function from `xdrv_52_9_berry.ino`. Still interesting to see how using AI tools can be useful in a development process. Asking it to look into the whole project from just the simple "chat" user interface resulted in general suggestions for code analysis, but outside the scope of that way of using it.

Going for the whole project would probably have to start from an integration with an IDE like VS Code. And Google does have a "Cloud Code for VS Code extension". Of course, other AI coding tools should be available too.

Taking a "random" driver like `xsns_09_bmp.ino`, it still did give quite a bit of output, mostly things of a general nature, including lauding some parts, but also pointed out things like many magic numbers, and concluded with some Recommendations
* Add memory deallocation (free) for dynamically allocated memory.
* Improve error reporting and logging.
* Simplify complex calculations and add comments.
* Consider using non-blocking delays.
* Thoroughly test the I2C communication and sensor data calculations.
* Add more comments to the code.
* Reduce the number of global variables.
While those things make sense as "good programming practice", there were no red flags, but still things worth considering in case of refactoring the code.

s-hadinger, 2025-03-12:
Thanks, I tried as well with Amazon Q for developer, it also spotted the bug above, and made some recommendations. I need to figure out how to make it more extensive

Jason2866, 2025-03-13:
I have activated [coderabbitai](https://github.com/apps/coderabbitai) in some of my repos.
The summary and code review in a PR is most of the time good and helpful

PR #23146 (2025-03-13): Fix unused ret_code2 in xdrv_52_9_berry
## Description:

**Related issue (if applicable):** fixes #23138

## Checklist:
  - [X] The pull request is done against the latest development branch
  - [X] Only relevant files were touched
  - [X] Only one feature/fix was added per PR and the code change compiles without warnings
  - [ ] The code change is tested and works with Tasmota core ESP8266 V.2.7.8
  - [ ] The code change is tested and works with Tasmota core ESP32 V.3.1.3.250302
  - [X] I accept the [CLA](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md#contributor-license- [... cut]