arendst/Tasmota #23157: GPIO1 and GPIO2 will be rewrited from Button to output from Tasmota

elektro-freak, 2025-03-16:
### PROBLEM DESCRIPTION
Tasmota overwrites input GPIO1 and GPIO2 to outputs when toggling one of them on ESP32S2

### REQUESTED INFORMATION

- [x ] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [x ] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [ x] Searched the problem in [discussions](https://github.com/arendst/Tasmota/discussions)
- [ x] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [ x] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [ x] Problem is not scripter related, in this case open a discussion and tag  gemu2015
- [ x] Device used (e.g., Sonoff Basic): ESP32S2
- - [x ] Tasmota binary firmware version number used: 14.5.0
  - [x ] Pre-compiled
  - [ ] Self-compiled
- [ x] Flashing tools used: webinstaller tasmota
- [ ] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```lua
  Configuration output here:

```
- [ ] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```lua
  Rules output here:

```
- [ ] Provide the output of this command: `Status 0`:
```lua
  STATUS 0 output here:

15:50:32.720 CMD: status 0
15:50:32.724 RSL: STATUS = {"Status":{"Module":0,"DeviceName":"Tasmota","FriendlyName":["Tasmota","Tasmota2","Tasmota3","Tasmota4","","","",""],"Topic":"tasmota_4B2BC2","ButtonTopic":"0","Power":"0000000000","PowerLock":"0000000000","PowerOnState":3,"LedState":1,"LedMask":"FFFF","SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0,"InfoRetain":0,"StateRetain":0,"StatusRetain":0}}
15:50:32.727 RSL: STATUS1 = {"StatusPRM":{"Baudrate":115200,"SerialConfig":"8N1","GroupTopic":"tasmotas","OtaUrl":"http://ota.tasmota.com/tasmota32/release/tasmota32s2.bin","RestartReason":"Software reset CPU","Uptime":"0T00:00:31","StartupUTC":"2025-03-16T14:50:01","Sleep":50,"CfgHolder":4617,"BootCount":16,"BCResetTime":"2024-05-27T06:47:29","SaveCount":44}}
15:50:32.730 RSL: STATUS2 = {"StatusFWR":{"Version":"14.5.0(release-tasmota32)","BuildDateTime":"2025-02-18T14:37:37","Core":"3_1_1","SDK":"5.3.2.250120","CpuFrequency":240,"Hardware":"ESP32-S2R2 v1.0","CR":"440/699"}}
15:50:32.733 RSL: STATUS3 = {"StatusLOG":{"SerialLog":0,"WebLog":2,"MqttLog":0,"FileLog":0,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["verschluesselt",""],"TelePeriod":300,"Resolution":"558180C0","SetOption":["0000C00B","2805C80001000600003C5A0A192800000000","40800080","00006000","00004000","00020000"]}}
15:50:32.745 RSL: STATUS4 = {"StatusMEM":{"ProgramSize":1946,"Free":933,"Heap":111,"StackLowMark":4,"PsrMax":2048,"PsrFree":2017,"ProgramFlashSize":4096,"FlashSize":4096,"FlashChipId":"164020","FlashFrequency":80,"FlashMode":"DIO","Features":["0809","9F9AD7DF","0015A001","B7F7BFCF","05DA9BC4","E0360DC7","480840D2","20200000","D4BC482D","810A80F1","00000814"],"Drivers":"1,2,!3,!4,!5,7,!8,9,10,11,12,!14,!16,!17,!20,!21,!24,26,27,29,!34,!35,38,50,52,!59,!60,62,!63,!66,!67,!68,!73,!75,82,!86,!87,!88,!91,!121","Sensors":"1,2,3,5,6,7,8,9,10,11,12,13,14,15,17,18,19,20,21,22,26,31,34,37,39,40,42,43,45,51,52,55,56,58,59,64,66,67,74,85,92,95,98,103,105,109,127","I2CDriver":"7,8,9,10,11,12,13,14,15,17,18,20,24,29,31,36,41,42,44,46,48,58,62,65,69,76,77,82,89"}}
15:50:32.755 RSL: STATUS5 = {"StatusNET":{"Hostname":"tasmota-4B2BC2-3010","IPAddress":"192.168.200.141","Gateway":"192.168.200.1","Subnetmask":"255.255.255.0","DNSServer1":"192.168.200.1","DNSServer2":"2003:e7:3707:4400:464e:6dff:fe64:7f34","Mac":"80:65:99:4B:2B:C2","IP6Global":"2003:e7:3707:4400:8265:99ff:fe4b:2bc2","IP6Local":"fe80::8265:99ff:fe4b:2bc2%st1","Ethernet":{"Hostname":"","IPAddress":"0.0.0.0","Gateway":"0.0.0.0","Subnetmask":"0.0.0.0","DNSServer1":"192.168.200.1","DNSServer2":"2003:e7:3707:4400:464e:6dff:fe64:7f34","Mac":"00:00:00:00:00:00","IP6Global":"","IP6Local":""},"Webserver":2,"HTTP_API":1,"WifiConfig":5,"WifiPower":15.0}}
15:50:32.765 RSL: STATUS6 = {"StatusMQT":{"MqttHost":"","MqttPort":1883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_4B2BC2","MqttUser":"DVES_USER","MqttCount":0,"MqttTLS":0,"MAX_PACKET_SIZE":1200,"KEEPALIVE":30,"SOCKET_TIMEOUT":4}}
15:50:32.770 RSL: STATUS7 = {"StatusTIM":{"UTC":"2025-03-16T14:50:32Z","Local":"2025-03-16T15:50:32","StartDST":"2025-03-30T02:00:00","EndDST":"2025-10-26T03:00:00","Timezone":"+01:00","Sunrise":"07:00","Sunset":"18:56"}}
15:50:32.778 RSL: STATUS10 = {"StatusSNS":{"Time":"2025-03-16T15:50:32","Shutter1":{"Position":0,"Direction":0,"Target":0,"Tilt":0},"Shutter2":{"Position":100,"Direction":0,"Target":100,"Tilt":0},"Shutter3":{"Position":16,"Direction":0,"Target":16,"Tilt":0},"Shutter4":{"Position":0,"Direction":0,"Target":0,"Tilt":0},"Shutter5":{"Position":0,"Direction":0,"Target":0,"Tilt":0}}}
15:50:32.787 RSL: STATUS11 = {"StatusSTS":{"Time":"2025-03-16T15:50:32","Uptime":"0T00:00:31","UptimeSec":31,"Heap":111,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":24,"MqttCount":0,"Berry":{"HeapUsed":4,"Objects":46},"POWER1":"OFF","POWER2":"OFF","POWER3":"OFF","POWER4":"OFF","POWER5":"OFF","POWER6":"OFF","POWER7":"OFF","POWER8":"OFF","POWER9":"OFF","POWER10":"OFF","Wifi":{"AP":1,"SSId":"verschluesselt","BSSId":"48:5D:35:63:1C:B9","Channel":6,"Mode":"HE20","RSSI":100,"Signal":-39,"LinkCount":1,"Downtime":"0T00:00:03"}}}
15:50:32.795 RSL: STATUS13 = {"StatusSHT":{"SHT0":{"Relay1":1,"Relay2":2,"Open":100,"Close":100,"50perc":50,"Delay":0,"Opt":"0000","Calib":[300,500,700,900,1000],"Mode":"1","TiltConfig":[0,0,0,0,0]},"SHT1":{"Relay1":3,"Relay2":4,"Open":100,"Close":100,"50perc":50,"Delay":0,"Opt":"0000","Calib":[300,500,700,900,1000],"Mode":"1","TiltConfig":[0,0,0,0,0]},"SHT2":{"Relay1":5,"Relay2":6,"Open":100,"Close":100,"50perc":50,"Delay":0,"Opt":"0000","Calib":[300,500,700,900,1000],"Mode":"1","TiltConfig":[0,0,0,0,0]},"SHT3":{"Relay1":7,"Relay2":8,"Open":100,"Close":100,"50perc":50,"Delay":0,"Opt":"0000","Calib":[300,500,700,900,1000],"Mode":"1","TiltConfig":[0,0,0,0,0]},"SHT4":{"Relay1":9,"Relay2":10,"Open":100,"Close":100,"50perc":50,"Delay":0,"Opt":"0000","Calib":[300,500,700,900,1000],"Mode":"1","TiltConfig":[0,0,0,0,0]}}}

```
- [ ] Set `weblog` to 4 and then, when you experience your issue, provide the output of the Console log:
```lua
  Console output here:

```

### TO REPRODUCE
I find that out with my shutter but if you take a blank esp32s2 configure the gpio1 and gpio2 as a button.
Than taglle them and you  will see in the gpio viewer that tasmota will rewrite both gpio to be an output instead of an input with pullup and it puts them to down. It is no  more longer able to be used as an input.
The ports stay to be an output also when your restart the esp.

### EXPECTED BEHAVIOUR
_A clear and concise description of what you expected to happen._

### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._

### ADDITIONAL CONTEXT
_Add any other context about the problem here._

**(Please, remember to close the issue when the problem has been addressed)**

sfromis, 2025-03-16:
You left out important details asked for in the form, to be able to see exact details of your gpio config.

Trying to replicate with a small test on an ESP32-S2, gpio 1+2 configured as "Button" inputs, I can toggle both multiple times, without any issue.

Your statement about "see in the gpio viewer that tasmota will rewrite both gpio to be an output instead of an input with pullup and it puts them to down" is not clear, as you do not tell what you see when you make that conclusion, and I can't recognize any of your issues. When I have two other pins configured for output (Relay), the gpio viewer has no indication of pins being input or output, only if they are high or low.

What I see in the gpio viewer is that both gpio 1 and 2 stays red (with text "High") after power-on boot with the gpio 1+2 config, and stays that way all the time, except when pushing either button, where the matching gpio indicator for each pin becomes green (with text "Low"), and pushing a button makes it appear as green while pushed.

What I see is the expected behavior without any issues.

elektro-freak, 2025-03-16:
Hello Thanks for coming back to my issue.The problem acours when you put the pins to GND. So not pushing the button in the gui.In addition to the gpio viewer. If you klick on a gpio a window pops up and say how the port is configured, like input with pull-up or output.Thanks

sfromis, 2025-03-16:
You are still not providing the details you "forgot" add first time. That's important info to be able to see what Tasmota is doing.

And again, when trying to replicate, I'm not seeing anything like the behavior you mention. That also goes for the content of the popup wjem clicking on a gpio dot in the gpio viewer. My input pins stays as "Pin Mode: Input with Pull-up" through several cycles of clicking. But as you did not provide the relevant details, no basis for speculating about what your situation is.

elektro-freak, 2025-03-17:
Thanks again,

 

I did a freahs install on my esp32s2 this morning with erasing the flash and install tasmota 14.5.0 with the webinstaller.

After installing I bind it to my wifi and configure the gpio0 to be button 1 and gpio2 to be button 2 under configuration module.

No other configuration. That‘s all.

Restart device. 

 

And now it works all fine. Looking on the release page (https://ota.tasmota.com/tasmota32/) I saw that the binarys are now from yesterday

 

Doing again a update via ota to my shutter device  (using http://ota.tasmota.com/tasmota32/release/tasmota32s2.bin)  I’m running again in the issue.

 

Picking up the file manualy from this page https://ota.tasmota.com/tasmota32/ and install the issue is gone.

Comparing the files https://ota.tasmota.com/tasmota32/tasmota32s2.bin and http://ota.tasmota.com/tasmota32/release/tasmota32s2.bin it looks like they are different.

 

So I think there was an issue with the gpio in the old comiled version but it is not deployed to all path on the server where it should be.

 

Thanks 

elektro-freak

 

Pic

Picking up the file manually 

You are still not providing the details you "forgot" to add first time. That's important info to be able to see what Tasmota is doing.

sfromis, 2025-03-17:
Again, do fill in all the requested info in the form used to create the issue, instead of merely posting some words about about your impression of events.

I'd also recommend going into the Github thread when you read or post to the issue, instead of having your email client post reams of quoted text making the "real" thread hardly readable in an ugly way.

closed 2025-03-17 by elektro-freak