arendst/Tasmota #23165: TuyaMCU energy meter is not reporting in the first SENSOR telemetry

ortegafernando, 2025-03-18:
### PROBLEM DESCRIPTION
Hi, I am using an EMYLO EAI-90. ZMAI-90 clone. It is an enegy meter

### REQUESTED INFORMATION
- [x] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [x] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [x] Searched the problem in [discussions](https://github.com/arendst/Tasmota/discussions)
- [x] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [ ] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [x] Problem is not scripter related, in this case open a discussion and tag  gemu2015
- [x] Device used (e.g., Sonoff Basic): Emylo EAI-90
- [X] Tasmota binary firmware version number used: 14.5.0(release-tasmota)-2_7_8(2025-02-18T14:34:10)
  - [X] Pre-compiled
  - [ ] Self-compiled
- [X] Flashing tools used: OTA
- [X] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```lua
  Configuration output here:
 bi/test/POWD1/stat/RESULT = {"NAME":"Generic","GPIO":[1,1,1,1,1,1,1,1,1,1,1,1,1,1],"FLAG":0,"BASE":18}
21:02:32.816 MQT: bi/test/POWD1/stat/RESULT = {"Module":{"54":"Tuya MCU"}}
21:02:33.024 MQT: bi/test/POWD1/stat/RESULT = {"GPIO0":{"0":"None"},"GPIO1":{"2272":"Tuya Tx"},"GPIO2":{"0":"None"},"GPIO3":{"2304":"Tuya Rx"},"GPIO4":{"0":"None"},"GPIO5":{"0":"None"},"GPIO9":{"0":"None"},"GPIO10":{"0":"None"},"GPIO12":{"0":"None"},"GPIO13":{"0":"None"},"GPIO14":{"0":"None"},"GPIO15":{"0":"None"},"GPIO16":{"0":"None"},"GPIO17":{"0":"None"}}

```
- [ ] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```lua
NO RULES
```
- [ ] Provide the output of this command: `Status 0`:
```lua
21:03:25.691 MQT: bi/test/POWD1/stat/STATUS = {"Status":{"Module":54,"DeviceName":"Tasmota","FriendlyName":["Tasmota"],"Topic":"POWD1","ButtonTopic":"","Power":"1","PowerLock":"0","PowerOnState":3,"LedState":1,"LedMask":"FFFF","SaveData":1,"SaveState":1,"SwitchTopic":"0","SwitchMode":[0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0],"ButtonRetain":0,"SwitchRetain":0,"SensorRetain":0,"PowerRetain":0,"InfoRetain":0,"StateRetain":0,"StatusRetain":0}}
21:03:25.699 MQT: bi/test/POWD1/stat/STATUS1 = {"StatusPRM":{"Baudrate":9600,"SerialConfig":"8N1","GroupTopic":"tasmotas","OtaUrl":"http://ota.tasmota.com/tasmota/release/tasmota.bin.gz","RestartReason":"Software/System restart","Uptime":"0T00:09:47","StartupUTC":"2025-03-18T19:53:38","Sleep":50,"CfgHolder":4617,"BootCount":73,"BCResetTime":"2025-03-08T19:46:05","SaveCount":176,"SaveAddress":"F4000"}}
21:03:25.704 MQT: bi/test/POWD1/stat/STATUS2 = {"StatusFWR":{"Version":"14.5.0(release-tasmota)","BuildDateTime":"2025-02-18T14:34:10","Boot":31,"Core":"2_7_8","SDK":"2.2.2-dev(38a443e)","CpuFrequency":80,"Hardware":"ESP8266EX","CR":"378/699"}}
21:03:25.710 MQT: bi/test/POWD1/stat/STATUS3 = {"StatusLOG":{"SerialLog":0,"WebLog":2,"MqttLog":0,"SysLog":0,"LogHost":"","LogPort":514,"SSId":["xxxxxxxxxx",""],"TelePeriod":300,"Resolution":"558180C0","SetOption":["00C08009","2805C80001000600003C5A0A192800000000","00000080","00006000","00004000","00000000"]}}
21:03:25.724 MQT: bi/test/POWD1/stat/STATUS4 = {"StatusMEM":{"ProgramSize":652,"Free":348,"Heap":21,"ProgramFlashSize":1024,"FlashSize":2048,"FlashChipId":"1540C8","FlashFrequency":40,"FlashMode":"DOUT","Features":["0809","8F9AC787","04368001","000000CF","010013C0","C000F981","00004004","00001000","54000020","00000080","00000000"],"Drivers":"1,2,3,!4,!5,!6,7,!8,9,10,12,16,!18,!19,!20,!21,!22,!24,26,!27,29,!30,!35,!37,!45,62,!68","Sensors":"1,2,3,4,5,6","I2CDriver":"7"}}
21:03:25.730 MQT: bi/test/POWD1/stat/STATUS5 = {"StatusNET":{"Hostname":"POWD1-5496","IPAddress":"192.168.1.xxx","Gateway":"192.168.1.1","Subnetmask":"255.255.255.0","DNSServer1":"192.168.1.1","DNSServer2":"0.0.0.0","Mac":"xxxxxxx","Webserver":2,"HTTP_API":1,"WifiConfig":2,"WifiPower":17.0}}
21:03:25.736 MQT: bi/test/POWD1/stat/STATUS6 = {"StatusMQT":{"MqttHost":"xxxxxx","MqttPort":xxxx,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_631578","MqttUser":"test","MqttCount":1,"MqttTLS":0,"MAX_PACKET_SIZE":1200,"KEEPALIVE":30,"SOCKET_TIMEOUT":4}}
21:03:25.742 MQT: bi/test/POWD1/stat/STATUS7 = {"StatusTIM":{"UTC":"2025-03-18T20:03:25Z","Local":"2025-03-18T21:03:25","StartDST":"2025-03-30T02:00:00","EndDST":"2025-10-26T03:00:00","Timezone":"+01:00","Sunrise":"06:56","Sunset":"18:59"}}
21:03:25.746 MQT: bi/test/POWD1/stat/STATUS9 = {"StatusPTH":{"PowerDelta":0,"PowerLow":0,"PowerHigh":0,"VoltageLow":0,"VoltageHigh":0,"CurrentLow":0,"CurrentHigh":0,"MaxPower":0,"MaxPowerHold":10,"MaxPowerWindow":30,"MaxEnergy":0,"MaxEnergyStart":0}}
21:03:25.845 MQT: bi/test/POWD1/stat/STATUS10 = {"StatusSNS":{"Time":"2025-03-18T21:03:25","ENERGY":{"TotalStartTime":"2025-03-08T19:46:05","Total":0.000,"Yesterday":0.000,"Today":0.000,"Power":0,"ApparentPower":0,"ReactivePower":0,"Factor":0.00,"Voltage":234,"Current":0.000}}}
21:03:25.852 MQT: bi/test/POWD1/stat/STATUS11 = {"StatusSTS":{"Time":"2025-03-18T21:03:25","Uptime":"0T00:09:47","UptimeSec":587,"Heap":19,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":23,"MqttCount":1,"POWER":"ON","Wifi":{"AP":1,"SSId":"xxxxxxx","BSSId":"xxxxxxxx","Channel":4,"Mode":"11n","RSSI":100,"Signal":-44,"LinkCount":1,"Downtime":"0T00:00:05"}}}

```
- [ ] Set `weblog` to 4 and then, when you experience your issue, provide the output of the Console log:
```lua
  Console output here:

```

### TO REPRODUCE
Only switch on, I have already read this 
https://templates.blakadder.com/ZMAi-90.html
About different versions, but even with rules "Rule1 on System#Boot do RuleTimer1 10 endon on Rules#Timer=1 do backlog TuyaSend8; RuleTimer1 10 endon" there is no way to get the first SENSOR telemetries with voltage value equals 0 (All values are 0, during my tests I have not connected anything)
I can not see any advantage using this Rule.
Even if I change teleperiod to 10 seconds and rule1 is off or on, same behaviour
See this console output, we can see that BEFORE the first sensor telemetry, TuyaMCU is reporting correctly.

```
21:11:13.309 {"TuyaReceived":{"Data":"55AA03070008140200040000092357","Cmnd":7,"CmndData":"1402000400000923","DpType2Id20":2339,"20":{"DpId":20,"DpIdType":2,"DpIdData":"00000923"}}}
21:11:13.312 TYA: fnId=33 is set for dpId=20
21:11:13.314 TYA: RX value 2339 from dpId 20 
21:11:13.316 TYA: Rx ID=20 Voltage=2339
```

is this important?:
```
21:11:14.298 QPC: Reset
21:11:15.243 NRG: Energy reset by invalid data
```

Complete console log:
[... 99 lines cut ...]
21:11:28.264 TYA: Send "55aa00000000ff"
21:11:28.277 DMP: 55 AA 03 00 00 01 01 04
21:11:28.279 {"TuyaReceived":{"Data":"55AA030000010104","Cmnd":0,"CmndData":"01"}}
21:11:28.281 TYA: Heartbeat
21:11:31.540 WIF: Checking connection...
21:11:37.263 MQT: bi/test/POWD1/tele/STATE = {"Time":"2025-03-18T21:11:37","Uptime":"0T00:00:31","UptimeSec":31,"Heap":24,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":1,"POWER":"ON","Wifi":{"AP":1,"SSId":"XXXXXXXX","BSSId":"XXXXXXXX","Channel":4,"Mode":"11n","RSSI":100,"Signal":-43,"LinkCount":1,"Downtime":"0T00:00:05"}}
21:11:37.274 MQT: bi/test/POWD1/tele/SENSOR = {"Time":"2025-03-18T21:11:37","ENERGY":{"TotalStartTime":"2025-03-08T19:46:05","Total":0.000,"Yesterday":0.000,"Today":0.000,"Period":0,"Power":0,"ApparentPower":0,"ReactivePower":0,"Factor":0.00,"Voltage":0,"Current":0.000}}
21:11:39.275 TYA: Send "55aa00000000ff"
21:11:39.288 DMP: 55 AA 03 00 00 01 01 04
21:11:39.290 {"TuyaReceived":{"Data":"55AA030000010104","Cmnd":0,"CmndData":"01"}}
21:11:39.293 TYA: Heartbeat
21:11:47.265 MQT: bi/test/POWD1/tele/STATE = {"Time":"2025-03-18T21:11:47","Uptime":"0T00:00:41","UptimeSec":41,"Heap":24,"SleepMode":"Dynamic","Sleep":50,"LoadAvg":19,"MqttCount":1,"POWER":"ON","Wifi":{"AP":1,"SSId":"XXXXXXXX","BSSId":"XXXXXXXX","Channel":4,"Mode":"11n","RSSI":100,"Signal":-44,"LinkCount":1,"Downtime":"0T00:00:05"}}
21:11:47.277 MQT: bi/test/POWD1/tele/SENSOR = {"Time":"2025-03-18T21:11:47","ENERGY":{"TotalStartTime":"2025-03-08T19:46:05","Total":0.000,"Yesterday":0.000,"Today":0.000,"Period":0,"Power":0,"ApparentPower":0,"ReactivePower":0,"Factor":0.00,"Voltage":0,"Current":0.000}}
```

### EXPECTED BEHAVIOUR
To get a value in the first SENSOR telemetry json data.

### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._

### ADDITIONAL CONTEXT
_Add any other context about the problem here._

**(Please, remember to close the issue when the problem has been addressed)**

ortegafernando, 2025-04-18:
Hi @Jason2866 better than close it, could you put it in "stand by"? Thanks.

Jason2866, 2025-04-18:
Bot closed.

closed 2025-05-18 by github-actions[bot]