arendst/Tasmota #23236: EnergyRex KNX not work

Sebastian3009, 2025-04-02:
### PROBLEM DESCRIPTION
When I set the EnergyRes to 0 it works on MQTT or in the console but not on KNX export.
The Export on KNX have many decimal places. I used Version Tasmota 14.5.0.3 (87b95f7-knx)

### REQUESTED INFORMATION
_Make sure your have performed every step and checked the applicable boxes before submitting your issue. Thank you!_

- [ ] Read the [Contributing Guide and Policy](https://github.com/arendst/Tasmota/blob/development/CONTRIBUTING.md) and [the Code of Conduct](https://github.com/arendst/Tasmota/blob/development/CODE_OF_CONDUCT.md)
- [ ] Searched the problem in [issues](https://github.com/arendst/Tasmota/issues)
- [ ] Searched the problem in [discussions](https://github.com/arendst/Tasmota/discussions)
- [ ] Searched the problem in the [docs](https://tasmota.github.io/docs/FAQ)
- [ ] Searched the problem in the [chat](https://discord.gg/Ks2Kzd4)
- [ ] Problem is not scripter related, in this case open a discussion and tag  gemu2015
- [ ] Device used (e.g., Sonoff Basic): Nous A1
- [ ] Tasmota binary firmware version number used: 14.5.0.3
  - [ ] Pre-compiled
  - [ ] Self-compiled
- [ ] Flashing tools used: was flashed
- [ ] Provide the output of command: `Backlog Template; Module; GPIO 255`:
```lua
  Configuration output here:

```
- [ ] If using rules, provide the output of this command: `Backlog Rule1; Rule2; Rule3`:
```lua
  Rules output here:

```
- [ ] Provide the output of this command: `Status 0`:
```lua
  STATUS 0 output here:

```
- [ ] Set `weblog` to 4 and then, when you experience your issue, provide the output of the Console log:
```lua
  Console output here:

```

### TO REPRODUCE
_Steps to reproduce the behavior:_

### EXPECTED BEHAVIOUR
_A clear and concise description of what you expected to happen._

### SCREENSHOTS
_If applicable, add screenshots to help explain your problem._

### ADDITIONAL CONTEXT
_Add any other context about the problem here._

**(Please, remember to close the issue when the problem has been addressed)**

arendst, 2025-04-05:
ALas, from what I see from the used library there is no means to change amount of shown decimals.

Sebastian3009, 2025-04-09:
Ok, when there is not solution I close thew issue.
Thank You

closed 2025-04-09 by Sebastian3009