arendst/Tasmota #23251: Accessing values from xdrv_10_scripter.ino

BerndPetermanns, 2025-04-05:
Hello.
Could someone please help me understand how the created variables are saved in xdrv_10_scripter.ino?
By default, 50 variables are specified – 45 of which are floats and 5 are strings. However, given the large number of uncommented "values" and their iterations, I don't understand where they are stored.

My problem is that I would like to access these variables from the script in xdrv_01_9_webserver.ino. I would like to display them there for testing purposes, in the Info menu. My problem now is accessing them.
The manual retrieval from the function SML_GetVal(); (Sensor.ino) always returns 0.00 as well as fvalues[]; (Scripter.ino).

Where is my error? Or where can the data be accessed?

Thank you very much
BernPetermanns

//////German

Könnte mir bitte jemand dabei helfen zu verstehen, wie in der xdrv_10_scripter.ino die angelegten Variablen abgespeichert werden?
Im Standardfall sind 50 Variablen vorgegeben - davon 45 Flots und 5 Strings. Ich verstehe bei der VIelzahl an unkommentierten "values" und deren Itterationen allerdings nicht, wo diese abgelegt werden?

Mein Problem ist es, dass ich gern auf diese Variablen aus dem Scripting in der xdrv_01_9_webserver.ino zugreifen möchte. Dort möchte ich diese zu Testzwekcne anzeigen lassen, in dem Info Menü. Mein Problem ist nun, auf diese zuzugreifen.
Der manuelle Abgriff aus der Funktion SML_GetVal(); (Sensor.ino) liefert immer 0,00 als auch fvalues[]; (Scripter.ino).

Wo liegt mein Fehler? Oder wo kann auf die Daten zugegriffen werden?

Vielen Dank
BernPetermanns

sfromis, 2025-04-05:
Creating an issue is not the way to ask questions, that should be in discussions instead. In your case, there even is a specific category for scripter/SML discussions:
https://github.com/arendst/Tasmota/discussions/categories/scripter-sml

sfromis, 2025-04-05:
That's no reason to post where it is off-topic.

I'm not using the scripter, but my general expectation would be that you can have your script produce html for your browser, including for debugging purposes, instead of trying to tear into internals.

BerndPetermanns, 2025-04-05:
I need the correct access or understanding of the internal storage process.
I'm aware of the test access via the URL, etc., but unfortunately it doesn't help me.
Do you perhaps have another idea or a documented scripter.ino?
Many thanks